esp8266/Arduino #8538: WiFiclient.setNoDelay has no effect until after connecting

bertmelis, 2022-04-10:
```cpp
WiFiClient client;
client.setDefaultNoDelay(false);
client.setNoDelay(true);  // no effect, Nagle stays enabled by previous default setting

client.connect(ip, port);
client.setNoDelay(true);  // Nagle disabled

client.close();
client.connect(ip, port);  // Nagle enabled again using default setting
```

WiFiclient has an option to set `noDelay` but this option is replaced during every call to connect by the default value. The default value is set by `WiFiClient::setDefaultNoDelay(bool)` but this sets a static variable so it might be overwritten by third party libraries.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.cpp#L174

A quick solution would be to add a non-static member variable and use this to set object specific options instead of using the static one.

The same applies for `wiFiClient::setSync(bool)` although there is a non-static member variable `_sync` in `ClientContext` so this can be used.

Obviously, a workaround is to set the options every time after connecting.

bertmelis, 2022-06-20:
I don't mind creating a PR. But I don't want to do it if it won't be considered.

Anyways, my opinion stays that the `noDelay` setting should persist during the lifetime of the `WiFiClient` object and not reset on every (re)connect as it is now.

Your thoughts?

bertmelis, 2022-06-20:
It is theoretically indeed, but that doesn't make it less true.

Why should disabling Nagle only be possible on an active connection? That doesn't seem to be consistent with setting a default, which is used before connecting.

Mind that I don't want to change any methods. I just want to persist noDelay on a per-client basis between reconnects and not return to the default.

bertmelis, 2022-06-20:
I created a hook so my derived class can do whatever it wants after connecting. Feel free to close this but my opinion is that this is counter-intuitive. I don't mind disagreeing 🤣 .

mcspr, 2022-06-21:
> Why should disabling Nagle only be possible on an active connection? That doesn't seem to be consistent with setting a default, which is used before connecting.

To avoid yet another state variable! :) See `tcp_{enable,disable}_nagle`, it's a flag set on internal TCP state object, which is only there when internal client object is 'active'

Suppose, we have a tri-state with default, enabled and disabled; like the WiFiServer does. If we don't call setNoDelay, it's default. If we do, it is the pre-set value. Still seems more complicated than the existing approach. Mostly, having default at all, I'd actually consider removing it if we really want some additional setting.

bertmelis, 2022-06-21:
I know the consequences. Coming from AVR, the ESP spoils us with memory but we shouldn't waste it.

I solved my problem. I'll close this for now.

I might create a PR to document the behaviour in the docs though. "Only to be called on an active connection".

closed 2022-06-21 by bertmelis