esp8266/Arduino #8547: There is heap-based buffer overflow

MucahitSaratar, 2022-04-21:
the function is vulnerable

you can fix it

https://medium.com/@Mr.deadbeef.py/esp-unusable-heap-based-buffer-overlow-ba6e8d99f813

mcspr, 2022-04-21:
We would appreciate a concise description of the issue, not the link to your blog :/
What's your suggestion for a fix here?

I think what you meant is that this line uses `char` to store the result of `strlen(username) + strlen(password) + 1`, and it must stay under 127 bytes to not overflow the value
https://github.com/esp8266/Arduino/blob/6c8c8cb2d6174ee1feb8ebc26a3127b64575e1d6/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h#L105

MucahitSaratar, 2022-04-21:
i prefer this solution, exactly, in this line should be validate string length lower than 127 bytes  before defining "toencodeLen" for username and password
if you are want to error exception or you can use snprintf(buffer,toencodeLen+1,...) instead of sprintf. 

the case is when remote user access the this function, they can trig to overflow

[image]

closed 2022-04-30 by mcspr