esp8266/Arduino #8550: After compiling with the latest version of GIT, the window does not display the memory usage

jie326513988, 2022-04-24:
After compiling with the latest version of GIT, the window does not display the memory usage

[image]

mcspr, 2022-04-24:
2nd window has Arduino Uno board selected, you'd need to switch to esp8266 first :)

jie326513988, 2022-04-25:
> 
Thanks for your reply. The 3.0.2 release I used before can show the percentage of memory usage, but it does not show after the CHANGE to THE GIT3.1.0 version, is there any need to set it.

mcspr, 2022-04-25:
Hm. Yeah, sorry, I missed the last part. Indeed, both 1.8.19 and arduino-cli latest build don't show anything
```
}
~\D\A\s\test> cat test.ino
void setup() {
}

void loop() {
}

~\D\A\s\test> C:\Users\maxim\Downloads\arduino-cli.exe compile -b esp8266com:esp8266:d1_mini
Executable segment sizes:
ICACHE : 32768           - flash instruction cache
IROM   : 232088          - code in flash         (default or ICACHE_FLASH_ATTR)
IRAM   : 26205   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...)
DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP
RODATA : 876   ) / 81920 - constants             (global, static) in RAM/HEAP
BSS    : 25520 )         - zeroed variables      (global, static) in RAM/HEAP

~\D\A\s\test>
```

Looking at arduino-cli code, it should come from this command
https://github.com/esp8266/Arduino/blob/02c1a502ca475d1bc202ba881fc5a2eb07a532d5/platform.txt#L61
https://github.com/esp8266/Arduino/blob/02c1a502ca475d1bc202ba881fc5a2eb07a532d5/platform.txt#L146

And diff'ing the platform.txt... it should've worked? The paths are there, for the local `tools/` dir
```
@@ -5,14 +5,14 @@
 # For more info:
 # https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

-name=ESP8266 Boards (3.0.2)
-version=3.0.2
+name=ESP8266 Boards (3.1.0-dev)
+version=3.1.0-dev

 # These will be removed by the packager script when doing a JSON release
+runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf
+runtime.tools.python3.path={runtime.platform.path}/tools/python3
```

mcspr, 2022-04-25:
Plus, adding `--show-properties` shows a path for the global packages dir instead of the one for the platform tools dir. I'd thought it would show the override instead.
```cmd
> arduino-cli compile --show-properties -b esp8266com:esp8266:d1_mini | select-string runtime.tools.xtensa-lx106-elf-gcc.path=
runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\maxim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed
```

mcspr, 2022-04-26:
So, the culprit is `upload.maximum_size` property. We don't have it set for any of the available boards, and with a slight tweak I'm finally able to see the size info.
```diff
@@ -3849,6 +3831,7 @@ d1_mini.name=LOLIN(WEMOS) D1 R2 & mini
 d1_mini.build.board=ESP8266_WEMOS_D1MINI
 d1_mini.build.variant=d1_mini
 d1_mini.upload.tool=esptool
+d1_mini.upload.maximum_size=1044464
 d1_mini.upload.maximum_data_size=81920
 d1_mini.upload.wait_for_upload_port=true
 d1_mini.upload.erase_cmd=
```
```
~\D\A\s\test> ~\arduino-cli\arduino-cli.exe compile -b esp8266com:esp8266:d1_mini
Executable segment sizes:
ICACHE : 32768           - flash instruction cache
IROM   : 232088          - code in flash         (default or ICACHE_FLASH_ATTR)
IRAM   : 26205   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...)
DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP
RODATA : 876   ) / 81920 - constants             (global, static) in RAM/HEAP
BSS    : 25520 )         - zeroed variables      (global, static) in RAM/HEAP
Sketch uses 260665 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27892 bytes (34%) of dynamic memory, leaving 54028 bytes for local variables. Maximum is 81920 bytes.
```
Still not sure why IDE somehow... finds this (apparently missing) value, while using the Core 3.0.2 version.

jie326513988, 2022-04-27:
> 所以，罪魁祸首是`upload.maximum_size`财产。我们没有为任何可用的板设置它，稍微调整一下，我终于能够看到尺寸信息。
> 
> ```diff
> @@ -3849,6 +3831,7 @@ d1_mini.name=LOLIN(WEMOS) D1 R2 & mini
>  d1_mini.build.board=ESP8266_WEMOS_D1MINI
>  d1_mini.build.variant=d1_mini
>  d1_mini.upload.tool=esptool
> + d1_mini.upload.maximum_size=1044464
>  d1_mini.upload.maximum_data_size=81920
>  d1_mini.upload.wait_for_upload_port=true
>  d1_mini.upload.erase_cmd=
> ```
> 
> ```
> ~\D\A\s\test> ~\arduino-cli\arduino-cli.exe compile -b esp8266com:esp8266:d1_mini
> Executable segment sizes:
> ICACHE : 32768           - flash instruction cache
> IROM   : 232088          - code in flash         (default or ICACHE_FLASH_ATTR)
> IRAM   : 26205   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...)
> DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP
> RODATA : 876   ) / 81920 - constants             (global, static) in RAM/HEAP
> BSS    : 25520 )         - zeroed variables      (global, static) in RAM/HEAP
> Sketch uses 260665 bytes (24%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 27892 bytes (34%) of dynamic memory, leaving 54028 bytes for local variables. Maximum is 81920 bytes.
> ```
> 
> 仍然不确定为什么 IDE 以某种方式......在使用 Core 3.0.2 版本时找到了这个（显然是缺失的）值。

I see. Thank you very much