esp8266/Arduino #8554: Unable to change STATION MAC address when using SOFT_AP

jam3st, 2022-05-01:
I have some old ESP8266 NOOS CODE that works but goes buggy after a few days. I want to port this to arduino as this uses 3.x of the SDK and I have an old 2.0 version of the NO OSSDK. In the old code I can randomize both the STATION and SOFT AP at the same time. But in this sketch only the first one is chnaged the station MAC doesn't change

```
#include <ESP8266WiFi.h>
#include <ESP8266TrueRandom.h>
/* Set these to your desired credentials for ESP8266 AP. */
const char *ssid = "SOFTAP";
const char *password = "01234567";

// Setup credentials for original WiFi, that we plan to repeat
const char* ssidExt = "ISPWIFI";

void setup() {
  uint8_t newMACAddress[6];
  Serial.begin(76800);

  
  ESP8266TrueRandom.mac(&newMACAddress[0]);
  Serial.printf("SOFTAP_IF %02x:%02x:%02x:%02x:%02x:%02x\r\n", newMACAddress[0],newMACAddress[1],newMACAddress[2],newMACAddress[3],newMACAddress[4], newMACAddress[5]);
  WiFi.softAPmacAddress( &newMACAddress[0]);

  ESP8266TrueRandom.mac(&newMACAddress[0]);
  Serial.printf("STATION_IF %02x:%02x:%02x:%02x:%02x:%02x\r\n", newMACAddress[0],newMACAddress[1],newMACAddress[2],newMACAddress[3],newMACAddress[4], newMACAddress[5]);
  wifi_set_macaddr(STATION_IF, &newMACAddress[0]);

  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssidExt, password);
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
  IPAddress apIP(192, 168, 4, 1);

  Serial.print("AP IP address: ");
  Serial.println(myIP);
}

bool connected = false;
void loop()
{
  bool newConnected =  (WiFi.status() == WL_CONNECTED);
  if(!connected && newConnected) {
    

      Serial.print("Connected to ");
      Serial.println(WiFi.macAddress());

      Serial.println(ssidExt);
      Serial.print("IP address: ");
      Serial.println(WiFi.localIP());
      Serial.print("dnsIP address: ");
      Serial.println(WiFi.dnsIP());
      Serial.print("gatewayIP address: ");
      Serial.println(WiFi.gatewayIP());
      Serial.print("subnetMask address: ");
      Serial.println(WiFi.subnetMask());
  } else if (connected && !newConnected) {
       Serial.print("Lost connection");
  }
  connected = newConnected;
  delay(5000);
}

```

d-a-v, 2022-05-02:
>I want to port this to arduino as this uses 3.x of the SDK and I have an old 2.0 version of the NO OSSDK

esp8266-Arduino core v3.x still uses esp8266 nonos-sdk firmware v2.x

d-a-v, 2022-05-02:
>   wifi_set_macaddr()

This firmware function *might* not change anything because mac addresses are stored elsewhere when using lwip2 compared to lwip1.4, which was an option until esp8266 arduino core 2.7.

d-a-v, 2022-06-18:
About AP, what I can say so far is that this:
> `WiFi.softAPmacAddress( &newMACAddress[0]);`

has never changed anything but overwrites `newMACAddress` with the AP address.
Using this instead currently works:
```
wifi_set_macaddr(SOFTAP_IF, &newMACAddress[0]);
```

mcspr, 2022-06-18:
Does it depend AP interface being 'UP'?
As always when mentioning 3.0.0 update, we have both interfaces down at boot. Code above sets MAC first, then brings the up the interface. NONOS SDK API reference example is to set mode to AP, then set MAC.

d-a-v, 2022-06-21:
You got it right @mcspr. 
STA interface needs to be up before changing its mac address:

```cpp
  WiFi.mode(WIFI_AP_STA);
  wifi_set_macaddr(SOFTAP_IF, newMACAddressAP);
  wifi_set_macaddr(STATION_IF, newMACAddressSTA);
 ...
  WiFi.begin(ssidExt, password);
  WiFi.softAP(ssid, password);
```
`wifi_set_macaddr` is mentioned nowhere in this core's source code nor in its documentation.
Indeed the FW API reference mentions a call to `wifi_set_opmode()` before changing mac addresses.

PR #8614 (2022-06-25): document side-effects for wifi-off-at-boot
fixes #8554
closed 2022-06-25 by d-a-v