esp8266/Arduino #8555: Ping Library fails in main branch,  works in 3.0.2 and 2.7.4

dduehren, 2022-05-03:
----------------------------- Delete below -----------------------------

### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:		NodeMCU 1.0 (ESP-12 Module)
- Core Version:			Github Desktop clone on 5/2/2022
- Development Env:		Arduino IDE 1.8.13
- Operating System:		Windows 10

### Settings in IDE

- Module:  			ESP-12
- Flash Mode:		Probably GPIO, not sure
- Flash Size: 		4MB (FS:2MB, OTA: 1019KB)
- lwip Variant:          v2 Lower Memory
- Reset Method:		Reset via UART and button on the device
- Flash Frequency:	Not sure, probably 
- CPU Frequency:		80 MHz
- Upload Using:		Serial for this case
- Upload Speed:        115200

### Problem Description

This example sketch for ping functionality works on 3.0.2 and 2.7.4 but does not work in Master Branch I downloaded yesterday (5/2/22)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```
/*
 * With this library an ESP8266 can ping a remote machine and know if it's reachable. 
 * It provides some basic measurements on ping messages (avg response time).
 */

#include <ESP8266WiFi.h>
#include <ESP8266Ping.h>

const char* ssid     = "ssid";
const char* password = "password";

const IPAddress remote_ip(192, 168, 0, 01;  //some suitable target

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println("Connecting to WiFi");
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  Serial.println();
  Serial.print("WiFi connected with ip ");  
  Serial.println(WiFi.localIP());

  Serial.print("Pinging ip ");
  Serial.println(remote_ip);

  if(Ping.ping(remote_ip)) {
    Serial.println("Success!!");
  } else {
    Serial.println("Error :(");
  }
}

void loop() { }

```

```
06:36:48.221 -> .......scandone
06:36:49.766 -> state: 0 -> 2 (b0)
06:36:49.766 -> .state: 2 -> 3 (0)
06:36:49.766 -> state: 3 -> 5 (10)
06:36:49.766 -> add 0
06:36:49.766 -> aid 8
06:36:49.766 -> cnt 
06:36:49.813 -> 
06:36:49.813 -> connected with TigerScout, channel 1
06:36:49.813 -> dhcp client start...
06:36:49.813 -> wifi evt: 0
06:36:49.859 -> ...ip:192.168.0.39,mask:255.255.255.0,gw:192.168.0.1
06:36:50.142 -> wifi evt: 3
06:36:50.189 -> .
06:36:50.189 -> WiFi connected with ip 192.168.0.39
06:36:50.189 -> Pinging ip 192.168.0.28
06:36:50.189 -> Error :(
06:36:55.305 -> ping 5, timeout 0, total payload 160 bytes, 5122 ms
06:36:59.764 -> pm open,type:2 0
06:38:52.241 -> ...........................scandone
06:38:55.753 -> state: 0 -> 2 (b0)
06:38:55.753 -> .state: 2 -> 3 (0)
06:38:55.753 -> state: 3 -> 5 (10)
06:38:55.753 -> add 0
06:38:55.753 -> aid 8
06:38:55.753 -> cnt 
06:38:55.801 -> 
06:38:55.801 -> connected with TigerScout, channel 1
06:38:55.801 -> dhcp client start...
06:38:55.801 -> wifi evt: 0
06:38:55.848 -> ...ip:192.168.0.39,mask:255.255.255.0,gw:192.168.0.1
06:38:56.081 -> wifi evt: 3
06:38:56.174 -> .
06:38:56.174 -> WiFi connected with ip 192.168.0.39
06:38:56.174 -> Pinging ip 192.168.0.35
06:38:56.174 -> Error :(
06:39:01.190 -> ping 5, timeout 0, total payload 160 bytes, 5022 ms
06:39:05.782 -> pm open,type:2 0
```

dduehren, 2022-05-05:
> I think this was only briefly discussed in gitter
> https://gitter.im/esp8266/Arduino?at=621609b3257a357825faa98a
>
> Library depends on the internal behaviour of esp_yeild(), which previously just suspended our USER task without doing anything else
> https://github.com/dancol90/ESP8266Ping/blob/master/src/ESP8266Ping.impl.h#L51
> However, since 3.0.0 it will also call esp_schedule(), so the logic it implements that waits on it's own esp_schedule() no longer works
>
> You should probably have notified the author first, since this is an external code and not something we maintain.
> I also wonder if we should just import the library here and clean up it a bit, so it uses our helpers that handle both situations. Meaning, when esp_schedule() is called, it will yield until ping either fails or succeeds.
> (similarly to what we have done with garp ping helper)

This makes me wonder what is your process for various types of “smoke testing” to be done to qualify a version for release?  Is the process described somewhere?

Thanks

David Duehren

dduehren, 2022-05-06:
Where is the documentation on esp_schedule and I will notify the author.

mcspr, 2022-05-06:
> Where is the documentation on esp_schedule

Not documented
(and idk how and where it would make sense to document it, since it is our own private api. and the code should be pretty self explanatory)

Effectively it is ets_post() / system_os_post(), see Espressif website and NONOS documentation for more info about that.

> This makes me wonder what is your process for various types of “smoke testing” to be done to qualify a version for release?  Is the process described somewhere?

CI here builds every example and also has a set of tests for our internal code. Tests are implemented for both 'device' and 'host' (done through mocking), but only 'host' is ran automatically. See `tests/` and our Actions setup.
In addition to CI, usually changes are verified using our 'device' tests and with various projects using this Core (espeasy, esphome, espurna, tasmota and etc.)

Feature wise, see https://github.com/esp8266/Arduino/milestone/24 for what was tagged so far for the next release.

Yo

dduehren, 2022-05-07:
What is the relationship between the NONOS and the Arduino development environment?  Can I access NON)S calls from C++?

And should yield() be used?  If not is there another way to achieve similar functionality?

Thanks,

David

dduehren, 2022-05-07:
The problem I'm really trying to solve is described here:  https://www.esp8266.com/viewtopic.php?f=29&t=23679  And I don't know if it's on the sending or receiving side, the connection fails, yet both sender and receiver succeed independently.  I can send http commands to the receiver from a browser, and I can send http GETs to my raspberry Pi running a python server. 
And as I show here,  https://github.com/esp8266/Arduino/issues/8513 where my crash/exception was due to my coding, it shows the example of success and then failure that I mention.
And I've been following this https://github.com/esp8266/Arduino/issues/5552 though we don't know what the GOPro5 is using, and as I mentioned, whether it's on the sending or receiving side.  Though it is the sending side code that changes between not working at all and working only the first time.

dduehren, 2022-05-07:
I'm about to open another issue. httpClient is causing an Exception 9

mcspr, 2022-05-08:
NONOS is SDK provided by Espressif, ESP* chip maker. Arduino side is built on top of it.
Just a matter of understanding where and when things are executed. Network, timers (os_timer, Ticker), network callbacks in externa Async* libs or the ping library code that calls schedule - happens in system context.

I meant something like
```cpp
while (ping_task_is_done) {
    yield(); // or, delay(100), so we jump back here every 100ms
}
```

dduehren, 2022-05-09:
The ping library works in 3.0.2 so something changed with the current code.

d-a-v, 2022-05-11:
I will try to fix it.
- ESP8266Ping library uses lwIP's `ping` app imported from lwIP's 1.4
- it is reported to be working with 3.0.2 but now anymore with master
- there were some changes in our internal yield/schedule api

dduehren, 2022-05-11:
I appreciate that.  My other issues were mainly my own code problems, and I'm looking into the environment where it still fails.

closed 2022-06-14 by mcspr

imayoda, 2023-04-16:
sorry to bother but it seems 3.1.2 is not pinging (dancol90) "again", while 3.0.2 works good ... I won't have "the machine" and the compiler under my hands for some days, but very reliable information was received about this problem.

PR #8596 (2022-06-14): fix lwip2 ping internal library
Fixes Ping internal library imported from lwIP-v1.4 to lwIP-v2

related: https://github.com/dancol90/ESP8266Ping/pull/42

fixes #8555 

cc @dduehren