esp8266/Arduino #8556: 100% reproductible: Exception in thread "Timer-0" java.lang.OutOfMemoryError: Java heap space -> crash

rin67630, 2022-05-06:
Problem: Arduino IDE throws a bench of errors and Raspberry completly crashes pon trying to compile a 

Exception in thread "Timer-0" java.lang.OutOfMemoryError: Java heap space ... 

**Hardware:** Raspberry Pi 3a+ 
**Software on Host:** Raspberry OS (Buster 32bit, freshly downloaded vanilla software:
     _Linux ***  5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux_
     VNC activated in preferences, headless. (Problem occurs with Monitor and Keyboard/Mouse as well)
**Software IDE:** Arduino IDE 1.8.19
**Board toolkit.** ESP8266 Community Version 3.0.2 
**Sketch loaded** as a common example : Arduino OTA from the examples, (Same behaviour form other sketches as well)

Memory used before loading Arduino IDE 127MB from 430MB
Memory used after loading Arduino IDE 263MB from 430MB
Memory used before crash 312MB from 430 MB

N.B. I have been using this configuration for years to remotely program my Wemos modules on dozens of identical Projects, even with Raspberry Pi zeros. It used to compile very slowly, but it did the job!

After an update/upgrade from Buster and upgrade to Boardtoolkit 3.0.2, it now crashes my Raspberries:

Already before attempting to compile the IDE Throws that bench of errors:
```
Exception in thread "Timer-0" java.lang.OutOfMemoryError: Java heap space
	at java.util.Arrays.copyOf(Arrays.java:3236)
	at java.io.ByteArrayOutputStream.grow(ByteArrayOutputStream.java:118)
	at java.io.ByteArrayOutputStream.ensureCapacity(ByteArrayOutputStream.java:93)
	at java.io.ByteArrayOutputStream.write(ByteArrayOutputStream.java:153)
	at org.apache.commons.compress.utils.IOUtils.copy(IOUtils.java:69)
	at org.apache.commons.compress.utils.IOUtils.copy(IOUtils.java:49)
	at org.apache.commons.compress.utils.IOUtils.toByteArray(IOUtils.java:167)
	at cc.arduino.contributions.GPGDetachedSignatureVerifier.verify(GPGDetachedSignatureVerifier.java:77)
	at cc.arduino.contributions.SignatureVerifier.isSigned(SignatureVerifier.java:55)
	at cc.arduino.contributions.DownloadableContributionsDownloader.checkSignature(DownloadableContributionsDownloader.java:211)
	at cc.arduino.contributions.libraries.LibraryInstaller.updateIndex(LibraryInstaller.java:89)
	at cc.arduino.contributions.ContributionsSelfCheck.updateLibrariesIndex(ContributionsSelfCheck.java:205)
	at cc.arduino.contributions.ContributionsSelfCheck.run(ContributionsSelfCheck.java:75)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
```
Hope I have submitted enough information, the problem is reproductible from vanilla installs using given examples.

Regards
Laszlo

earlephilhower, 2022-05-06:
Everything you've listed is related to the IDE (Java-based for 1.x) and not this core.  We have no control over anything java-related.  I suggest using https://forum.arduino.cc for more assistance.  There may be command-line options to increase the java heap space for the IDE.

closed 2022-05-06 by earlephilhower

rin67630, 2022-05-06:
I must however mention that no error is thrown upon compiling for Arduino modules. How do you explain that?

earlephilhower, 2022-05-06:
Java's reporting OOM and not any of the tools we ship so, again, there's not much we can do here.  We have 0 processes of our own running at idle.  The IDE just shells out and calls our `gcc` or `g++` to compile and once those exit we're completely out of the loop.

I really do suggest using the Arduino.cc forums as they do know the IDE in and out and would know of any command-line switches to suggest, or debugging options.