esp8266/Arduino #8557: HTTPClient::Get leads to exception 9

dduehren, 2022-05-07:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				Node MCU 1.0	
- Core Version:			Refreshed 5 days ago
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[ESP12-E]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz probably
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other])

### Problem Description

HTTPClient GET leads to exception 9

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
Not yet available.  This occurred in full application - fairly large.  Current MCVE has not shown this error.  For now I've included the corresponding function
```
bool OtaUpdate::RlyOnOff(bool OnOff, uint8_t relNo){
  //relNo is for future expansion, requires multiple swtichIPs
    Serial.println("RlyOnOFf1");
    int pass = 0;
    int httpCode = 0;
    httpSW.setReuse(true);
    sprintf_P(MTemp, "%s%s:%d%s" ,string_urlBase,switchIP,HTTPport,string_switchCmd);
    strcat(MTemp,"_");
    if(OnOff){
      strcat_P(MTemp, string_on);
    } else {
      strcat_P(MTemp,string_off);
    }
    Serial.println(MTemp);
    httpSW.begin(client, MTemp);
    httpCode = httpSW.GET();
       if ((httpCode > 0)){
            httpSW.writeToStream(&Serial);
            Serial.println(" ");
            Debug.print(DBG_INFO, F("HTTP %d"), httpCode);
            httpSW.end();
            return true;
          } else {
            Debug.print(DBG_ERROR, F("HTTP GET invalid response http code %d %s"),httpCode, httpSW.errorToString(httpCode).c_str());
            // something wrong with connection, try to reconnect
            httpSW.end();
            return false;
          }
    }
```

### Debug Messages

```
12:55:11.134 -> [hostByName] request IP for: us.pool.ntp.org
[... 95 lines cut ...]
0x40236a85: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x401014b8: umm_malloc(size_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 912
0x40236cb3: new_linkoutput at glue-lwip/lwip-git.c line 272
0x4023711e: ethernet_output at netif/ethernet.c line 312
0x4023ef6c: etharp_output_to_arp_index at core/ipv4/etharp.c line 769
0x4022937c: uart_write(uart_t*, char const*, size_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\uart.cpp line 547
0x4023f040: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x40101054: check_poison_block(umm_block*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_poison.c line 99
0x40240920: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40101279: umm_malloc_core(umm_heap_context_t*, size_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 47
0x40240952: ip4_output_if_src at core/ipv4/ip4.c line 1589
0x4023720b: ip_chksum_pseudo at core/inet_chksum.c line 392
0x4023b9fc: udp_sendto_if_src at core/udp.c line 893
0x40101054: check_poison_block(umm_block*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_poison.c line 99
0x401010e9: check_poison_neighbors(umm_heap_context_t*, uint16_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 72
0x40101138: umm_free_core(umm_heap_context_t*, void*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 642
0x40227747: __esp_suspend() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_features.h line 65
0x4022788a: __esp_delay(unsigned long) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 158
0x402285b4: __delay(unsigned long) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 37
0x40219abe: NTPClient::forceUpdate() at E:\Development\Arduino\libraries\NTPClient\NTPClient.cpp line 75
0x40219b62: NTPClient::update() at E:\Development\Arduino\libraries\NTPClient\NTPClient.cpp line 99
0x4020582f: sec30Update() at C:\Users\ddueh\AppData\Local\Temp\arduino_modified_sketch_629138/MushBox.ino line 249
0x402058e2: loop() at C:\Users\ddueh\AppData\Local\Temp\arduino_modified_sketch_629138/MushBox.ino line 157
0x4022794c: loop_wrapper() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 247
```

earlephilhower, 2022-05-07:
This is a null pointer deref (`EXECVADDR`):
> Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4010652c
EXCVADDR: 0x00000003

Most likely OOM (malloc() returning `nullptr`, then deref'ing a structure/object inside w/o checking).  Full debugging would be useful with a capture at the alloc failure.

dduehren, 2022-05-07:
Just prior to this, it had succeeded.  Same code.

```
12:53:56.141 -> Hum OnOff 0
12:53:56.141 -> Hum Relay test ota.SWconn 1
12:53:56.141 -> RlyOnOFf1
12:53:56.141 -> http://192.168.0.28:80/cm?cmnd=Power_off
12:53:56.141 -> [HTTP-Client][begin] url: http://192.168.0.28:80/cm?cmnd=Power_off
12:53:56.188 -> [HTTP-Client][begin] host: 192.168.0.28 port: 80 url: /cm?cmnd=Power_off
12:53:56.188 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
12:53:56.188 -> [hostByName] Host: 192.168.0.28 is a IP!
12:53:56.188 -> :ref 1
12:53:56.234 -> [HTTP-Client] connected to 192.168.0.28:80
12:53:56.234 -> [HTTP-Client] sending request header
12:53:56.234 -> -----
12:53:56.234 -> GET /cm?cmnd=Power_off HTTP/1.1
12:53:56.234 -> Host: 192.168.0.28
12:53:56.283 -> User-Agent: ESP8266HTTPClient
12:53:56.283 -> Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
12:53:56.283 -> Connection: keep-alive
12:53:56.283 -> Content-Length: 0
12:53:56.283 -> 
12:53:56.283 -> -----
12:53:56.283 -> :wr 180 0
12:53:56.283 -> :wrc 180 180 0
12:53:56.283 -> :ack 180
12:53:56.283 -> :rn 139
12:53:56.283 -> [HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
12:53:56.283 -> [HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 34
'
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain
'
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: none
'
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] RX: '
'
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] code: 200
12:53:56.331 -> [HTTP-Client][handleHeaderResponse] size: 34
12:53:56.331 -> Has param cmnd and value Power_off:c0 1, 139
12:53:56.363 -> [HTTP-Client][end] tcp stop
12:53:56.363 -> :close
12:53:56.363 ->  
12:53:56.363 -> HTTP 200
12:53:56.363 -> :ur 1
12:53:56.363 -> :dsrcv 0
12:53:56.363 -> :del
12:53:56.363 -> [HTTP-Client][end] tcp is closed
```

dduehren, 2022-05-07:
I refreshed the code and have the same results.  The MCVE sketch below 1) Does not produce the failure above, does not produce the success above either, it always fails, which is also the  httpClient problem I'm trying to figure out.  The target device is a 8266 running an ESP Async Web Server compiled with 2.7.4.  It also fails against a Tasmota  if the  "_" is changed to "%20".
```
/**
    RealBasicHttp.ino
    Created on: 24.05.2015
*/
#include <Arduino.h>
#include "Arduino_DebugUtils.h"
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266Ping.h>

const char string_BSQ[] PROGMEM = "\"";
const char string_Host[] PROGMEM ="Host: ";
const char string_GET[] PROGMEM ="GET ";
const char string_switchCmd[] PROGMEM = "/cm?cmnd=Power";
const char string_ContentLen[] PROGMEM ="Content-Length: ";
const char string_ContentTypeTxt[] PROGMEM ="Content-Type: text/plain";
const char string_HTTP1P1[] PROGMEM =" HTTP/1.1";
const char string_urlBase[] PROGMEM = "http://";
const char string_ipBase[] PROGMEM = "192.168.0.35";
const uint16_t serverPort = 5000;  //5000 for Rasp Pi, 80 for Ubuntu server
const char string_off[] PROGMEM = "off";
const char string_on[] PROGMEM = "On";

#define ipLen 17
#define HTTPport 80
#define MTempSize 160
char MTemp[MTempSize];  //General purpose buffer for many operations
char workBuffer[21]; //General purpose buffer when two are needed
const IPAddress test_ip(192,168,0,71);

bool RlyOnOff(bool OnOff, uint8_t relNo);

HTTPClient httpSW;
WiFiClient client;
char switchIP[ipLen];

int httpCode;
void setup() {
Serial.begin(115200);

  establishContact();
  Serial.println();

  strcpy(switchIP, "191.168.0.28");  //ip address of the wireless switch (28 for Myswitch, 71 for Tasmota)
//  strcpy(switchIP, "191.168.0.71");  //ip address of the wireless switch (28 for Myswitch, 71 for Tasmota)
  
  Serial.println("Connecting to wifi");
  wifi_set_phy_mode (PHY_MODE_11G);
  WiFi.mode(WIFI_STA);
  WiFi.begin("ssid", "password");    //Need to insert credentials here
  // wait for WiFi connection
  if(WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(500);
  }
  Serial.println("Wifi Connected");   

}
bool ONoff = true;
void loop() {
    Serial.println("Begin loop");
/*
    if(Ping.ping(test_ip)){
        Debug.print(DBG_INFO, F("ping succeeded"));
      } else {
        Debug.print(DBG_INFO, F("ping failed"));
[... 153 lines cut ...]
14:10:15.182 -> 
14:10:15.182 -> connected with TigerScout, channel 1
14:10:15.182 -> dhcp client start...
14:10:15.182 -> wifi evt: 0
14:10:15.557 -> ip:192.168.0.39,mask:255.255.255.0,gw:192.168.0.1
14:10:15.557 -> wifi evt: 3
14:10:21.955 -> Begin loop
14:10:21.955 -> RlyOnOff, Switch IP: 191.168.0.71
14:10:21.955 -> http://191.168.0.71:80/cm?cmnd=Power%20off
14:10:21.955 -> [HTTP-Client][begin] url: http://191.168.0.71:80/cm?cmnd=Power%20off
14:10:21.955 -> [HTTP-Client][begin] host: 191.168.0.71 port: 80 url: /cm?cmnd=Power%20off
14:10:21.955 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
14:10:21.955 -> [hostByName] Host: 191.168.0.71 is a IP!
14:10:21.955 -> :ref 1
14:10:25.153 -> pm open,type:2 0
14:10:26.969 -> :ctmo
14:10:26.969 -> :abort
14:10:26.969 -> :ur 1
14:10:26.969 -> :dsrcv 0
14:10:26.969 -> :del
14:10:26.969 -> [HTTP-Client] failed connect to 191.168.0.71:80
14:10:26.969 -> [HTTP-Client][returnError] error(-1): connection failed
14:10:26.969 -> HTTP GET invalid response http code -1 connection failed
14:10:27.023 -> [HTTP-Client][end] tcp is closed
```

dduehren, 2022-05-07:
Can anyone suggest a strategy for debugging this?  The connection fails except the first time in a complex app that I have not been able to reproduce in a simpler app.  Are there ways to check the wifi or tcp connection and/or environment variables to see what might be different between success and failure?  And also to figure out why the exception.

dduehren, 2022-05-08:
> This is a null pointer deref (`EXECVADDR`):
> 
> > Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
> > PC: 0x4010652c
> > EXCVADDR: 0x00000003
> 
> Most likely OOM (malloc() returning `nullptr`, then deref'ing a structure/object inside w/o checking). Full debugging (including the Serial debug log) would be useful with a capture at the alloc failure.

Good insight.  I forget the multi-tasking nature of this and think I know what may have happened.  If so, that will address the exception.  htttpClient still fails in the simple cases, but perhaps that should be another entry.  It's this latter issue that I really want to solve.   Will check shortly.  Exceptions get people's attention but may distract from the real problem.

How do I debug wifi connections, where it connects and succeeds in one case but fails to connect, in another using the same code, and simplified versions always fail?

dduehren, 2022-05-08:
Exception converted to Soft WDT time reset.  I thought the problem was that MTemp was getting clobbered because it's a global variable.  I created a variable local to the routine posted above., thinking that would resolve the issue.  Instead it has converted it to a time-out.  Here is the debug output followed by the exception decoded:  Debug level is one with SSL+TLS, ..+HTTP_CLIENT, ...+CORE+WIFI. +OTA+OOM+MDNS.  I have attached the full serial debug including the exception. The highlights are below.  wdt appears to come after __esp_delay
[MBTest0507gitSoftware.txt](https://github.com/esp8266/Arduino/files/8646495/MBTest0507gitSoftware.txt)

```
23:35:27.265 -> Hum OnOff 1
23:35:27.265 -> Hum Relay test ota.SWconn 1
23:35:27.265 -> RlyOnOFf1
23:35:27.265 -> http://192.168.0.28:80/cm?cmnd=Power_On
23:35:27.265 -> [HTTP-Client][begin] url: http://192.168.0.28:80/cm?cmnd=Power_On
23:35:27.265 -> [HTTP-Client][begin] host: 192.168.0.28 port: 80 url: /cm?cmnd=Power_On
23:35:27.265 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
23:35:27.265 -> [hostByName] Host: 192.168.0.28 is a IP!
23:35:27.265 -> :ref 1
23:35:29.605 -> 
```
```
Decoding stack results
0x40100609: millis() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 174
0x40227844: __esp_delay(unsigned long) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 152
0x40227897: esp_try_delay(unsigned int, unsigned int, unsigned int) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 171
0x40218f71: WiFiClient::connect(IPAddress, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/coredecls.h line 66
0x40101476: umm_free(void*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 688
0x40218e88: WiFiClient::connect(IPAddress, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp line 149
0x402184c8: WiFiClient::connect(char const*, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp line 137
0x4021f32a: HTTPClient::beginInternal(String const&, char const*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x4021e89f: HTTPClient::connect() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 348
0x4021f494: HTTPClient::sendRequest(char const*, unsigned char const*, unsigned int) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 459
0x401010e9: check_poison_neighbors(umm_heap_context_t*, uint16_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 72
0x40101138: umm_free_core(umm_heap_context_t*, void*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 642
0x40101494: umm_poison_free_fl(void*, char const*, int) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 157
0x4021f62e: HTTPClient::GET() at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 370
0x4020e023: OtaUpdate::RlyOnOff(bool, unsigned char) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_918350\sketch\otaUpdate.cpp line 77
0x4020b449: humOnOff(unsigned char) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_918350\sketch\boardconfig.cpp line 48
```
The above starts with the the previous stack push and corresponds to the debug output at 23:35:27,265 Hum OnOff 1
```
0x4020904c: std::_Function_handler   >::_M_invoke(const std::_Any_data &, AsyncWebServerRequest *&&) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_918350\sketch\WebUI.cpp line 210
0x402264f2: String::operator=(String&&) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\WString.cpp line 296
0x4021460c: AsyncWebServerRequest::_parsePlainPostChar(unsigned char) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x40216e48: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at e:\development\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 617
0x40226d06: String::startsWith(char const*) const at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x40214e48: AsyncWebServerRequest::_onData(void*, unsigned int) at E:\Development\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 173
0x40100609: millis() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 174
0x40214e9c: std::_Function_handler   >::_M_invoke(const std::_Any_data &, void *&&, AsyncClient *&&, void *&&, unsigned int &&) at E:\Development\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 76
0x40211c4c: AsyncClient::_recv(std::shared_ptr &, tcp_pcb*, pbuf*, long) at E:\Development\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.h line 103
0x401010e9: check_poison_neighbors(umm_heap_context_t*, uint16_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 72
[... cut ...]
0x4020df34: OtaUpdate::updateSW() at C:\Users\ddueh\AppData\Local\Temp\arduino_build_918350\sketch\otaUpdate.cpp line 36
0x4010023c: ets_post(uint8, ETSSignal, ETSParam) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 227
0x4010025d: esp_schedule() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 132
0x40227926: loop_wrapper() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 251

```

dduehren, 2022-05-09:
I started over with the BasicHttpClient code and got it to work by using "http://192.168.0.28/cm?cmnd=Power%20On" This led me to three main areas of difference between that sketch and the MVCE above.  1) Example uses WifiMulti, 2) I build the string with character strings from Progmem vs in place strings.  3) In the example, client and http are instantiated inside the loop - local variables.  I started with the strings and all of a sudden it worked.  The revised MVCE subroutine is below.  What it means is that somehow building the strings with strcat_P creates results that create problems inside httpConnect.   I don't know if this is a bug or not, or if there's something in my concatenation that is creating the problem. 

THE PROBLEM IS THAT THE IP ADDRESS OF THE REMOTE DEVICE HAS TO BE USER SUPPLIED, SO I NEED A WAY TO MAKE IT PROGRAMMABLE.  NEED A WAY TO CLEANUP THE BUILT-UP RESULTS.
```
bool RlyOnOff(bool OnOff, uint8_t relNo){
  //relNo is for future expansion, requires multiple swtichIPs
```
Previous version of code, building up the connect string with sprintf and strcat
```
/*
    int httpCode = 0;
    httpSW.setReuse(false);
    Debug.print(DBG_INFO, "RlyOnOff, Switch IP: %s", switchIP);
    sprintf_P(MTemp, "%s%s:%d%s" ,string_urlBase,switchIP,HTTPport,string_switchCmd);
//    sprintf_P(MTemp, "%s%s:%d%s" ,string_urlBase,srvIP,serverPort,string_switchCmd);
    strcat(MTemp,"%20");  //"%20" for Tasmota devices
//    strcat(MTemp,"%20"); 
    if(OnOff){
      strcat_P(MTemp, string_on);
    } else {
      strcat_P(MTemp,string_off);
    }
    Serial.println(MTemp);
*/
```
New version of code, using F() macro on the full string.
```
    if(OnOff){
    httpSW.begin(client,F("http://192.168.0.28/cm?cmnd=Power%20On"));
    } else {
    httpSW.begin(client,F("http://192.168.0.28/cm?cmnd=Power%20off"));
    }
    httpCode = httpSW.GET();
       if ((httpCode > 0)){
            String payload = httpSW.getString();
            Serial.println(payload);
            Debug.print(DBG_INFO, F("HTTP %d"), httpCode);
            httpSW.end();
            return true;
          } else {
            Debug.print(DBG_ERROR, F("HTTP GET invalid response http code %d %s"),httpCode, httpSW.errorToString(httpCode).c_str());
            // something wrong with connection
            httpSW.end();
            return false;
          }
 }
```

dduehren, 2022-05-09:
The complex app still has the soft WDT reset with a stack trace that looks very much the same as the one above.   The httpClient subroutine is the same one as above.  I'm using a local buffer mtemp rather than the global MTemp so I don't think it's getting clobbered, or if so, don't know how.

dduehren, 2022-05-09:
Even a fixed string leads to the soft WDT reset and same stack dump on the exception.  How do I figure out what's causing this? 
It doesn't happen in the simple sketches.  Here's the subroutine again, using fixed strings in the http.begin.  I may try this in an earlier version of the code, one maybe where Ping still works.
```
bool OtaUpdate::RlyOnOff(bool OnOff, uint8_t relNo){
  //relNo is for future expansion, requires multiple swtichIPs
    Serial.println("RlyOnOFf");

//    sprintf_P(mtemp, "%s%s:%d%s" ,string_urlBase,switchIP,HTTPport,string_switchCmd);  //If default port, don't need it.
//    strcat(mtemp,"_");
/*
    sprintf_P(mtemp, "%s%s%s" ,string_urlBase,switchIP,string_switchCmd);
    strcat(mtemp,"%20");  //"%20" for Tasmota devices
    if(OnOff){
      strcat_P(mtemp, string_on);
    } else {
      strcat_P(mtemp,string_off);
    }
    Serial.println(mtemp);
    httpSW.begin(client, mtemp);
*/
    if(OnOff){
    httpSW.begin(client, F("http://192.168.0.28/cm?cmnd=Power%20On"));
    } else {
    httpSW.begin(client, F("http://192.168.0.28/cm?cmnd=Power%20off"));
    }
    httpCode = httpSW.GET();
       if ((httpCode > 0)){
            httpSW.writeToStream(&Serial);
            Serial.println(" ");
            Debug.print(DBG_INFO, F("HTTP %d"), httpCode);
            httpSW.end();
            return true;
          } else {
            Debug.print(DBG_ERROR, F("HTTP GET invalid response http code %d %s"),httpCode, httpSW.errorToString(httpCode).c_str());
            // something wrong with connection, try to reconnect
            httpSW.end();
            return false;
          }
    }
```
Here's the serial dump
[MBTest0507gitSoftwareFixedIPstring.txt](https://github.com/esp8266/Arduino/files/8648624/MBTest0507gitSoftwareFixedIPstring.txt)
And the exception decoding leading up to the WDT
```
Decoding stack results
0x40100609: millis() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 174
0x40227800: __esp_delay(unsigned long) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 152
0x40227853: esp_try_delay(unsigned int, unsigned int, unsigned int) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 171
0x40218f2d: WiFiClient::connect(IPAddress, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/coredecls.h line 66
0x40101476: umm_free(void*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 688
0x40218e44: WiFiClient::connect(IPAddress, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp line 149
0x40218484: WiFiClient::connect(char const*, unsigned short) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp line 137
0x4021f2e6: HTTPClient::beginInternal(String const&, char const*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x4021e85b: HTTPClient::connect() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 348
0x4021f450: HTTPClient::sendRequest(char const*, unsigned char const*, unsigned int) at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 459
0x401010e9: check_poison_neighbors(umm_heap_context_t*, uint16_t) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 72
0x40101138: umm_free_core(umm_heap_context_t*, void*) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 642
0x40101494: umm_poison_free_fl(void*, char const*, int) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_local.c line 157
0x4021f5ea: HTTPClient::GET() at E:\Development\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 370
0x4020dfde: OtaUpdate::RlyOnOff(bool, unsigned char) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_185466\sketch\otaUpdate.cpp line 84
0x4020b449: humOnOff(unsigned char) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_185466\sketch\boardconfig.cpp line 48
```
Here's the call from the UI to the humOnOff function.
```
0x4020904c: std::_Function_handler   >::_M_invoke(const std::_Any_data &, AsyncWebServerRequest *&&) at C:\Users\ddueh\AppData\Local\Temp\arduino_build_185466\sketch\WebUI.cpp line 210
0x402264ae: String::operator=(String&&) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\WString.cpp line 296
0x402145c8: AsyncWebServerRequest::_parsePlainPostChar(unsigned char) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x40216e04: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at e:\development\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 617
0x40226cc2: String::startsWith(char const*) const at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 115
0x40214e04: AsyncWebServerRequest::_onData(void*, unsigned int) at E:\Development\Arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 173
[... 39 lines cut ...]
0x402341e8: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
[... cut ...]
0x4020df34: OtaUpdate::updateSW() at C:\Users\ddueh\AppData\Local\Temp\arduino_build_185466\sketch\otaUpdate.cpp line 36
0x4010023c: ets_post(uint8, ETSSignal, ETSParam) at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 227
0x4010025d: esp_schedule() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 132
0x402278e2: loop_wrapper() at E:\Development\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 251
```

dduehren, 2022-05-09:
In 3.0.2 in one path, the connection fails but does not have WDT problem.  In another path it succeeds, so maybe it's a memory problem??? Note sure what next step should be.  The ping library also works in 3.0.2.

php4fan, 2023-12-09:
Any progress on this??

I'm getting LoadStoreAlignmentCause triggered by a `Serial.printf`

Here's the decoded stack trace:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x402086d2:  is in Print::printf(char const*, ...) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.cpp:66).
EXCVADDR: 0xe0000003

Decoding stack results
0x4020277d: loadImage() at /home/teo/Documents/...***.ino:723
0x40203f00:  is in WiFiClient::WiFiClient(WiFiClient const&) (/home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/include/ClientContext.h:111).
0x40203438: ESP8266WiFiSTAClass::status() at /home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:558
0x402028fa: clientModeSetup() at /home/teo/Documents/...***.ino:577
0x40202961: setup() at /home/teo/Documents/...***.ino:151
0x4020a2fc: loop_wrapper() at /home/teo/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp:255
```

The line where the crash happens (as indicated by the stack trace) is:
```
Serial.printf("...((ticks = %d)) %d bytes downloaded; %d remaining\n", ticks, bytesdownloaded, len);
```
 `ticks`, `bytesdownloaded` and `len` are all int and I'm not doing anything weird with them, there's no way it's something I'm doing.

Even assuming there are bugs in my code, something must be very fucked up somewhere very deep for it to be affecting the variables involved in the printf.

I'm reporting here becaue this is also happening while doing a http download.

This is code that I had been working on for a long time and never observed anything like this. The only changes I made reently are that instead of downloading a stream of characters that were all letters of the alphabet, now I'm downloading a stream of "binary" bytes.
I'm pretty sure that has nothing to do with the nature of the error, and that it's something completely unrelated and outside of my code that I've now stumbled upon for completely random reasons.

php4fan, 2023-12-09:
Is there an easy way to check if I'm running out of memory?

TD-er, 2023-12-09:
`Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address`

Sounds more like the code is perhaps doing some kind of reinterpret_cast to a larger type from an address which isn't a multiple of the size of that object.
For example when accessing a 32-bit type, you need to work with addresses which are a multiple of 4.

php4fan, 2023-12-09:
> Sounds more like the code is perhaps doing some kind of reinterpret_cast 

Definitely not _my_ code.

php4fan, 2023-12-09:
If I run the exact same code only without the Serial.print()s, it hangs instead (probably around the same point, difficult to say without the debug output) and then triggers a WDT reset.

d-a-v, 2023-12-13:
@dduehren network code can also be sorted out through [the testing on host filter](https://github.com/esp8266/Arduino/blob/master/tests/host/README.txt). gdb and valgrind can help.
Running with all debug options enabled can also help.

mcspr, 2023-12-20:
Looking at logs, looks like it was WiFiClient being used inside of AsyncWebServer callbacks? Those happen in raw lwip, in sys context. WiFiClient would want to yield() at some point causing a panic.