esp8266/Arduino #8558: esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

devilution1979, 2022-05-08:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [v ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [v ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [v ] I have tested that the issue is present in current master branch (aka latest git).
- [v ] I have searched the issue tracker for a similar issue.
- [v ] If there is a stack dump, I have decoded it.
- [ v] I have filled out all fields below.

#### Platform

- Hardware:					|ESP-01|
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|
- Operating System:		[Windows 10|

### Settings in IDE

- Module:  			[Generic ESP8266 Module|
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.
have same issue, I used Windows 10, and the error logs as below:
` Arduino: 1.8.15 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, no dtr (aka ck), 26 MHz, 40MHz, DOUT (compatible), 4MB (FS:1MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

ICACHE : 32768 - flash instruction cache

IROM : 250788 - code in flash (default or ICACHE_FLASH_ATTR)

IRAM : 28257 / 32768 - code in IRAM (IRAM_ATTR, ISRs...)

DATA : 1556 ) - initialized variables (global, static) in RAM/HEAP

RODATA : 2428 ) / 81920 - constants (global, static) in RAM/HEAP

BSS : 25832 ) - zeroed variables (global, static) in RAM/HEAP

Sketch uses 283029 bytes (27%) of program storage space. Maximum is 1044464 bytes.

Global variables use 29816 bytes (36%) of dynamic memory, leaving 52104 bytes for local variables. Maximum is 81920 bytes.

esptool.py v3.0

Serial port COM9

Connecting......................................____Traceback (most recent call last):

File "C:\Users\anggolak\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/upload.py", line 66, in

esptool.main(cmdline)

File "C:/Users/anggolak/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 3552, in main

esp.connect(args.before, args.connect_attempts)

File "C:/Users/anggolak/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 529, in connect

raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

mcspr, 2022-05-08:
Make sure to switch into flashing mode first.

btw we are missing try: ... except FatalError: ..., python tracebacks are scary to look at
https://github.com/espressif/esptool/blob/4fa0bd7b0d1f69f5ff22b043adc07c5e562a8931/esptool.py#L3960-L3965
https://github.com/esp8266/Arduino/blob/bcb5464167ddf3b9d8035e81e2d5813d98c8461b/tools/upload.py#L66

closed 2022-05-08 by mcspr

d-a-v, 2022-05-12:
Some brands have an integrated flasher circuitry, some have a button, and some have nothing so you must wire it.
Check the documentation. https://arduino-esp8266.readthedocs.io/en/latest/boards.html

devilution1979, 2022-05-22:
`Make` sure to switch into flashing mode first.

Thank you for your reply

devilution1979, 2022-05-22:
I tried to flashing my ESP6266 using TTL USB and firmware is using ESP8266_NonOS_AT_Bin_V1.7.4,, 

but still failed (have same error message) when integrated with Arduino sketch, may I know what is the procedure for flashing mode ? I take a look at (https://www.espressif.com/en/support/download/at)home page that ESP8266 NonOS AT Bin already Deprecated.

Is there any solution for my problem ? thanks for your help

decoded-cipher, 2022-07-24:
The following procedure solved my issue:
- Close the Serial Monitor (if active)
- Disconnect the board from the circuit (At least the RX and TX pins)
- Then upload the code.

perlchamp, 2023-12-08:
Hallo at all,
i had the problems with the ESP8266 D1 mini.
I solved this problem as follow (no software or special drivers needed):
1. i bought [this one](https://www.amazon.de/DSD-TECH-SH-U09G-USB-zu-TTL-Kabel-Eingebauter/dp/B083HVM7VZ)
2. Connection to the board: vcc to 5V, gnd to GND, rxd to TX, txd to RX
3. Connected D3 (GPIO0) with GND (board) via jumper-wire (D1 sitting on a breadboard).
4. Open Arduino IDE, choose board 'LOLIN (WEMOS) D1 R2 & Mini', choose the right COM-Port
5. starting flashing... works !
6. Have fun ...

Good luck
So long
Pc

usersina, 2023-12-15:
It was working with Arduino IDE but not PlatformIO in my case, I simply added  the `upload_resetmethod = nodemcu` config [as specified here](https://community.platformio.org/t/esp01-code-upload-works-with-arduino-ide-but-not-with-pio/25893/3?u=usersina).