esp8266/Arduino #8562: Problem with download certs.

Mamol27, 2022-05-12:
Platform

    Hardware: ESP8266
    Core Version: idk
    Development Env: [Arduino IDE 1.8.19]
    Operating System: [Ubuntu 20.04]

Settings in IDE

    Module: [Generic ESP8266 Module]
    Flash Mode: [DIO]
    Flash Size: [4MB/1MB]
    lwip Variant: [v2 Lower Memory]
    Reset Method: [nodemcu]
    Flash Frequency: [80Mhz]
    CPU Frequency: [160MHz]
    Upload Using: [SERIAL]
    Upload Speed: [115200]

When i tried download certs i've got next messages:

```
python2 ./certs-from-mozilla.py 
AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org -> data/ca_000.der
unable to load certificate
140369867158848:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
AC Camerfirma, S.A.:AC Camerfirma S.A.: -> data/ca_000.der
unable to load certificate
140540298077504:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org -> data/ca_000.der
unable to load certificate
140321021482304:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
AC Camerfirma, S.A.:AC Camerfirma S.A.: -> data/ca_000.der
unable to load certificate
140144222418240:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Actalis:Actalis S.p.A./03358520967: -> data/ca_000.der
unable to load certificate
140221143942464:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Amazon Trust Services:Amazon: -> data/ca_000.der
unable to load certificate
140341341365568:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Amazon Trust Services:Amazon: -> data/ca_000.der
unable to load certificate
140430061745472:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Amazon Trust Services:Amazon: -> data/ca_000.der
unable to load certificate
140170498626880:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Amazon Trust Services:Amazon: -> data/ca_000.der
unable to load certificate
140578244822336:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Amazon Trust Services:Starfield Technologies, Inc.: -> data/ca_000.der
unable to load certificate
139692824077632:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Sp. z o.o.: -> data/ca_000.der
unable to load certificate
140680946144576:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Asseco Data Systems S.A. (previously Unizeto Certum):Asseco Data Systems S.A.:Certum Certification Authority -> data/ca_000.der
unable to load certificate
139699367388480:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority -> data/ca_000.der
unable to load certificate
139956524582208:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority -> data/ca_000.der
unable to load certificate
140007230719296:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Asseco Data Systems S.A. (previously Unizeto Certum):Asseco Data Systems S.A.:Certum Certification Authority -> data/ca_000.der
unable to load certificate
139751913936192:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Atos:Atos: -> data/ca_000.der
unable to load certificate
[... 404 lines cut ...]
unable to load certificate
140487318398272:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Taiwan-CA Inc. (TWCA):TAIWAN-CA:Root CA -> data/ca_000.der
unable to load certificate
140688351614272:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Taiwan-CA Inc. (TWCA):TAIWAN-CA:Root CA -> data/ca_000.der
unable to load certificate
139933706556736:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Telia Company:Telia Finland Oyj: -> data/ca_000.der
unable to load certificate
140218335561024:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
Telia Company:TeliaSonera: -> data/ca_000.der
unable to load certificate
140594203587904:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_000.der
unable to load certificate
140336781669696:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_000.der
unable to load certificate
139842223412544:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority -> data/ca_000.der
unable to load certificate
140194676061504:error:0909006C:PEM routines:get_name:no start line:../crypto/pem/pem_lib.c:745:Expecting: TRUSTED CERTIFICATE

```

Mamol27, 2022-05-12:
I said, I use ubuntu 20.04
```
 python3 ./certs-from-mozilla.py 
  File "./certs-from-mozilla.py", line 49
    print names[i] + " -> " + certName
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(names[i] + " -> " + certName)?
```
This is why i used python 2

earlephilhower, 2022-05-12:
I think you're using a very obsolete version of the core.  That line does not exist and the last change to the file was 2 years ago:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_CertStore/certs-from-mozilla.py

I suggest updating to 3.0.2 or the `git `master branch.

Mamol27, 2022-05-12:
Oh, I'm  sorry, there was link for old commit, i didn't notice it.
https://www.dzombak.com/blog/2021/10/HTTPS-Requests-with-a-Root-Certificate-Store-on-ESP8266-Arduino.html

earlephilhower, 2022-05-12:
No problem, good luck!  That's a nice write up.  Pity he picked some specific branch and not master.  Closing.