esp8266/Arduino #8580: class WiFiServer' has no member named 'accept'

MaxFun, 2022-05-24:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

[Arduino](https://github.com/esp8266/Arduino)/[doc](https://github.com/esp8266/Arduino/tree/master/doc)/[esp8266wifi](https://github.com/esp8266/Arduino/tree/master/doc/esp8266wifi)/server-examples.rst

Using Arduino IDE 1.8.16 and compiling with Sketch->Verify/Compile, the code produces an error "'class WiFiServer' has no member named 'accept'; "

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

const char* ssid = "********";
const char* password = "********";

WiFiServer server(80);

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  server.begin();
  Serial.printf("Web server started, open %s in a web browser\n", WiFi.localIP().toString().c_str());
}

// prepare a web page to be send to a client (web browser)
String prepareHtmlPage()
{
  String htmlPage;
  htmlPage.reserve(1024);               // prevent ram fragmentation
  htmlPage = F("HTTP/1.1 200 OK\r\n"
               "Content-Type: text/html\r\n"
               "Connection: close\r\n"  // the connection will be closed after completion of the response
               "Refresh: 5\r\n"         // refresh the page automatically every 5 sec
               "\r\n"
               "<!DOCTYPE HTML>"
               "<html>"
               "Analog input:  ");
  htmlPage += analogRead(A0);
  htmlPage += F("</html>"
                "\r\n");
  return htmlPage;
}

void loop()
{
  WiFiClient client = server.accept();
  // wait for a client (web browser) to connect
  if (client)
  {
    Serial.println("\n[Client connected]");
    while (client.connected())
    {
      // read line by line what the client (web browser) is requesting
      if (client.available())
      {
        String line = client.readStringUntil('\r');
        Serial.print(line);
        // wait for end of client's request, that is marked with an empty line
        if (line.length() == 1 && line[0] == '\n')
        {
          client.println(prepareHtmlPage());
          break;
        }
      }
    }

    while (client.available()) {
      // but first, let client finish its request
      // that's diplomatic compliance to protocols
      // (and otherwise some clients may complain, like curl)
      // (that is an example, prefer using a proper webserver library)
      client.read();
    }

    // close the connection:
    client.stop();
    Serial.println("[Client disconnected]");
  }
}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2022-05-24:
You didn't fill out the information requested in the template, so we don't know what version you're running, but I think you are not running the `git` master branch that you're using for the docs.  The `accept` was a recent addition and is most certainly in the core:
https://github.com/esp8266/Arduino/blob/b080c50713fad40a7c002292129a9f6cafde5926/libraries/ESP8266WiFi/src/WiFiServer.h#L84

Use `WiFiSerfver::available` if you're running an older version.

closed 2022-05-24 by earlephilhower

MaxFun, 2022-05-24:
I ran the from "https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/server-examples.rst", how do I get to "git master branch"?
I filled out what I knew or understood, I am not an expert and I read what I thought was applicable.
Thanks  for the response

earlephilhower, 2022-05-24:
If you are using an older release, just replace `.accept()` with `.available()` in the sketch and is should just work.  No need to use the git master.

The issue is you're using the documentation from the current, development version against your older, released version.  Use the online docs which have prior versions, not the git repo, for documents.  For example, if you look at the 3.0.2 version of the docs the server example uses `.available()`:
https://arduino-esp8266.readthedocs.io/en/3.0.2/esp8266wifi/server-examples.html

MaxFun, 2022-05-24:
Thanks
Is there somewhere that indicates which version of Arduino or later can run the code, this more of a general question of github.