esp8266/Arduino #8592: I think that it's time to release a new version. isn't it?

sblantipodi, 2022-06-02:
As title.
Is there any sense in not releasing a new version ?

d-a-v, 2022-06-22:
1. https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version
2. https://d-a-v.github.io/esp8266/Arduino/index.htm
3. there is no due date yet for the 3.1 release (but fwiw we're on it)

enjoyneering, 2022-07-29:
> 1. https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version
> 2. https://d-a-v.github.io/esp8266/Arduino/index.html
> 3. there is no due date yet for the 3.1 release (but fwiw we're on it)

Hi d-a-v can you update your snapshot to the latest dev?

d-a-v, 2022-07-30:
> Hi d-a-v can you update your snapshot to the latest dev?

done

closed 2024-01-18 by sblantipodi