esp8266/Arduino #8593: Slow WiFi connection with core v3.0.x w.r.t. v2.7.4

fabianoriccardi, 2022-06-04:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			9e2103f27e9eacba39d25697bacc5142a7aa5e66
- Development Env:	Arduino IDE

### Settings in IDE

- Module:  		Wemos D1 mini r2
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Flash Frequency:	40Mhz
- CPU Frequency:		80MHz
- Upload Using:		SERIAL

### Problem Description

The wifi connection in v3.x.x is slower than what observed in v2.7.4. With the old version (v2.7.4), the fastest connection time was about 150ms, but using v3.0.2 and the latest commit I get about 1sec.

What can I do to reach the previous connection time?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>

const char* ssid     = "ssid";
const char* password = "pwd";

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  uint8_t mac[6] = { 0x20, 0x00, 0x00, 0x00, 0x00, 0x00 };

   WiFi.config(
    IPAddress(192,168,1,100), 
    IPAddress(192,168,1,1),
    IPAddress(255,255,255,0),
    IPAddress(192,168,1,1),
    IPAddress(192,168,1,1));

  WiFi.begin(ssid, password, 6, mac);

  Serial.println();
  Serial.println();
  Serial.print("Wait for WiFi... ");
  unsigned long a = millis();

  while (WiFi.status() != WL_CONNECTED) {
    delay(1);
  }
  unsigned int b = millis();

  Serial.println(b-a);

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}
```

### Serial output

With v3.1.0-dev
```
fpm close 1 
mode : sta(dc:4f:00:00:00:00)
add if0

Wait for WiFi... scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt 

connected with SSID, channel 7
ip:192.168.1.100,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.100,mask:255.255.255.0,gw:192.168.1.1
1067

WiFi connected
IP address: 
192.168.1.100

```

with 2.7.4
```
scandone

Wait for WiFi... scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt 

connected with SSID, channel 7
ip:192.168.1.100,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.100,mask:255.255.255.0,gw:192.168.1.1
148

WiFi connected
IP address: 
192.168.1.100

```

mcspr, 2022-06-04:
3.0.0 putting WiFi to sleep on boot may be the reason?
https://github.com/esp8266/Arduino/blob/15e7d35d6e7c430915a5a86d854ae855d10a8da5/cores/esp8266/core_esp8266_main.cpp#L390-L394

b/c of Arduino IDE quirks, it is yet another weak-symbol-changing-link-order-magic to disable this behaviour
https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/libraries/ESP8266WiFi/src/enable_wifi_at_boot_time.cpp
https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/libraries/ESP8266WiFi/src/enable_wifi_at_boot_time.cpp#L9-L16

Also note of 'channel' and 'bssid' args to begin(), if AP stays on the same channel.

fabianoriccardi, 2022-06-05:
Thnkas for the suggestion, `enableWiFiAtBootTime` solves the issue. Moreover, I noted that calling WiFi.mode(WIFI_STA) (which doesn't change the actual mode since WIFI_STA is the default) breaks the fast connection.

Will it be possible to set `enableWiFiAtBootTime` at runtime? is it a limitation of the NONOS SDK?

d-a-v, 2022-06-05:
Here is the reason for [enableWiFiAtBootTime()](https://arduino-esp8266.readthedocs.io/en/3.0.2/esp8266wifi/generic-class.html?highlight=enableWiFiAtBootTime#persistent).
You can do this in a portable way:
```cpp
void setup ()
{
    #ifdef WIFI_IS_OFF_AT_BOOT
        enableWiFiAtBootTime(); // speed up wifi on boot
    #endif
    ...
}
```

fabianoriccardi, 2022-06-11:
Thanks for the link! I verified that, even if `enableWiFiAtBootTime()` is called, calling `WiFi.persistent(false);` prevents the fast connection.

closed 2022-09-12 by mcspr