esp8266/Arduino #8611: Error compiling for board NodeMCU 1.0 (ESP-12E Module)on one Windows-based machine but is able to compile on other Windows-based machines

ryuenma, 2022-06-20:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[3.0.2 / Latest Stable]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[NodeMCU 1.0]
- Flash Mode:			[Arduino IDE Default]
- Flash Size: 		[Arduino IDE Default]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[NodeMCU]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I am currently using a Windows 10 machine to code my sketches. I used the SimpleEspNowConnection library from [here](https://github.com/saghonfly/SimpleEspNowConnection) to get more utility out of the feature. I was currently coding for a versatile code that can be applied to either ESP32 or ESP8266. But when i compiled the code for ESP8266, the sketch compiler returned an error in the screenshot below;

[image]

[image]

first i thought that my code was wrong, and assuming that my code was mistyped i tried to compile the example sketch that was included in the library. but it turns out that the example sketch got the same error. I was confused, as last night i was able to compile it for the ESP32 one, but somehow it won't compile for the ESP8266. the library excplicitly stated that it supports both ESP32 and ESP8266. then out of curiosity, i had a Windows 11 machine with an Intel cpu. i tried compiling the code in that machine, and suddenly it compiled without a problem, as seen in the screenshot below;

[image]

I thought maybe my main machine was acting up, so i already did a restart multiple times, and re ran the comile side by side with the ESP32 one. the ESP32 still ran without a hitch, and is able to be flashed normally, meanwhile the ESP8266 one gave me a compiler error as seen in the screenshot below;

[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

This is the sketch example that is included within the library / documentation.

```#include "SimpleEspNowConnection.h"

SimpleEspNowConnection simpleEspConnection(SimpleEspNowRole::SERVER);

typedef struct struct_message {
  char type;
  char a[32];
  int b;
  float c;
  bool e;
} struct_message;

String inputString;
String clientAddress;

bool sendBigMessage()
{
  char bigMessage[] = "\n\
There was once a woman who was very, very cheerful, though she had little to make her so; for she was old, and poor, and lonely. She lived in a little bit of a cottage and earned a scant living by running errands for her neighbours, getting a bite here, a sup there, as reward for her services. So she made shift to get on, and always looked as spry and cheery as if she had not a want in the world.\n\
Now one summer evening, as she was trotting, full of smiles as ever, along the high road to her hovel, what should she see but a big black pot lying in the ditch!\n\
\"Goodness me!\" she cried, \"that would be just the very thing for me if I only had something to put in it! But I haven't! Now who could have left it in the ditch?\"\n\
And she looked about her expecting the owner would not be far off; but she could see nobody.\n\
[... 287 lines cut ...]
C:\Users\Kiki\Documents\Arduino\libraries\SimpleEspNowConnection\src\SimpleEspNowConnection.cpp:776:16: warning: array subscript 14 is above array bounds of 'char [13]' [-Warray-bounds]

  776 |  sendMessage[14] = 0;

      |  ~~~~~~~~~~~~~~^

C:\Users\Kiki\Documents\Arduino\libraries\SimpleEspNowConnection\src\SimpleEspNowConnection.cpp:765:7: note: while referencing 'sendMessage'

  765 |  char sendMessage[13];

      |       ^~~~~~~~~~~

cc1plus.exe: some warnings being treated as errors

Using library SimpleEspNowConnection at version 1.2.0 in folder: C:\Users\Kiki\Documents\Arduino\libraries\SimpleEspNowConnection 

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Kiki\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi 

Using library Ticker at version 1.0 in folder: C:\Users\Kiki\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\Ticker 

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

```

mcspr, 2022-06-20:
```cpp
...\src\SimpleEspNowConnection.cpp: In member function 'bool SimpleEspNowConnection::DeviceMessageBuffer::deleteBuffer(const uint8_t*, long int)':

...\src\SimpleEspNowConnection.cpp:195:1: error: no return statement in function returning non-void [-Werror=return-type]

  195 | }

      | ^

```
Please read `error:` and `warning:` lines, library code contains an error that will cause a crash at runtime when running Core 3.0.0+. `deleteBuffer()` **must** return a value

ESP32 has the same issue btw, it's just hidden due to warning not being fatal since the Core does not force it to become an error
https://github.com/espressif/arduino-esp32/pull/5321
https://github.com/espressif/arduino-esp32/issues/5867

Another issue is IDE warning settings were unintentionally disabling `-Werror=return-type`. Core 3.1.0 will have a fix for the different build results you are seeing. Just our misunderstanding of how GCC handles warning flags and `-w`
https://github.com/esp8266/Arduino/pull/8495

closed 2022-06-20 by mcspr

mcspr, 2022-06-21:
I should empathize that disabling warnings is not a workaround! Missing return value may cause a crash, or may not. It is simply undefined behaviour, anything could happen.

Fixing the library would be best.
I have noticed that you already opened an issue at https://github.com/saghonfly/SimpleEspNowConnection/issues/14
Brief search shows we could probably just change the return type from `bool` to `void` in both .cpp and .h, nothing in the library code checks what deleteBuffer returns

mcspr, 2022-09-22:
Make sure of two things
* that you have modified the correct `deleteBuffer` method. notice that there are two of those - one with `bool`, correctly returning `bool` and has `return true;` / `return false;`; and another one, where it also has `bool` but it does not return anything and should be `void`
* .h and .cpp should have matching function signatures. e.g.
```cpp
// .h
void deleteBuffer(const uint8_t *device, long id);

// .cpp
void SimpleEspNowConnection::DeviceMessageBuffer::deleteBuffer(const uint8_t *device, long id) {
```

ryuenma, 2022-09-22:
Thank you, i have fixed the problem and modified the library for now.