esp8266/Arduino #8617: file.size() produces wrong number

dduehren, 2022-06-25:
- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).  Cloned branch but couldn't get Arduino to recognize it so I couldn't select an8266 board.  Produced under 3.02 and 2.74
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ ]x I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12|
- Core Version:			[l3.02]
- Development Env:		[Arduino IDE
- Operating System:		[Windows

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB
- lwip Variant:   |v2 Lower Memory
- Reset Method:		[|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz
- Upload Using:		SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
While investigating code that worked before no longer worked _int error = httpSRV.writeToStream(&f);  //this used to work 
I was using file.size() to check if the file download was successful.  Eventually I determined that a) the code above indeed no longer works,  and b) file.size() also falsely reports 0 for the file size when in fact there is a valid file with a size much bigger than 0.

The sketch below reproduces the problem.  I have not tested this with the development branch because I had trouble getting that set up, even though I had it working earlier this year.  And this bug isn't crucial to my code working, but I want to raise it because all bugs are shallow given enough eyeballs.  It shows the problem in 2.74 and 3.02.
The code downloads files from a server running on Digital Ocean.  Right now there are two files to be downloaded, fan.html and adj.html.   The line:
```
Debug.print(DBG_ERROR, F("HTTP GET, File saved as %s, file size %d, http size %d"),workBuffer, f.size(), size);
```
has the term f.size() to report where f is the name of the LittleFS file.
The main loop uses the dir method to see what files have been downloaded and produce the correct result, which I verified by inspecting the contents of the file.
The debug is set to CORE.

Or am I doing something stupid?

### [MCVE]
```
/**
    RealBasicHttp.ino
    Created on: 24.05.2015
*/
#include <Arduino.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266HTTPClient.h>
#include <LittleFS.h>
#include "Arduino_DebugUtils.h"
#include <WiFiUdp.h>

#define BD_VER 6// set to board version being targetted
const char string_urlBase[] PROGMEM = "http://";
const char string_ipBase[] PROGMEM = "147.182.231.246";
const uint16_t serverPort = 80;  //5000 for Rasp Pi, 80 for Ubuntu server
//The next 4 lines are needed in exact order for binary to have matching strings for detection by ota upload server.
const char string_VERSION[] PROGMEM = "v9.0.19";  // version is BD_VER.SigFeature.build/bugfix
const char string_Ver[] PROGMEM = "&ver=";
const char string_DEVICE_PLATFORM[] PROGMEM = "ND_01"; //MAC is unique to a platform
const char string_Dev[] PROGMEM = "?dev=";
const char string_Mac[] PROGMEM = "&mac=";
#define MTempSize 160
#define ipLen 17
char workBuffer[21];
char MTemp[MTempSize];
[... 179 lines cut ...]
12:23:47.586 -> :wr 238 0
12:23:47.586 -> :wrc 238 238 0
12:23:47.633 -> :ack 238
12:23:47.633 -> :rn 536
12:23:47.633 -> :rch 536, 536
12:23:47.633 -> :rch 1072, 536
12:23:47.633 -> :rch 1608, 435
12:23:47.774 -> :c 1, 536, 2043
12:23:47.774 -> :c 1, 536, 1507
12:23:47.774 -> :c 1, 536, 971
12:23:47.774 -> :c0 1, 435
12:23:47.821 -> :rn 536
12:23:47.821 -> :c0 1, 536
12:23:47.821 -> :rn 536
12:23:47.821 -> :rch 536, 536
12:23:47.821 -> :rch 1072, 328
12:23:47.821 -> :c 1, 536, 1400
12:23:47.867 -> :c 1, 536, 864
12:23:47.867 -> :c0 1, 328
12:23:52.869 -> lfs_file_close: fd=0x3fff0688
12:23:52.869 -> HTTP GET, File saved as Adj.html, file size 0, http size 3683

12:23:52.869 -> Adj.html, size: 3683
12:23:52.869 -> Fan.html, size: 1046
```

earlephilhower, 2022-06-25:
You're closing the file before printing the size, but when you close the file the `File` is no longer valid for things like `.size`, etc.

````
....
                          f.close();
                          Debug.print(DBG_ERROR, F("HTTP GET, File saved as %s, file size %d, http size %d"),workBuffer, f.size(), size);
....
````
This hasn't changed for at least 3 years:
https://github.com/esp8266/Arduino/blob/8decdc380a13d8bdc26356dffcf91d0b90429665/libraries/LittleFS/src/LittleFS.h#L468-L470

Moving the printout before the `close` should get you running.

dduehren, 2022-06-25:
Thanks for printing that out but it’s hardly intuitive and using the Dir based file size isn’t particularly convenient for a quick check.

David Duehren

https://github.com/esp8266/Arduino/blob/8decdc380a13d8bdc26356dffcf91d0b90429665/libraries/LittleFS/src/LittleFS.h#L468-L470

dduehren, 2022-06-25:
I did that, thanks.

closed 2022-06-25 by dduehren