esp8266/Arduino #8618: Device hangs when client disconnects while file upload

PaSaSaP, 2022-06-25:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware: Wemos D1 Mini PRO (ESP8266)
- Core Version: 3.0.2 (official release),  Jul 27, 2021
- Development Env: Arduino IDE
- Operating System: MacOS

### Settings in IDE

- Module: LOLIN (WEMOS) D1 mini pro (esp8266)
- Flash Mode: -
- Flash Size: 16MB (FS: 14MB, OTA:~1019KB)
- lwip Variant: v2 Higher Bandwidth
- Reset Method: -
- Flash Frequency: -
- CPU Frequency: 80Mhz
- Upload Using: SERIAL
- Upload Speed: 3000000

### Problem Description

I have checked it on Wemos D1 mini pro and on ESP-01 (default setting + debug messages) and behavior is the same.

Listening socket is not closed when client uploads file to HTTP Server and while uploading client disconnects from WiFi. 

To reproduce it on PC with WiFi:
- Upload https://github.com/PaSaSaP/ESP8266_upload_file_issue sketch to device (change ssid/password, change/delete IP setting)
- in that repo there is script: request.py which uploads some data to server (change IP), execute: ```./request.py False```
- request.py prints: **Disconnect WiFi now...** so do it; after WiFi is disconnected stop request.py script or wait 15 seconds
- Server device has opened socked and hangs here: https://github.com/esp8266/Arduino/blob/3.0.2/libraries/ESP8266WebServer/src/Parsing-impl.h#L339
- Enable WiFi so internet connection is back to work!

Device need to be restarted manually to "fix" it by hand.
To fix it for now for myself I have introduced timeout (the simplest solution?): https://github.com/esp8266/Arduino/compare/master...PaSaSaP:uploadReadByte-timeout but I think that client should be disconnected in similar way just like when socket is closed on client side (which works).

I will try to reproduce it in latest git revision.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
Simple sketch available at: https://github.com/PaSaSaP/ESP8266_upload_file_issue

### Debug Messages
All serial output with enabled all debug messages available at: https://github.com/PaSaSaP/ESP8266_upload_file_issue/blob/master/serial-output.log

d-a-v, 2022-06-25:
It might be interesting to retest after #8607 is merged.

PaSaSaP, 2022-06-25:
Tested with latest git revision and still can be reproduced.
Tested with https://github.com/esp8266/Arduino/pull/8607 on 3.0.2 and latest git and problem still exists.

mcspr, 2022-06-26:
I think the issue here is with the abrupt disconnection of a client while upload is still in progress? If I remember values correctly, we end up with 'disconnected' state only after ~10-15 minutes.

Note that your proposal needs update timestamp in the loop as well, you might end up timing out when upload takes too long