esp8266/Arduino #8629: ESP8266 Example Sketch Not Working

AlperenKosem, 2022-07-06:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP - 07 
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Ubuntu 20.04

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dout
- Flash Size: 		        1 MB (144K)
- lwip Variant:                   v2 Lower Memory
- Reset Method:		dtrset
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I try to flash esp8266-07 module and there is no problem with the flashing. It says done uploading and not giving error but there is nothing happen in the esp and its functionalities after flashing. Arduino ide output after flashing is given in debug messages section. After flashing i cannot see any messages in the serial monitor and there is no blinking led.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// the setup function runs once when you press reset or power the board
void setup() {
    Serial.begin(115200);
    Serial.println("aa");

  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}
 
// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("HIGH");
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  Serial.println("LOW");
  delay(1000);                       // wait for a second
}
```

### Debug Messages

```
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to dtrset
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 267600 bytes from /tmp/arduino_build_348535/esp07.ino.bin to flash at 0x00000000
	erasing flash
	size: 041550 address: 000000
	first_sector_index: 0
	total_sector_count: 66
	head_sector_count: 16
	adjusted_sector_count: 50
	erase_size: 032000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 1000
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 30% ]
................................................................................ [ 61% ]
................................................................................ [ 91% ]
......................                                                           [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader

```

mcspr, 2022-07-06:
> esptool v0.4.13-1-gf80ae31

since 9790e1cb7cfa01ac74056fb835e6282af123aba0 we are using esptool.py, I don't think you are using latest version of the Core
further updates also included a different mechanism for flash erasing (see menu options). at least looking at other issues related to esp07, it might help cleaning up existing data

plus, opening up serial monitor on 74880 baud will show initial boot info

AlperenKosem, 2022-07-06:
Yes, i wasnt use esptool. After i used esp tool i get md5 error. And i write `esptool.py verify_flash --diff yes 0x0 esp07.ino.generic.bin ` command but i didnt understand the outputs.

Issue https://github.com/espressif/esptool/issues/273 says:
power issue... need to connect to USB vcc instead of 3.3v

```
esptool.py --baud 115200 --port /dev/ttyUSB0 write_flash --flash_mode dout --flash_size detect 0x0 esp07.ino.generic.bin 
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:55:19:12:8d:93
Stub is already running. No upload is necessary.
Configuring flash size...
Auto-detected Flash size: 1MB
Flash will be erased from 0x00000000 to 0x00041fff...
Compressed 267600 bytes to 195141...
Wrote 267600 bytes (195141 compressed) at 0x00000000 in 17.8 seconds (effective 120.0 kbit/s)...
File  md5: 9412f039b3640ac7f0278316cd27c1a5
Flash md5: 3adf43e75ba9012241b2dd0fbb3b5976
MD5 of 0xFF is 2c5e0f848f359ab1b4548800bb92c11b

A fatal error occurred: MD5 of file does not match data in flash!
```

```
   000412e9 00 2d
   000412ea 00 64
   000412eb 00 65
   000412ec 00 76
   000412ed 00 28
   000412ee 00 63
   000412ef 00 30
   000412f0 00 66
   000412f1 00 37
   000412f2 00 62
   000412f3 00 34
   000412f4 00 34
   000412f5 00 29
   000412f8 00 20
   000412f9 00 80
   000412fa 00 fe
   000412fb 00 3f
   00041300 ff 00
   00041301 ff 00
   00041302 ff 20
   00041303 ff 41
   00041304 ff 04
   00041305 ff 08
   00041306 ff 10
   00041307 ff 0c
   00041308 00 04
   00041309 00 08
   0004130a 00 10
   0004130b 00 0c
   0004130c 00 64
   00041310 00 14
   00041311 00 08
   00041312 00 40
   00041314 00 dd
   00041315 00 07
   00041317 00 50
   00041318 00 f2
   00041319 00 02
   0004131b 00 01
[... 436 lines cut ...]
   00041532 00 20
   00041533 00 40
   00041534 00 90
   00041535 00 2a
   00041537 00 40
   00041538 00 60
   00041539 00 87
   0004153a 00 fe
   0004153b 00 3f
   0004153c 00 d0
   0004153d 00 e8
   0004153e 00 fe
   0004153f 00 3f
   00041540 ff 00
   00041541 ff 00
   00041542 ff 00
   00041543 ff 00
   00041544 ff 00
   00041545 ff 00
   00041546 ff 00
   00041547 ff 00
   0004154f 00 17

A fatal error occurred: Verify failed.
```

mcspr, 2022-07-06:
As usual with ESP
- check the wires, make sure physical connection isn't the issue (e.g. there are broken / half-connected joints)
- make sure power supply is sufficient, ESP needs a stable 3v3. if power comes from serial adapter, try out a different one or just use external 3v3. just make sure everything has common GND

esptool.py diff meant to show contents difference... which does not really tell much, just that reading from flash after writing have different last 615 bytes; file total is 0x41550, diff shows that starting from offset 0x412e9 bytes are not the same

AlperenKosem, 2022-07-07:
i'm sure there is no problem with the wiring. 
I use external power source and i followed this wiring:
[image]
*(ref: https://www.mischianti.org/2021/05/08/esp12-esp07-esp8266-flash-pinout-specs-and-arduino-ide-configuration-6/)

#https://github.com/espressif/esptool/issues/291#issuecomment-377806241 in there, person says i solved problem by connecting 5 volt. 
I connected 5 volt and get this error: 
` A fatal error occurred: Failed to write compressed data to flash after seq 0 (result was C100: Unknown result)`

What should i do? am i continue with the 3.3V power supply and try to solve "md5 error" or connect to 5V and try to solve "Failed to write compressed data to flash after seq 0 error"?

mcspr, 2022-07-12:
Simplify wiring by losing io15 and en? 5v is not really solving anything, just changing the voltage to an unexpected level.
You are experiencing some physical / hw issue in both situations, at least what it looks like.