esp8266/Arduino #8638: WPA2 enterprise auth - is there support for TLSv1.2

aheinzel, 2022-07-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[a2982f96]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[26Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Using latest version from git (a2982f96) with nonos-sdk pre 3 ESP8266 can successfully connect to WPA2 enterprise using EAP-PEAP as long as the access point supports TLSv1.1. When, however, the access point requires TLSv1.2 it appears as ESP8266 is still trying to use TLSv1.1 and thus connection establishment fails during EAP phase. Is there support for TLSv1.2 for WPA2 enterprise authentication and how can it be enabled?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include  <ESP8266WiFi.h>

extern "C" {
#include "wpa2_enterprise.h"
}

char ssid[] = "WLAN1";
char username[] = "test1";
char identity[] = "test1";
char password[] = "secretpass1";

void setup() {
  Serial.begin(115200);
  Serial.printf("SDK version: %s\n", system_get_sdk_version());
  Serial.printf("free heap: %d\n", ESP.getFreeHeap());

  WiFi.mode(WIFI_STA);
  wifi_set_opmode(STATION_MODE);

  struct station_config wifi_config;
  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  strcpy((char*)wifi_config.password, password);

  wifi_station_set_config(&wifi_config);
 
  //clear enterprise
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_clear_enterprise_identity();
  wifi_station_clear_enterprise_username();
  wifi_station_clear_enterprise_password();
  wifi_station_clear_enterprise_new_password();

  //setup enterprise auth
[... 528 lines cut ...]
EAP: EAP entering state SELECT_ACTION
EAP: getDecision: method failed -> FAILURE
EAP: EAP entering state FAILURE
EAP: Building EAP-Failure (id=199)
wlan0: CTRL-EVENT-EAP-FAILURE ac:0b:fb:d8:03:46
IEEE 802.1X: ac:0b:fb:d8:03:46 BE_AUTH entering state FAIL
wlan0: STA ac:0b:fb:d8:03:46 IEEE 802.1X: Sending EAP Packet (identifier 199)
IEEE 802.1X: ac:0b:fb:d8:03:46 AUTH_PAE entering state HELD
nl80211: Set STA flags - ifname=wlan0 addr=ac:0b:fb:d8:03:46 total_flags=0x60 flags_or=0x0 flags_and=0xfffffffe authorized=0
wlan0: STA ac:0b:fb:d8:03:46 IEEE 802.1X: unauthorizing port
wlan0: STA ac:0b:fb:d8:03:46 IEEE 802.1X: authentication failed - EAP type: 0 (unknown)
wlan0: STA ac:0b:fb:d8:03:46 IEEE 802.1X: Supplicant used different EAP type: 25 (PEAP)
wlan0: IEEE 802.1X: Force disconnection of ac:0b:fb:d8:03:46 after EAP-Failure in 10 ms
IEEE 802.1X: ac:0b:fb:d8:03:46 BE_AUTH entering state IDLE
wlan0: IEEE 802.1X: Scheduled disconnection of ac:0b:fb:d8:03:46 after EAP-Failure
wlan0: ap_sta_disconnect STA ac:0b:fb:d8:03:46 reason=23
nl80211: sta_remove -> DEL_STATION wlan0 ac:0b:fb:d8:03:46 --> 0 (Success)
wlan0: STA ac:0b:fb:d8:03:46 WPA: event 3 notification
wpa_driver_nl80211_set_key: ifindex=3 (wlan0) alg=0 addr=0x556c2aff10 key_idx=0 set_tx=1 seq_len=0 key_len=0
   addr=ac:0b:fb:d8:03:46
nl80211: set_key failed; err=-22 Invalid argument)
RSN: PTK removal from the driver failed
WPA: ac:0b:fb:d8:03:46 WPA_PTK entering state DISCONNECTED
```
</details>

mcspr, 2022-07-31:
We depend on the SDK implementation of enterprise connection, so only if it supports then we could support it. You could check standalone SDK examples, though. (obviously, both are without Arduino / Wiring api or libs from our Core)

  * NONOS - https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise
  * RTOS - https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/wifi/wpa2_enterprise

mcspr, 2022-09-12:
Depends on #5790

closed 2022-09-12 by mcspr