esp8266/Arduino #8639: UpdaterClass error [10] Magic byte is wrong most of the time

BrotherV, 2022-07-16:
### Update firmware returns error [10] Magic byte is wrong, not 0xE9 while the first byte of bin file is 0xE9

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in the current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[3.0.2]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I am trying to download the ESP firmware bin file from AWS, however, since I could not pass the HTTPclient object to the ESPhttpUpdateobject, I tried to create a method to download firmware and upload on the ESP chip.

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266HTTPClient.h>
#include "AWSClient4.h"
#include "Utils.h"

void executeOTAFirmware()
{
DateTimeProvider *dateTimeProvider = new DateTimeProvider;
  char *dateTime = dateTimeProvider->getDateTime();
  Serial.printf("Date and Time: %s\r\n", dateTime);
  
  AWSClient4 *awsClient4 = new AWSClient4;
  awsClient4->setQuery("");
  awsClient4->setMethod("GET");
  awsClient4->setSignedHeaders("host;x-amz-content-sha256;x-amz-date");
  awsClient4->setAWSRegion(awsRegion);
  awsClient4->setAWSEndpoint(awsEndpoint);
  awsClient4->setAWSDomain(awsDomain);
  awsClient4->setAWSService("s3");
  awsClient4->setAWSKeyID(awsAccessKey);
  awsClient4->setAWSSecretKey(awsSecretKey);
  awsClient4->setDateTime(dateTime);
  awsClient4->setAWSPath(awsPath_WiFi);
  awsClient4->createSignature("");

  HTTPClient *httpClient = new HTTPClient;
  WiFiClientSecure *client = new WiFiClientSecure;
  if (client)
  {
    client->setInsecure();
    Log("\n\t[HTTPS] begin...");
    httpClient->begin(*client, url_WiFi);
    // httpClient.begin(client, awsDomain, 80, awsPath, true);
    httpClient->addHeader("x-amz-content-sha256", awsClient4->getPayloadHash());
[... 251 lines cut ...]
    90 E1 00 40 A8 DE 00 40 E8 2B 00 40 00 4A 00 40 
    4C 4A 00 40 BF 20 FF 3F C9 20 FF 3F C3 20 FF 3F 
    C7 20 FF 3F 00 07 00 60 F0 30 00 40 A0 2F 00 40 
    B7 1D C1 04 00 12 00 60 00 EE FF 9F 80 12 00 60 
    00 10 00 EB 7C 12 00 60 88 DC 00 40 CD 20 FF 3F 
    04 20 FF 3F 7C 20 FF 3F 40 20 FF 3F 9A 20 FF 3F 
    FF FF 00 00 12 C1 F0 31 DA FF C9 21 D9 11 09 31 
    D1 D9 FF CD 02 39 02 21 D7 FF 0C 14 29 1C 28 0D 
    40 44 11 01 D5 FF C0 00 00 28 0D 08 31 22 D2 10 
    29 0D 28 0C D8 11 1B 32 39 0C 22 02 00 C8 21 12 
    C1 10 0D F0 52 A1 01 12 C1 E0 C0 55 11 0C 44 3D 
    01 5A 22 C9 61 09 71 01 C8 FF C0 00 00 CD 02 CC 
    D2 38 01 21 C6 FF 01 C6 FF C0 00 00 86 00 00 00 
    0C 1C 08 71 2D 0C C8 61 12 C1 20 0D F0 00 00 00 
    12 C1 E0 0C 84 D9 51 4A 31 DD 02 22 D2 10 09 71 
    C2 61 06 E2 61 04 01 B8 FF C0 00 00 56 22 07 C2 
    A2 01 D0 CC B0 ED 02 DD 02 C6 14 00 0C 84 3D 01 
    2D 0C D9 01 D9 11 01 B0 FF C0 00 00 FC 42 38 01 
    22 AB FF C0 22 01 51 AF FF 2A 23 48 11 8B CC 27 
    B5 0F 7C F2 20 22 01 51 AC FF 2A 23 27 35 02 C6 
```

Based on the debug message, the first byte of the file is 0xE9. I have tested "UpdaterClass" 10 times and it could only update ESP firmware just two times.

The question is, what is wrong with that? I checked it several times.

BrotherV, 2022-07-16:
I added a method in "ESP8266HTTPUpdate" to pass down only "HTTPClient" object instead of "WifiClient" and "URL". In this way, I can add some specific headers to the "HTTPClient". I do not know why the above sketch does not work all the time, however, the modified "ESP8266HTTPUpdate" class can update the sketch with no problem.

**ESP8266httpUpdate.h file**
```
HTTPUpdateResult ESP8266HTTPUpdate::update(HTTPClient &http, const String &currentVersion)
{
    return handleUpdate(http, currentVersion, false);
}
```

**ESP8266httpUpdate.cpp** file, added in public section.**
```
t_httpUpdate_return update(HTTPClient &http, const String &currentVersion = "");
```

**CPP file**
```
WiFiClientSecure *client = new WiFiClientSecure;
  if (client)
  {
    client->setInsecure();
    Log("\n\t[HTTPS] begin...");

    HTTPClient *httpClient = new HTTPClient;
    httpClient->begin(*client, url_WiFi);
    httpClient->addHeader(F("x-amz-content-sha256"), awsClient4->getPayloadHash());
    httpClient->addHeader(F("x-amz-date"), awsClient4->getDateTime());
    httpClient->addHeader(F("Authorization"), awsClient4->getAuthorization());

    ESP8266HTTPUpdate *otaUpdater = new ESP8266HTTPUpdate;
    otaUpdater->setLedPin(LED_BUILTIN, LOW);

    // Add optional callback notifiers
    otaUpdater->onStart(update_started);
    otaUpdater->onEnd(update_finished);
    otaUpdater->onProgress(update_progress);
    otaUpdater->onError(update_error);

    otaUpdater->rebootOnUpdate(false); // remove automatic update

    Serial.println(F("Update start now!"));

    t_httpUpdate_return ret = otaUpdater->update(*httpClient);
    // Or:
    // t_httpUpdate_return ret = ESPhttpUpdate.update(client, "bvtech.parsaspace.com", 443, "/firmware.bin");

    switch (ret)
    {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      Serial.println("HTTP_UPDATE_OK");
      delay(1000); // Wait a second and restart
      ESP.restart();
      break;
    }

    delete otaUpdater;

    client->flush();
    client->stop();

    httpClient->end();
  }
[... 163 lines cut ...]
CALLBACK:  HTTP update process at 475136 of 535920 bytes...
CALLBACK:  HTTP update process at 479232 of 535920 bytes...
CALLBACK:  HTTP update process at 483328 of 535920 bytes...
CALLBACK:  HTTP update process at 487424 of 535920 bytes...
CALLBACK:  HTTP update process at 491520 of 535920 bytes...
CALLBACK:  HTTP update process at 495616 of 535920 bytes...
CALLBACK:  HTTP update process at 499712 of 535920 bytes...
CALLBACK:  HTTP update process at 503808 of 535920 bytes...
CALLBACK:  HTTP update process at 507904 of 535920 bytes...
CALLBACK:  HTTP update process at 512000 of 535920 bytes...
CALLBACK:  HTTP update process at 516096 of 535920 bytes...
CALLBACK:  HTTP update process at 520192 of 535920 bytes...
CALLBACK:  HTTP update process at 524288 of 535920 bytes...
CALLBACK:  HTTP update process at 528384 of 535920 bytes...
CALLBACK:  HTTP update process at 532480 of 535920 bytes...
CALLBACK:  HTTP update process at 535920 of 535920 bytes...
CALLBACK:  HTTP update process at 535920 of 535920 bytes...
CALLBACK:  HTTP update process at 535920 of 535920 bytes...
Staged: address:0x0027D000, size:0x00082D70
[HTTP-Client][end] tcp is closed
CALLBACK:  HTTP update process finished
[HTTP-Client][end] tcp is closed
HTTP_UPDATE_OK

```

vortigont, 2022-08-08:
This seems to be wrong here

> ```c++
>         if (canBegin)
>         {
>           size_t written = update->writeStream(*client);
> ```

you should work on `httpClient` object, not on `client` secured stream.
Try it like this

```c++
    WiFiClient *stream = httpClient.getStreamPtr();
    if (!stream){
      // do some error processing
          return;
    }

   size_t written = update->writeStream(*stream);
```

closed 2022-09-12 by mcspr