esp8266/Arduino #8646: Empty reply from server / no data found for resource with given identifier

mofada, 2022-07-21:
#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

##### expected
I wrote a piece of code to connect to WiFi via http and when I enter the correct ssid and password everything works as expected.
```json
{"message":"connect success!","code":0,"data":{"ssid":"fada","password":"fada-8888","rssi":-62,"localIP":192.168.0.103,"deviceId":6303019}}
```

#### fail
When I pass in a non-existing ssid, or a wrong password
chrome `no data found for resource with given identifier`
```
C:\Users\fada>curl http://192.168.4.1/wifi?ssid=fada1
curl: (52) Empty reply from server`
```
My expected return:
`{"message":"wifi password wrong!","code":203}`

### code
```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer esp8266_server(80);

void setup()
{
  Serial.begin(9600);

  WiFi.mode(WIFI_STA);
  WiFi.softAP("khc-" + String(ESP.getChipId()), "88888888");

  // start server
  esp8266_server.begin();
  esp8266_server.keepAlive(true);
  esp8266_server.on("/wifi", HTTP_GET, handlePutWifi);

  WiFi.printDiag(Serial);
  Serial.setDebugOutput(true);
}

void loop()
{
  esp8266_server.handleClient();
}

/**
   @brief connect wifi

*/
void handlePutWifi()
{
  if (!esp8266_server.hasArg("ssid"))
  {
    // check params
    esp8266_server.send(200, "application/json;charset=utf-8", "{\"message\":\"message is required\",\"code\":-1}");
    return;
  }

  // get params
  String ssid = esp8266_server.arg("ssid");
  String password = esp8266_server.arg("password");
  // set not connect
  WiFi.setAutoReconnect(false);
  // connect wifi
  WiFi.begin(ssid.c_str(), password.c_str());
  // timeout  15s
  int8_t status = WiFi.waitForConnectResult(15 * 1000L);

  if (status == WL_CONNECTED)
  {
    // success return ip/deviceId
    esp8266_server.send(200, "application/json;charset=utf-8", "{\"message\":\"connect success!\",\"code\":0,\"data\":{\"ssid\":\"" + WiFi.SSID() + "\",\"password\":\"" + WiFi.psk() + "\",\"rssi\":" + WiFi.RSSI() + ",\"localIP\":" + WiFi.localIP().toString() + ",\"deviceId\":" + ESP.getChipId() + "}}");
  }
  else if (status == WL_NO_SSID_AVAIL)
  {
    esp8266_server.send(200, "application/json;charset=utf-8", "{\"message\":\"wifi ssid not avail!\",\"code\":202}");
  }
  else if (status == WL_WRONG_PASSWORD)
  {
    esp8266_server.send(200, "application/json;charset=utf-8", "{\"message\":\"wifi password wrong!\",\"code\":203}");
  }
  else
  {
    esp8266_server.send(200, "application/json;charset=utf-8", "{\"message\":\"connect fail!\",\"code\":204}");
  }
}

```

### Debug Messages

```
station: 08:1f:71:ac join, AID = 1
scandone
scandone
switch to channel 6
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
state: 5 -> 0 (2)
rm 0
```

enjoyneering, 2022-07-21:
Looks like the ["waitForConnectResult()"](https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L429) returns **-1** insted of actual status.

mofada, 2022-07-21:
> Looks like the ["waitForConnectResult()"](https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L429) returns **-1** insted of actual status. By the way default timeout is 60sec.

If -1 is returned, the last else will be hit, and then return
`{"message":"connect fail!","code":204}`

enjoyneering, 2022-07-21:
i

> If -1 is returned, the last else will be hit, and then return
> `{"message":"connect fail!","code":204}`

no, because esp is not connected to netwok :)

mofada, 2022-07-21:
Ah? Then it should also hit the `send`，and will the station mode be turned off?

[image]

mofada, 2022-07-22:
If the password is wrong, 12s ends the request and no message is returned
[image]

[image]

mofada, 2022-07-30:
@enjoyneering  can anyone help me？

mofada, 2022-07-30:
[image]

mcspr, 2022-07-30:
> The client appears to be lost halfway through the connection

https://github.com/esp8266/Arduino/blob/cf6ff4c4f834f7b288c19b3fadbf5aedc0e8e08f/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L228

try removing this line in your Core installation folder, usually at `%LOCALAPPDATA%\Arduino15\packages\esp8266\hardware\...`

mofada, 2022-07-31:
> https://github.com/esp8266/Arduino/blob/cf6ff4c4f834f7b288c19b3fadbf5aedc0e8e08f/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L228
>
> try removing this line in your Core installation folder, [...]

Solved my problem, which has been bothering me for a long time, thanks a lot

mofada, 2022-07-31:
> https://github.com/esp8266/Arduino/blob/cf6ff4c4f834f7b288c19b3fadbf5aedc0e8e08f/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L228
>
> try removing this line in your Core installation folder, [...]

My problem was solved by modifying the source code, is there any other solution without modifying the source code? Such a solution does not seem to be the optimal solution

mcspr, 2022-07-31:
> My problem was solved by modifying the source code, is there any other solution without modifying the source code? Such a solution does not seem to be the optimal solution

Next release will no longer include the code in question.

closed 2022-07-31 by mcspr