esp8266/Arduino #8650: WiFi.softAPConfig() always returns false

enjoyneering, 2022-07-29:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[git:'3.0.2-118-gdc6b15e3' (Wed Jun 22 21:37:42 UTC 2022)]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]
- Espressif FW: [nonos-sdk 2.2.1+61 (190313)]

### Problem Description

Hi, I'm using [d-a-v](https://github.com/d-a-v) snapshot - [version 0.0.1: git:'3.0.2-118-gdc6b15e3' (Wed Jun 22 21:37:42 UTC 2022)](https://github.com/esp8266/Arduino/issues/8592). Don't think git has changed a lot since then. The problem is that WiFi.softAPConfig() always return false. Didn't have this issue with v3.0.2.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

#define AP_IP         IPAddress(192, 168, 4, 1)
#define AP_GATEWAY    IPAddress(0, 0, 0, 0) 
#define AP_SUBNET     IPAddress(255, 255, 255, 0)

#define AP_HIDDEN     false
#define AP_SSID       "mySSID"
#define AP_PASSWORD   "12345678"
#define AP_CHANNEL    11
#define AP_MAX_USERS  3

void setup()
{
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
  Serial.println();

  if (WiFi.softAPConfig(AP_IP, AP_GATEWAY, AP_SUBNET) != true)
  {
    Serial.println(F("Oops! failed to assign softAP settings"));
  }

  if (WiFi.softAP(AP_SSID, AP_PASSWORD, AP_CHANNEL, AP_HIDDEN, AP_MAX_USERS) != true)
  {
    Serial.println(F("Oops! failed to start softAP"));
  }
}

void loop()
{
  Serial.println(F("loop"));

  delay(1000);
}

```

mcspr, 2022-07-29:
> Don't think git has changed a lot since then. 

Re-try with git installation, updating to the current master?
DHCPs code *was* changed, as well as the related AP code. I am not seeing error `println`s with the example above. Enabling WiFi debug in the menu would also show some kind of error message on the serial port, there are a bunch of `DEBUG_WIFI(...)` in the softAP code

> #define AP_GATEWAY    IPAddress(0, 0, 0, 0) 

Maybe unrelated... but what exactly happens here, when we call `wifi_set_ip_info` with this gw? Keeps the default?

enjoyneering, 2022-07-29:
Well, I kind of avoid installing the git version as it requires a lot of extra software on the PC and manual work. I tried once and messed up arduino IDE. Looks like I'm bad at console. That's why I prefer the [d-a-v](https://github.com/d-a-v) snapshot, but for some reason he rarely update it. I really like what {arduino-esp32 did](https://github.com/espressif/arduino-esp32) does - they have stable & development release for download.

I've also tryed `#define AP_GATEWAY IPAddress(8, 8, 8, 8)` - same story. Since git has changed a lot since Wednesday June 22nd, I'll ask d-a-v to update the snapshot and then post the result here.

enjoyneering, 2022-07-30:
I installed the current git and the problem went away. Thank you [mcspr](https://github.com/mcspr) for help.

closed 2022-07-30 by enjoyneering