esp8266/Arduino #8659: Signing broke by #8507 and arduino-builder update

earlephilhower, 2022-08-21:
See discussion in https://github.com/earlephilhower/arduino-pico/issues/783

`arduino-builder` no longer includes `{build.path}` in the includes, so the generated `Updater_Signing.h` is never used.

#8507 adjusts  `_size` here but it should be `binSize`:
https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/cores/esp8266/Updater.cpp#L245-L248

That means the signature will be calculated over the entire blob, including the signature itself, causing failure.

See the prior line:
https://github.com/esp8266/Arduino/blob/520233f73e997e2a3899271fbf670fedbdeb6718/cores/esp8266/Updater.cpp#L238

mcspr, 2022-08-21:
https://github.com/esp8266/Arduino/pull/8545 ?

earlephilhower, 2022-08-21:
Lots of stuff happening in that PR other than the incorrect signature calc.  I'll need some time to grok exactly what's going on in it.

FWIW undoing the 1-liner change above seems to be enough to unbreak checks in the common case.

The platform.txt changes are also needed or the generated signing file will be ignored, it seems (with current Arduino cores....earlier revs worked but operated differently.)

mcspr, 2022-08-21:
No rush, PR stayed there a long time anyway :>
The idea is to also update the structure on updater touching on these vars; one line change fixes specific stuff, but not the underlying issue of type handling.

(also I don't really remember what I did there, but I do remember that I definitely tested it)

mcspr, 2022-08-21:
> The platform.txt changes are also needed or the generated signing file will be ignored, it seems (with current Arduino cores....earlier revs worked but operated differently.)

I think it is already handled? Unless I misunderstood the issue with the builder / arduino-cli in IDE. 
https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/platform.txt#L73
https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/platform.txt#L134
BTW Is the issue with the orig java IDE or the new one? (or, if that matters at all)

earlephilhower, 2022-08-21:
Yeah, that's the same thing we ended up on.  So, no worries here then once #8545 gets in!  Will close this, too.

closed 2022-08-21 by earlephilhower