esp8266/Arduino #8660: OTA text error messages lost due to multiple USB packets

earlephilhower, 2022-08-21:
Ref:  https://github.com/earlephilhower/arduino-pico/pull/787

Basically, `OTA::printError` is calling `printf()` multiple times, causing multiple UDP packets to be sent to `espota.py`.

`espota` will throw out anything after the first packer with a `E` (i.e. `ERROR`) meaning the text messages are lost.

Simple fix is to concat the errors into a `String` prior to sending, then use a single `out.println()` call.

mcspr, 2022-08-21:
...could be taken from https://github.com/esp8266/Arduino/pull/8653/files
maybe even with the additional `String`-returning method

earlephilhower, 2022-08-21:
:+1: 

The title "Updater lifetime callbacks" didn't really cover this but I see it's part of the changed, but thx for pointing it out!  I'll close this, no worries.

Just trying to keep track of things I find in the Pico and make sure they are covered upstream!

closed 2022-08-21 by earlephilhower

mcspr, 2022-08-21:
It is complicated, yes:)

Just checking, I don't see error string being pushed multiple times in this implementation as everything is buffered in StreamString first
https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/libraries/ArduinoOTA/ArduinoOTA.cpp#L253-L257
This part seems to do this in sync-mode, but not sure if it matters
https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/libraries/ArduinoOTA/ArduinoOTA.cpp#L342