esp8266/Arduino #8662: DHCP Close Websocket Error 1006

ahdaib, 2022-08-24:
Dear All
there is Problem with DHCP connection in STA Mode, I See When i connect DHCP to Router the WIFI hit onStationModeGotIP
every 50 Sec 
When i Use Web-socket Connection it Disconnect After 50 Sec because WIFI Got IP

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```cpp
 #include <ArduinoWebsockets.h>
 #include <ESP8266WiFi.h>
 const char* ssid = "SSID"; //Enter SSID
 const char* password = "PASSWORD"; //Enter Password
 const char* websockets_server_host ="ws.postman-echo.com"; 
 const uint16_t websockets_server_port = 443;
 using namespace websockets;
 WebsocketsClient client;
 void onMessageCallback(WebsocketsMessage message) {
     Serial.print("Got Message: ");
     Serial.println(message.data());
 }
 void onEventsCallback(WebsocketsEvent event, String data) {
     if(event == WebsocketsEvent::ConnectionOpened) {       
         Serial.println("Connnection Opened");
     } else if(event == WebsocketsEvent::ConnectionClosed) {
         Serial.println(client.getCloseReason());            
         Serial.println("Connnection Closed");
     } else if(event == WebsocketsEvent::GotPing) {
         Serial.println("Got a Ping!");
     } else if(event == WebsocketsEvent::GotPong) {
         Serial.println("Got a Pong!");
     }
 }
 void setup() {
     Serial.begin(115200);
       WiFi.mode(WIFI_STA);     
     WiFi.begin(ssid, password);
     while ( WiFi.status()!= WL_CONNECTED)
     {
       Serial.print(".");
        delay(1000);
       }
    
     client.onMessage(onMessageCallback);
     client.onEvent(onEventsCallback);
     client.connectSecure(websockets_server_host, websockets_server_port,"/raw");
     client.ping();
 }
 
 void loop() {
    client.poll();
 }

```
in Debug i See loop of onStationModeGotIP Handler repeated every 50 sec 
i think this broken the websocket connection 
### Debug Messages
```
wifi evt: 2
..scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with SSID, channel 6
dhcp client start...
wifi evt: 0
..........pm open,type:2 0
ip:192.168.100.119,mask:255.255.255.0,gw:192.168.100.1
wifi evt: 3
[hostByName] request IP for: ws.postman-echo.com
[hostByName] Host: ws.postman-echo.com IP: 35.173.121.207
Connnection Opened
Got a Pong!
Got a Ping!
Got a Ping!
1006
Connnection Closed
ip:192.168.100.119,mask:255.255.255.0,gw:192.168.100.1
wifi evt: 3
ip:192.168.100.119,mask:255.255.255.0,gw:192.168.100.1
wifi evt: 3
ip:192.168.100.119,mask:255.255.255.0,gw:192.168.100.1
wifi evt: 3

```

###
but when i use Static IP every thing work fine and the  there is no loop for onStationModeGotIP Handler 

is there any solution to stop  StationModeGotIP after get the IP from Router or stop the loop of StationModeGotIP

Thank you

Pablo2048, 2022-08-24:
Did you check the DHCP lease time in your DHCP server settings?

ahdaib, 2022-08-24:
Thank you @Pablo2048 
i increase DHCP lease time and it is work fine

closed 2022-08-24 by ahdaib