esp8266/Arduino #8673: Printf 64 bit integers in HEX only shows lower 32-bit part

bertmelis, 2022-09-12:
Printing 64 integers using printf doesn't seem to work properly.
While it does print when formatted as decimal, the hex output only takes the first 32-bit part.

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(74880);
  delay(5000);

  uint64_t testvar = 0xFFFFFFFFFFFFFFFF;
  Serial.printf("%llu\n", testvar);
  Serial.printf("0x%016X\n", testvar);
  Serial.printf("%"PRIu64"\n", testvar);
  Serial.printf("workaround: 0x%lx%lx\r\n", (uint32_t)(testvar>>32), (uint32_t)testvar);
 }

void loop() {
}
```

Output:
```
18446744073709551615
0x00000000FFFFFFFF
18446744073709551615
workaround: 0xffffffffffffffff
```

(issue is equally valid for signed integers)

mcspr, 2022-09-12:
Wasn't this already fixed in master branch?
https://arduino-esp8266.readthedocs.io/en/latest/installing.html
https://docs.platformio.org/en/stable/platforms/espressif8266.html#using-arduino-framework-with-staging-version

I would assume the linked issue is dealing with the latest boards manager / PIO release

bertmelis, 2022-09-12:
corrected code, tested with git version

```cpp
  Serial.printf("%llu\n", testvar);
  Serial.printf("0x%016llX\n", testvar);
  Serial.printf("%" PRIu64 "\n", testvar);
  Serial.printf("workaround: 0x%x%x\r\n", (uint32_t)(testvar>>32), (uint32_t)testvar);
```

closed 2022-09-12 by bertmelis

mcspr, 2022-09-12:
Yep, looking at warning output should also be pretty apparent
```
src/main.cpp:14:34: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'long long unsigned int' [-Wformat=]
   14 |     Serial.printf("serial x - %16x\n", test);
      |                               ~~~^     ~~~~
      |                                  |     |
      |                                  |     long long unsigned int
      |                                  unsigned int
      |                               %16llx
```