esp8266/Arduino #8675: More for fixing MDNS

d-a-v, 2022-09-13:
After @hreintke 's fix (#8641 from #8308), this [notice](https://github.com/esp8266/Arduino/issues/8308#issuecomment-1186495220) and these lwIP guidelines [brought to attention](https://github.com/esp8266/Arduino/issues/8308#issuecomment-1221656181) by @mcspr,

our mDNS implementation will benefit from part or all these suggestions:
- [ATTENTION: netif_set_up() now always has to be called, even when dhcp/autoip is used!](https://git.savannah.nongnu.org/cgit/lwip.git/commit/?id=4e520cdd3009cf2e04c50c173737b379ff7d72a2)
- `LwipIntf::stateUpCB()` should trigger also when IP address change

mcspr, 2022-09-13:
> LwipIntf::stateUpCB()

s/LwipIntf::stateUpCB()/LwipIntf::ipChangeCB()/ ?
in case of DHCP, that would be called twice

Might be confusing for static ip? Need to check examples, though. Is it 'STATE UP -> set ip addresses' or 'set ip addresses -> STATE UP'?
In the referenced issue that was 'solved' by optional EXT callbacks feature within LWIP core, I don't see any callbacks in dhcp client. Same for auto-ip :/

d-a-v, 2022-09-13:
We can also considerate bringing interface DOWN then UP when IP changes

mcspr, 2022-09-13:
Also seems weird :/ Can we replace `netif_status_changed` with `netif->status_callback`?
https://github.com/lwip-tcpip/lwip/blob/239918ccc173cb2c2a62f41a40fd893f57faf1d6/src/core/netif.c#L350-L352
https://github.com/lwip-tcpip/lwip/blob/239918ccc173cb2c2a62f41a40fd893f57faf1d6/src/core/netif.c#L504 (called on IP change)
https://github.com/lwip-tcpip/lwip/blob/239918ccc173cb2c2a62f41a40fd893f57faf1d6/src/core/netif.c#L882 (called on UP / DOWN)

closed 2022-12-14 by mcspr