esp8266/Arduino #8681: Wemos D1 mini ESP8266 wifi got stuck at debug message 'pm open,type:0 0'

opc241, 2022-10-02:
Hi there,

i have a Wemos D1 mini with ESP8266 on it and can connect to my wifi with the following code:

`#include <ESP8266WiFi.h>
#include <NTPClient.h>
#include <WiFiUdp.h>
#include <string>

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org");

const char* SSID = "MYSSID";
const char* PSK = "MYPSK";

WiFiClientSecure espClient;         /* Secure client connection class, as opposed to WiFiClient */
PublicKey key(pubkey);

void setup_wifi() {
    WiFi.setSleepMode(WIFI_NONE_SLEEP);
    WiFi.mode(WIFI_STA);    
    WiFi.disconnect();
    WiFi.begin(SSID, PSK);
    Serial.println("Try to connect to wifi ");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }    
    Serial.println(WiFi.localIP());
    espClient.setX509Time(timeClient.getEpochTime());
    Serial.write(timeClient.getEpochTime());
}

void setup() {
     
    // Serial Monitor
    Serial.begin(115200);
   
    // STATUS LED
    pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output

    // WIFI
    setup_wifi(); 
}

 
void loop()
{     
 
}`

So i also set the "not go to sleep" command for wifi. 
The problem is, that wifi connects only when i restart my wifi in router. If i don't restart it, then sometimes the wifi will connect properly, but only for a few times. Then the wifi connecting will get stuck with the following wifi debug message: 

`connected with <MY_SSID>, channel 11
dhcp client start...
wifi evt: 0
pm open,type:0 0`

So it seems to be connected, but there is no ip address allocated and the wifi status never reaches the WL_CONNECTED state. 
The connection got stuck at **pm open,type:0 0**

I thought this is a problem with 5GHz wifi, so i turned it off in my router, which is now only set up to 2,4 GHz. 
And i also changed the wifi channel selection from below channel 11 in my router settings. 

I am very looking forward for any hints. 

Thanks in advance!

TD-er, 2022-10-02:
Can you remove (or comment) the `WiFiClientSecure` related code?
Does it then connect?

Next step would be to set a flag whether you're connected (or simply check the `WiFi.status()`) and try to perform the NTP call from the `loop()`

opc241, 2022-10-02:
Yes, so when i run the code without the secured WifiClientSecure it will connect properly. 
(But i will need the security stuff for X509 certs against my MQTT broker). 

The WiFi.status() is returning the return code: 6 and therefore it will never break the while loop.

I am really not sure if there is a proper dhcp allocation for the wemos. My router sometimes says yes (wemos has 192.168.2.113), but when it comes to the wifi connection failure my router says it has the ip 0.0.0.0

TD-er, 2022-10-02:
And when you try to initialize the `WiFiClientSecure` object + setup the NTP in the loop, not in the `setup()` ?

mcspr, 2022-10-03:
Check out our docs related to git installation and see if that helps with the connection issue.
Does it work when you *don't* use DHCP, and set a static IP?
There was #8319 , but it seems to only have fixed hostname assignment.

Related to code... you don't have to use NTPClient lib. We always include SNTP lwip app, either use configTime or sntp\_... functions ([like here]( https://github.com/esp8266/Arduino/blob/a0c7a85649a3c48c05e70f3014769af23d01ec3b/cores/esp8266/time.cpp#L232-L242)). Client will use the posix `time()` and you don't have to update it each connection since it will sync periodically.