esp8266/Arduino #8684: BasicOTA upload button remain disabled after successfully uploading code.

MorgothCreator, 2022-10-09:
MacOSX 12.6, Arduino IDE 2.0.0

Everything is working OK except after uploading the code Arduino IDE still is showing the message "Uploading......" and upload button remain disabled.

The target board is reseting as desired and run the new code.

The only workaround is to close Arduino IDE and reopening it, annoying as hell :)

mcspr, 2022-10-10:
Does not look like it is from our side, IDE interface is not under our jurisdiction :)
https://github.com/arduino/arduino-ide/issues/1356
https://github.com/arduino/arduino-ide/blob/6f077173693cdfd1ed2a65a5754f05b4fcca8ac6/arduino-ide-extension/src/browser/contributions/upload-sketch.ts#L137-L147

I'd subscribe to that issue instead and try to find what should trigger `Uploading...` notification removal in the IDE code. If it is something in our tooling exit codes / stdout / etc. we could fix that. Until then, closing in favour of the upstream issue

closed 2022-10-10 by mcspr

Jens869, 2022-10-11:
Or just use a timeout. If the upload is not finished after 1 minutes, then it either failed or finished. 
When the button becomes activatable again, the user can try again. Or use a cable.
If the button remains grey, only the sketch can be closed and open again.