esp8266/Arduino #8689: User is unable to define is WiFi.disconnect should null SSID and Password

cziter15, 2022-10-12:
This problem is not really related to any board, but directly to the source code, so I've decided to avoid unnecessary template here.

The problem is that WiFi.disconnect function always set SSID and Password to NULL. On ESP32's version of Arduino, user can define if the config should be saved or not. When manually implementing reconnection, this requires user to store credentials between disconnect and connect.

This should be unified somehow or at least eraseAp boolean should be added to allow calling underlying methods without zeroing credentials.

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L345
https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L347

d-a-v, 2022-12-12:
> On ESP32's version of Arduino, user can define if the config should be ever touched or not

Would you be ready to make an API addition proposal [to match esp32's](https://github.com/espressif/arduino-esp32/blob/9275dbff942bb2ccce62632607500ad27dd0513a/libraries/WiFi/src/WiFiSTA.cpp#L345) ?

cziter15, 2022-12-12:
I will issue PR, but give me few hours :)

cziter15, 2022-12-12:
I think it's better to erase whole AP data. We don't know how user will use it, but in case we want to forget AP data, we should erase at least whole password field.

cziter15, 2022-12-12:
I've re-issued PR !8758

closed 2022-12-15 by d-a-v