esp8266/Arduino #8691: Smartconfig fails to retrieve WiFi credentials

BarrMan, 2022-10-14:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01|
- Core Version:			3.0.2
- Development Env:		Arduino IDE/Platformio
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dout
- Flash Size: 		1MB
- lwip Variant:   Unknown
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz|
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I’m trying to use esp8266 smart config but it fails to connect. There’s no error message shown and after inspecting the smart config app UDP message using WireShark, I see that the UDP messages are sent correctly.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include "smartconfig.h"

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);
  delay(500);
  esptouch_set_timeout(20);
  
  // timeout
  WiFi.beginSmartConfig();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.smartConfigDone());
  }
  Serial.println("WiFi connected");  
  Serial.println(WiFi.localIP());

}

```

### Debug Messages
[image]

Wireshark logs:
[image]

I’ve tried adding some more logs in _smartConfigCallback to find out what actually happens and it looks like it’s trying to find a channel.

Also, I saw that my smartconfig version is v2.5.4 and in other posts, I see that people are using version 2.9.0. Tried looking for a way to upgrade the version but found no resources on that as well.

Is it an issue with the version of smartconfig that I’m using? Is there a way to view more debugging logs so I could figure out what the exact issue is?

Thanks a lot!

Pontifex42, 2023-01-19:
Same issue on my ESP8266-12E / -12F. additionaly tested on a D1 mini, too. SmartConfig does not work on any of my ESP8266, while other ways of connecting to WiFi work well there. I tried same SmartConfig-Connect code on an ESP32 dev module and there, it works. So I am sure the problem is not my App nor my Android device.
I tried ESP8266 board managers V3.02, V3.1.0 and 3.1.1. Environment is Arduino IDE V2.0 and the VisualStudio-Plugin "VisualMicro" (which has other issues with V3.1 board manager). Firmware always prompts "ets Jan  8 2013".
I'm pretty sure there is something rotten in the state of denmark... and in the WiFi lib of the board manager.
Any help somewhere?

muzhiyun, 2024-05-23:
It seems that this problem still occurs. Does anyone have any updates or ideas?

prashanthsun9, 2024-11-30:
Does anyone have any update on this, it is not working even now