esp8266/Arduino #8692: WiFiClientSecure returns 0 for remoteIP(), remotePort(), localIP(), localPort() calls

JiriBilek, 2022-10-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [n/a ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01 (but doesn't matter)
- Core Version:			latest git (d3eddeb5014d04aa9e61b97474ef98d0214eab1a)
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT
- Flash Size: 		1MB
- lwip Variant:   v2.1 Lower Memory
- Reset Method:		dtr
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

The functions `WiFiClientSecure.remoteIP(), remotePort(), localIP(), localPort()` return 0. This is caused by the fact, that `WiFiClientSecure` is derived from `WiFiClient` but doesn't use its `ClientContext* _client` variable. It uses `WiFiClient` class that is the base of `WiFiClientSecureCtx* _ctx` instead, as is written in the documentation of the `WiFiClientSecure` class.

The problem is, that `WiFiClientSecure.remoteIP()` calls `WiFiClient.remoteIP()` with `_client=nullptr` instead of calling something like `WiFiClientSecure._ctx->remoteIP()`. The same holds with the other 3 function mentioned in the title.

The solution is to make these functions virtual and override them in `WiFiClientSecure` class.
I am creating a PR in no time.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

The sketch is a modified WiFiClient example

```cpp
/*
    This sketch establishes a TCP connection to a "quote of the day" service.
    It sends a "hello" message, and then prints received data.
*/

#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "xxx"
#define STAPSK  "xxx"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "github.com";
const uint16_t port = 443;

void setup() {
  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  static bool wait = false;

  Serial.print("connecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  // Use WiFiClient class to create TCP connections
  WiFiClientSecure client;
  client.setInsecure();
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    delay(5000);
    return;
  }

  Serial.print(client.remoteIP().toString().c_str());
  Serial.print(':');
  Serial.println(client.remotePort());

  // Close the connection
  Serial.println();
  Serial.println("closing connection");
  client.stop();

  if (wait) {
    delay(300000); // execute once every 5 minutes, don't flood remote service
  }
  wait = true;
}

```

### Debug Messages

expected:
```
WiFi connected
IP address:
192.168.1.137
connecting to github.com:443
140.82.121.4:443
```

got:
```
WiFi connected
IP address:
192.168.1.137
connecting to github.com:443
(IP unset):0
```

regdomain, 2022-10-19:
Thanks for your fix, you're great, man! I've dug through the whole Internet trying to figure out why `server.client().remoteIP().toString()` returns "(IP unset)"!

JiriBilek, 2022-11-02:
Fixed with https://github.com/esp8266/Arduino/pull/8693

closed 2022-11-02 by JiriBilek