esp8266/Arduino #8696: ESP8266 Crashing on conn.connect

Chopan25, 2022-10-20:
when running the example pro the basic insert esp8266 on my board it started crashing nonstop until i remove the connection to de database and it worked fine. this was the error:
```
Connecting to wifi.......
Connected to network
My IP address is: 192.168.0.88
...trying...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40202558 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000049 depc=0x00000000

>>>stack>>>
sp: 3ffffc40 end: 3fffffc0 offset: 0190
3ffffdd0:  00000000 00000000 00000000 fffffffe  
3ffffde0:  ffffffff 3fffc6fc 00000001 402025ab  
3ffffdf0:  3ffee83c 00000000 3ffe8629 40202598  
3ffffe00:  3ffef9f0 3ffeeb5c 3ffef5d0 40206730  
3ffffe10:  fffffffc 00000040 3fffff40 00000cea  
3ffffe20:  00000014 3ffef5cc 000000ff 3ffee60c  
3ffffe30:  402066fc 3ffe8629 3ffee674 40202ee8  
3ffffe40:  3ffef9ac 3ffee674 00000000 40202f0c  
3ffffe50:  3ffef9ac 3ffee674 3ffee674 40203068  
3ffffe60:  3ffeeb5c 00000000 00000000 4020295f  
3ffffe70:  40201705 402016e2 40100205 40215196  
3ffffe80:  00000170 00000170 3ffe8664 401006bf  
3ffffe90:  00000000 4bc6a7f0 f126e978 00000001  
3ffffea0:  3ffefbb4 00000020 3ffefbb4 4010089e  
3ffffeb0:  3ffee1a0 00000000 00000000 40100294  
3ffffec0:  00001316 3ffee83c 0000012c 40215bfc  
3ffffed0:  0000012f 0000012f 3ffe8664 401006bf  
3ffffee0:  3ffe8629 00000000 3ffef68c 3ffe8629  
3ffffef0:  3ffe8614 00000035 00000020 3ffe8629  
3fffff00:  3ffe8614 00000035 3ffee60c 40202aa2  
3fffff10:  00000010 3ffef9ac 3ffee60c 40202816  
3fffff20:  0000004d 00000100 0001c200 00000003  
3fffff30:  00000001 3ffee74c 3ffee60c 402021fa  
3fffff40:  40206bac 73752ab9 3ffee74c 40202c78  
3fffff50:  3fffff90 00000cea 3ffe863e 3ffe8629  
3fffff60:  3ffe8614 40201f54 3ffee74c 40203098  
3fffff70:  5800a8c0 00ffffff 0100a8c0 3ffee7d8  
3fffff80:  3ffe85cc 3ffee60c 3ffee74c 402010fb  
3fffff90:  40206bac 73752ab9 feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee7c4 40203c0c  
3fffffb0:  feefeffe feefeffe 3ffe8660 40100c25  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00044e10
~ld

```
After a while of messing around with no success i downgraded the esp board librery to the 2.7.0 version and it just worked on the first try

Chopan25, 2022-10-20:
I think i haven't explain myself well enough.
the library I'm using is: MySQL Connector Arduino
I'm uploading the library example, changing my network and database parameters and uploading it, it compiles perfectly and uploads it. it runs untill it get to the database conection, line 66 of the example, that is wen the esp throws the exception and restarts. that is with the version 3.0 of the esp core install on the Arduino IDE version 2.0. what i did was uninstalling the board library and install the version 2.7 of the esp core and then it workedwith no changes to the code.
So there is some incopatibility between the new version of the board library and the method MySQL_Connection.connect. that has appeared between the version 2.7.0 and the lastest version of the ESP8266 library for the Arduino IDE. i will try to decode the exception and post it as a reply to this issue.
If you think the one resposable for the error is the library i'm using please tell me and I will post in their github repo.

Chopan25, 2022-10-21:
this is what it has decoded 

```
Decoding 28 results
0x402025ab: Encrypt_SHA1::write(unsigned char) at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Encrypt_Sha1.cpp line 88
0x40202598: Encrypt_SHA1::addUncounted(unsigned char) at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Encrypt_Sha1.cpp line 84
0x40206730: Print::write(unsigned char const*, unsigned int) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.cpp line 39
0x402066fc: IPAddress::fromString4(char const*) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/IPAddress.cpp line 103
0x40202ee8: Print::write(char const*) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.h line 59
0x40202f0c: Print::write(char const*) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.h line 60
0x40203068: Print::print(__FlashStringHelper const*) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.cpp line 114
0x4020295f: MySQL_Packet::scramble_password(char*, unsigned char*) at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Packet.cpp line 186
0x40201705: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 204
0x402016e2: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 195
0x40100205: millis at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40215196: ip4_output_if at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 1547
0x401006bf: umm_free_core at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 549
0x4010089e: free at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 595
:  (inlined by) free at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 574
0x40100294: vPortFree at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/heap.cpp line 373
0x40215bfc: mem_malloc at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 221
0x401006bf: umm_free_core at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 549
0x40202aa2: MySQL_Packet::send_authentication_packet(char*, char*, char*) at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Packet.cpp line 131
0x40202816: MySQL_Packet::parse_handshake_packet() at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Packet.cpp line 345
0x402021fa: MySQL_Connection::connect(IPAddress, int, char*, char*, char*) at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/src/MySQL_Connection.cpp line 99
0x40206bac: std::_Function_handler ::_M_manager(std::_Any_data&, std::_Function_handler  const&, std::_Manager_operation) at time.cpp line ?
0x40202c78: HardwareSerial::availableForWrite() at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 182
0x40201f54: WiFiClient::WiFiClient() at ?? line ?
0x40203098: Print::print(char) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.cpp line 126
0x402010fb: setup at /home/chopan/Arduino/libraries/MySQL_Connector_Arduino/examples/basic_insert_esp8266/basic_insert_esp8266.ino line 66
0x40206bac: std::_Function_handler ::_M_manager(std::_Any_data&, std::_Function_handler  const&, std::_Manager_operation) at time.cpp line ?
0x40203c0c: loop_end at ?? line ?
0x40100c25: schedule_recurrent_function_us(std::function  const&, unsigned int, std::function  const&) at /home/chopan/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Schedule.cpp line 134
```

Chopan25, 2022-10-21:
ok, going throw the issues in the Mysql library i found out this is a [known issue](https://github.com/ChuckBell/MySQL_Connector_Arduino/issues/185) for them, and they have just stayed on the version 2.7.4
reading the code i think a i foud a type error on an argument the mysql library is calling client.connect(IPAdresse ip, int port) and it expects connect(IPAddress ip, uint16_t port)

mcspr, 2022-10-21:
See https://github.com/ChuckBell/MySQL_Connector_Arduino/pull/177
I don't think library version was updated, though, so neither Arduino registry or PIO see the release
(btw I think I misread pro as PIO, sorry :)

Chopan25, 2022-10-21:
you are right, i just cloned the repo and tested with esp core 3.0.2 and it is working fine. thaks!