esp8266/Arduino #8702: ESP reset due to longer execution time of function

mihirgraspberry, 2022-10-27:
ESP 8266 reboots whenever a function takes more than say 60 seconds to finish execution. Is there any way to implement a time out for software watchdog to prevent this. 
I am doing a serial read and write with another device , this entire communication is done in one function. This function takes anywhere between 2minutes to. 10 minutes to finish execution.

fsommer1968, 2022-10-27:
Try using yield(); inside the function

mcspr, 2022-11-03:
See above.

ref. https://github.com/esp8266/Arduino/tree/master/tools/sdk/include/user_interface.h and https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf for SDK documentation.

Arduino level API yield() switches to the system task that does the feeding. And which you'd need to periodically do anyway, if you want networking to work. Otherwise, feed it manually 
https://github.com/esp8266/Arduino/blob/d3eddeb5014d04aa9e61b97474ef98d0214eab1a/cores/esp8266/Esp.cpp#L113-L115

closed 2022-11-03 by mcspr