esp8266/Arduino #8710: Functions / symbols defined in sketch may override framework .a

mcspr, 2022-11-06:
>        I tested the MCVE and OTA works again perfectly fine. I found the problem in my full sketch. I had a global object called crc32! The same name as the function name with problem. Sorry for bothering you and thank you @earlephilhower for your help. Closing this issue...

_Originally posted by @kamran00f in https://github.com/esp8266/Arduino/issues/7651#issuecomment-707327428_

mcspr, 2022-11-06:
Linking happens like this right now
> xtensa-lx106-elf-g++ -o .pio/build/d1_mini/firmware.elf -T eagle.flash.4m1m.ld -Os -nostdlib -Wl,--no-check-sections -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -u app_entry -u _printf_float -u _scanf_float -u _DebugExceptionVector -u _DoubleExceptionVector -u _KernelExceptionVector -u _NMIExceptionVector -u _UserExceptionVector .pio/build/d1_mini/src/main.cpp.o -L.pio/build/d1_mini -L.pio/build/d1_mini/ld -L/home/runner/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lib -L/home/runner/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/ld -L/home/runner/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lib/NONOSDK22x_190703 -Wl,--start-group .pio/build/d1_mini/lib2ea/libESP8266WiFi.a .pio/build/d1_mini/libb9e/libESP8266mDNS.a .pio/build/d1_mini/libc61/libArduinoOTA.a .pio/build/d1_mini/libFrameworkArduinoVariant.a .pio/build/d1_mini/libFrameworkArduino.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group

With a sketch c/p from the issue, lets add our own crc32 func
```cpp
uint32_t crc32 (const void* data, size_t length, uint32_t crc /*= 0xffffffff*/) {
    return 0;
}
```

Resulting .elf has our function and not the original one, crc32 returning 0.
```gdb
> xtensa-lx106-elf-gdb --batch --ex "disassemble crc32" .pio/build/d1_mini/firmware.elf
Dump of assembler code for function crc32(void const*, size_t, uint32_t):
   0x4020d0f0 <+0>:     movi.n  a2, 0
   0x4020d0f2 <+2>:     ret.n
End of assembler dump.
```

Moving main .o to the back of the 'library group' seems to help
```console
/home/runner/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini/src/main.cpp.o: in function `crc32':
/home/runner/dev/arduino7651/src/main.cpp:8: multiple definition of `crc32'; .pio/build/d1_mini/libFrameworkArduino.a(crc32.cpp.o):/home/runner/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/crc32.cpp:39: first defined here
collect2: error: ld returned 1 exit status
```
Not sure if that has any more side-effects, or whether this feature is used by any sketches.
(iirc tasmota might override wificlient this way)

TD-er, 2022-11-15:
Is it possible to add a warning when a (global) function is redefined? Maybe even better to stop further compilation. I rather would have a build to fail compared to nearly impossible to find issues which may appear and disappear nearly random on new builds.
We've seen enough of those the last few years.