esp8266/Arduino #8713: Can not upload sketch to nodemcu

omercantasdemir, 2022-11-13:
I installed CH340 drivers, checked mutiple data cables, pressed flash button during upload, grounded D3 pin, tried multiple arduino ide versions but i cant seem to upload even the simplest sketch to my wemos nodemcu v3. My pc detects the board but cant communicate. I am sure I selected correct board and correct port.

omercantasdemir, 2022-11-13:
So far i got "timed out" and comm_open failed errors

omercantasdemir, 2022-11-13:
here is an error message
Arduino: 1.8.19 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

ICACHE : 32768           - flash instruction cache 

IROM   : 231724          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26793   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 

DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 876   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 25608 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 260889 bytes (24%) of program storage space. Maximum is 1044464 bytes.

Global variables use 27980 bytes (34%) of dynamic memory, leaving 53940 bytes for local variables. Maximum is 81920 bytes.

esptool.py v3.0

Serial port COM3

Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):

  File "C:\Users\omerc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/upload.py", line 66, in <module>

    esptool.main(cmdline)

  File "C:/Users/omerc/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 3552, in main

    esp.connect(args.before, args.connect_attempts)

  File "C:/Users/omerc/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 529, in connect

    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_

the selected serial port _

 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mcspr, 2022-11-13:
^ `C:\Users\omerc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\tools\esptool\esptool.py` could be used to debug the upload further (needs `pyserial`), or manual installation with `pip install esptool` (https://pypi.org/project/esptool/). It would check whether the board comms work, and it has some more tools to check flash size, get chip info, etc.

Nodemcu (most of time) should work with our defaults, we do some magic to automatically reboot to flash. Also note that you mustn't hold D3 too long; start with GPIO0 <-> GND, press RST, release GPIO0 -> board is ready for the upload.
Using 74880 baudrate you should see the boot process happening in any terminal emulator; if there is no serial communication happening, upload can't happen. So it is good to know it actually works before trying anything else.
---

Prefer our [gitter](https://gitter.im/esp8266/Arduino) chat or https://esp8266.com/ for general questions, though. Timeout usually means hardware errors :>

closed 2022-11-13 by mcspr