esp8266/Arduino #8714: HTTPClient fixed timeout sending payload HTTPC_ERROR_SEND_PAYLOAD_FAILED

asturcon3, 2022-11-14:
ESP8266HTTPClient.cpp does 
```
int HTTPClient::sendRequest(const char * type, const uint8_t * payload, size_t size)
...
        // transfer all of it, with send-timeout
        if (size && StreamConstPtr(payload, size).sendAll(_client.get()) != size)
            return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);

```

StreamConstPtr derives from Stream which defines 
```
    protected:
        unsigned long _timeout = 1000;  // number of milliseconds to wait for the next char before aborting timed read
```

So no matter what your HTTPClient.setTimeout is, 1 second pause on your server side causes a HTTPC_ERROR_SEND_PAYLOAD_FAILED.

It's obvious, but httpclient should create the StreamConstPtr, set its timeout to either the full or the remaining millisecs, and then issue the sendAll.

mcspr, 2022-11-14:
If you have specific code change in mind, no need to be formal and create an issue - just go with the PR.
Do you have a specific code example to reproduce? Just at a glance, we always do `_client->setTimeout()` via `::setTimeout` method that updates both internal var and client when connected.

asturcon3, 2022-12-02:
I'm leaving the PR to the people with the expertise, what I can do is a sample based on two ESP's, one will be a dumb server with a pause when receiving data, the other will be a client sending a bunch of data.

ESP8266 1. The server
```
#include <ESP8266WiFi.h>

WiFiServer server(80);

void setup() {
	Serial.begin(74880);
	Serial.print("setup");
	WiFi.begin("ssid","pwd");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
	Serial.print("Conectado, dirección IP: ");
  Serial.println(WiFi.localIP());
	server.begin(80);
}

void loop() {
	WiFiClient client = server.available();
	if (client) {
		Serial.print("Connected ");
		Serial.println(client.available());
		bool bPaused=false;
		while (client.connected()) {
			int count=0;
			while (client.available()){
				char c=client.read();
				//Serial.print(String(c));
				count++;
				if (!bPaused && count+client.available()>300) {
					bPaused=true;
					Serial.println(""); Serial.print("pause at "); Serial.println(count+client.available());
					unsigned long ms=millis();
					noInterrupts();
					while (millis()-ms<2000) ; // busy loop, freeze server
					interrupts();
					Serial.println("/pause");
				}
			}
			delay(10);
		}
		Serial.println("");
		Serial.println("disconnected");
	}
}

```

ESP 2, the client
```

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

#define dataSize (16*1024)

char *buffer;

void setup() {
	Serial.begin(74880);
	Serial.print("setup");
	WiFi.begin("ssid","pwd");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
	Serial.print("Conectado, dirección IP: ");
  Serial.println(WiFi.localIP());
	
	buffer=(char *)malloc(dataSize);
	if (!buffer) Serial.println("No mem!!");
	for (int i=0;i<dataSize;i++) buffer[i]=65+i%32;
}

void loop() {
	Serial.println("Go...");
	WiFiClient cli;
	HTTPClient http;
	http.setReuse(false); http.useHTTP10(true); // no reuse please, just plain
	http.setTimeout(5000);
	bool r=http.begin(cli,"http://192.168.1.143"); // Server IP here
	if (!r) {
		Serial.print("begin fail");
	} else {
		unsigned long ms=millis();
		int httpCode= http.POST((const uint8_t*)buffer,(size_t)dataSize);
		Serial.print(millis()-ms); Serial.print("ms : ");
		Serial.print(httpCode); Serial.println(http.errorToString(httpCode));
	}
	delay(1000);
}
```

Note the `http.setTimeout(5000);` line

Now, the client side log:
```
9:40:35.361 -> ...............Conectado, dirección IP: 192.168.1.138
19:40:43.115 -> Go...
19:40:44.763 -> 1624ms : -3send payload failed
19:40:45.751 -> Go...
19:40:47.117 -> 1361ms : -3send payload failed
19:40:48.106 -> Go...
19:40:53.332 -> 5215ms : -1connection failed
19:40:54.321 -> Go...
19:40:58.746 -> 4419ms : -3send payload failed
19:40:59.734 -> Go...
19:41:01.100 -> 1364ms : -3send payload failed
19:41:02.136 -> Go...
19:41:03.596 -> 1482ms : -3send payload failed
19:41:04.585 -> Go...
19:41:09.857 -> 5235ms : -1connection failed
19:41:10.846 -> Go...
19:41:12.165 -> 1326ms : -3send payload failed
19:41:13.153 -> Go...
19:41:14.614 -> 1445ms : -3send payload failed
19:41:15.602 -> Go...
19:41:16.967 -> 1352ms : -3send payload failed
19:41:17.956 -> Go...
19:41:19.321 -> 1352ms : -3send payload failed
19:41:20.311 -> Go...
19:41:21.676 -> 1357ms : -3send payload failed
19:41:22.664 -> Go...
19:41:24.170 -> 1486ms : -3send payload failed
19:41:25.159 -> Go...
19:41:29.659 -> 4521ms : -3send payload failed
19:41:30.696 -> Go...
19:41:32.108 -> 1401ms : -3send payload failed
19:41:33.098 -> Go...
19:41:34.464 -> 1372ms : -3send payload failed
19:41:35.453 -> Go...
19:41:40.675 -> 5236ms : -1connection failed
19:41:41.711 -> Go...
19:41:43.030 -> 1324ms : -3send payload failed

```

Note the sort interval for most errors, much lower than expected. For failed connections (it's a dumb single client server) it's ok.

If you comment out the server's pause, the client's log is as expected, a bunch of `5297ms : -11read Timeout` lines (the server doesn't write)