esp8266/Arduino #8715: Longer delays for Ticker does not work

enjoyneering, 2022-11-15:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

The new [longer Ticker delays](https://github.com/esp8266/Arduino/commit/27c0591756c7841f0241638cb54b1102c844fd60) does not work with 86400s (24h) delays. Also tested 28800s (8h) and seems to work fine.

enjoyneering, 2022-11-15:
Closed for now. I think there is an error in my calculation.

closed 2022-11-15 by enjoyneering

mcspr, 2022-11-15:
My initial hunch is to look here, but numbers seem to be different for ok and failing cases. Maybe I was wrong setting this as limit?
https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/libraries/Ticker/src/Ticker.h#L183-L185
(iirc tick calc itself should be ok, but I haven't actually waited for 24h at any point; tests were done using a lower limit to see whether algorithm works)

10h -> we 'tick' every 4500000ms / 75min 8 times (assumed ok)
15h -> every 6750000ms / ~112min 8 times (larger than 24h, but ok?)
24h -> 5400000ms / 90min 16 times (fails above)