esp8266/Arduino #8716: exception.what() not a string

mhightower83, 2022-11-16:
### Basic Infos

- [*] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [*] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [*] I have tested that the issue is present in current master branch (aka latest git).
- [*] I have searched the issue tracker for a similar issue.
- [*] If there is a stack dump, I have decoded it.
- [*] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01S]
- Core Version:			[80bf71662551fb5e6579713dc15e541ac48ccf98]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600 (serial upload only)

### Problem Description

For the case of a missing `try/catch` around `new`, a secondary crash within postmortem occurs while processing the unhandled C++ Exception.

The second crash is caused when `e.what()` returns a value of `0x12`.
https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/cores/esp8266/core_esp8266_main.cpp#L296
resulting in a new ESP exception in postmortem at
https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/cores/esp8266/core_esp8266_postmortem.cpp#L144

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
/*
   Build with Tools->C++ Exceptions: "enable"
*/
char *p = NULL;

void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.printf("\nUp and running ...\n");

  // loops - crashing in postmortem
  p = new char[128000];

  if (p) {
    Serial.printf("\n%p, \"new\" worked, it should not have!\n", p);
  }
}

void loop() {}
```

### Debug Messages

```
Up and running ...

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset

```

mcspr, 2022-11-16:
... #8717 as an alternative way of overloading new. note c++17 also has aligned ones, seems like a waste writing all of those yet again :/

`std::bad_alloc::what()` is indeed returning `0x12`. 
```gdb
arduino8716 [1]> xtensa-lx106-elf-gdb --batch --ex "disassemble 'std::bad_alloc::what'" --ex "x 0x40204674" .pio/build/d1_mini/firmware.elf
Dump of assembler code for function std::bad_alloc::what() const:
   0x4020d614 <+0>:     l32r    a2, 0x40204674
   0x4020d617 <+3>:     ret.n
End of assembler dump.
0x40204674:     0x00000012
```

Here's the string, though
```
> xtensa-lx106-elf-objdump -s -j .irom.exceptiontext  ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/lib/libstdc++-exc.a
...
bad_alloc.o:     file format elf32-xtensa-le

Contents of section .irom.exceptiontext:
 0000 7374643a 3a626164 5f616c6c 6f6300    std::bad_alloc.
...
```
([see toolchain .patch](https://github.com/earlephilhower/esp-quick-toolchain/blob/master/patches/gcc10.3/gcc-exception-what-to-pmem.patch))

mcspr, 2022-11-16:
this seems to fix the discarded address, just not sure why the rule(s) below try to put things in .text
```diff
diff --git a/tools/sdk/ld/eagle.app.v6.common.ld.h b/tools/sdk/ld/eagle.app.v6.common.ld.h
index 77c834ae..ed838918 100644
--- a/tools/sdk/ld/eagle.app.v6.common.ld.h
+++ b/tools/sdk/ld/eagle.app.v6.common.ld.h
@@ -181,8 +181,8 @@ SECTIONS
     *libgcc.a:_fixsfsi.o(.literal .text)
     *libgcc.a:_cmpdf2.o(.literal .text)
     *libgcc.a:_cmpsf2.o(.literal .text)
     *libstdc++.a:( .literal .text .literal.* .text.*)
-    *libstdc++-exc.a:( .literal .text .literal.* .text.*)
+    *libstdc++-exc.a:( .literal .text .literal.* .text.* .irom.exceptiontext)
     *libsmartconfig.a:(.literal .text .literal.* .text.*)
     *liblwip_gcc.a:(.literal .text .literal.* .text.*)
     *liblwip_src.a:(.literal .text .literal.* .text.*)
```

mhightower83, 2022-11-16:
@mcspr 
Interesting options.
Your change to `eagle.app.v6.common.ld.h` worked for me. 
Putting the strings into `.irom0.text` appears to land in the PROGMEM address range.
Thanks!

mhightower83, 2022-11-22:
@mcspr Also for me, removing the comma here allows things to work. https://github.com/esp8266/Arduino/blob/80bf71662551fb5e6579713dc15e541ac48ccf98/tools/sdk/ld/eagle.app.v6.common.ld.h#L223

closed 2022-11-24 by mhightower83

mcspr, 2022-11-24:
*(somewhat late comment, thanks for the fix!)*

^ re above and [Earles comment about ld](https://github.com/esp8266/Arduino/pull/8726#pullrequestreview-1192118426), there is actually nothing about such syntax option in up-to-date binutils documentation.

https://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_chapter/ld_toc.html
> January 1994

vs.
https://sourceware.org/binutils/docs/ld/Input-Section-Basics.html

Nothing in the testsuite either; `git log -p -- ./ld/testsuite/**.ld`, search for `, .` gives nothing useful. Noticed this 'ordered' list, but not really sure of the difference between the two (manual sorting order?)
```diff
+  .got            : { *(.got.plt) *(.igot.plt) *(.got) *(.igot) }
```
https://sourceware.org/git?p=binutils-gdb.git;a=tree;f=ld/testsuite;h=1f226f4c011ade55ee276991a21631a19ef675fd;hb=HEAD