esp8266/Arduino #8719: Help documentation improvements - adding nonos-sdk version information

enjoyneering, 2022-11-20:
### Problem Description

Hi,

Hi, is it possible to add information about nonos-sdk versions used in Arduino? So far we have 7!!! And not a word about the differences between them. A quick search gives a little clarification for "nonos-sdk 2.2.1+61 (190313)" and that's it.

[image]

My experience with "nonos-sdk 2.2.1+119 (191122)" shows high heap memory usage and eventually crashes ESP, especially when using BearSSL.

TD-er, 2022-12-06:
Welcome to the world of Espressif Releases.
Best known for:
- No release info, whatsoever, nothing, noppes, nada!
- No source code of the WiFi parts which obviously have bugs which are nextto impossible to pinpoint.

So far I know:

190313:
Allows to connect using 802.11n, later revisions of the 2.2.x don't, even though they claim to do so.
In my experience the only SDK which allows some really troublesome ESP boards (e.g. some Sonoff Basic units) to connect to WiFi with a stable connection.
However on other boards, the opposite may happen as they no longer have a stable and good connection when using this SDK.

190703:
Deemed the most stable by the Tasmota guys.

I myself use these two, but as I said, there is no "best" SDK build as neither of these will generate builds that will perfect on all ESP8266/8285 boards out there.

enjoyneering, 2022-12-07:
Hi TD-er,

Thanks for the comment. So far I'm using 190313. It shows the best uptime (weeks) with my code. The same project and 191122 often crashes with BearSSL, but this may be due to my poor programming skills. I will definitely try 190703.