esp8266/Arduino #8725: xtensa-lx106-elf-g++: bad CPU type in executable on a M2 Mac without rosetta emulation installed

aleroot, 2022-11-23:
I'm trying to compile and upload a sketch to a NodeMCU Lolin V3 board using arduino-cli_0.29.0_macOS_ARM64 but I get the following error whenever I try to compile:

```
alessio@AleMacBookPro Tools % ./arduino-cli compile --fqbn esp8266:esp8266:nodemcuv2 example

Used platform   Version Path                                                                    
esp8266:esp8266 3.0.2   /Users/alessio/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2

Error during build: fork/exec /Users/alessio/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++: bad CPU type in executable
```

Looks clear that the executable file xtensa-lx106-elf-g++ has been built for x86-64 Mac instead of AMR64... How can I download the correct files for my architecture?

I have been following [this procedure](https://ericslenk.com/posts/getting-started-with-the-nodemcu-esp8266-and-arduino-cli), the problem is that `arduino-cli core install esp8266:esp8266` downloads the wrong architecture executables, it should download executables built for ARM64 instead of those for x86-64.

earlephilhower, 2022-11-23:
Sorry, for now you need to use Rosetta/etc.  None of the maintainers use Macs, and all tools are cross-built under Linux using a Docker image w/full toolchains set up.

If you, or some other Mac user, can add Linux-hosted cross-building tools for the Apple ARM workflow to https://github.com/earlephilhower/gcc-cross then we can easily add it to the build process.

earlephilhower, 2022-11-23:
On further inspection, looks like OSXCross was updated to later Mac SDKs to include ARM target support.

https://github.com/tpoechtrager/osxcross

The issue now is that by moving the SDK forward, we'll lose support for older Macs stuck on obsolete/EOL versions of MacOS.  Also will need to rebuild a portable Python for ARM...

aleroot, 2022-11-25:
@earlephilhower can't you create a new 4.X version of the tool with the new SDK and compiled for ARM64 ? 
The older Macs could still point to the older branch version 3.X ...

seemywingz, 2023-10-28:
I realize this issue states ***"without rosetta emulation installed"***, but I was able to get this working on a 2022 M2 MacBook Air using rosetta
```
Software:

    System Software Overview:

      System Version: macOS 14.1 (23B74)
      Kernel Version: Darwin 23.1.0
      Boot Volume: Macintosh HD
      Boot Mode: Normal
      Computer Name: MacBook Air
      Secure Virtual Memory: Enabled
      System Integrity Protection: Enabled
      Time since boot: 1 hour, 58 minutes

Hardware:

    Hardware Overview:

      Model Name: MacBook Air
      Model Identifier: Mac14,2
      Model Number: MLY33LL/A
      Chip: Apple M2
      Total Number of Cores: 8 (4 performance and 4 efficiency)
      Memory: 8 GB
      System Firmware Version: 10151.41.12
      OS Loader Version: 10151.41.12
      Activation Lock Status: Enabled

```
### Install Rosetta, HomeBrew , Arduino IDE, and CLI (if not already installed) 
```shell
softwareupdate --install-rosetta --agree-to-license
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
brew install arduino-ide arduino-cli
```
### Install the VSCode Arduino Extension
https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino
### Configure the Arduino Extention to use `arduino-cli`
**Arduino: _Command Path_:** `arduino-cli`

**Arduino: _Path_:** `/opt/homebrew/bin/`

**Arduino: _Use Arduino CLI_:** `checked`

ℹ️ Important! Restart Visual Studio Code

Jason2866, 2026-09-27:
It would be nice to have native Apple ARM toolchains available. With iOS 27 Rosetta needs manually installing. With iOS 28 Rosetta support will not be available anymore.

mcspr, 2026-09-27:
> It would be nice to have native Apple ARM toolchains available. With iOS 27 Rosetta needs manually installing. With iOS 28 Rosetta support will not be available anymore.

^ see above, it is wip. but... funny you should ask, as I am looking into fixing the linux builds for 14.4 right this moment :) 

while the [container builder](https://github.com/earlephilhower/gcc-cross/blob/master/Dockerfile) is not up to date, but I don't see any issues (yet?) while running the produced binaries on the latest Fedora.
macos & win python stuff are probably going to be uv aka astral python standalone prebuilt ones, so no building and not much to figure out on that front

and as much as I like the makefile format, it does need some getting used to after a while