esp8266/Arduino #8727: Exception: Unable to find entry point in file  "C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/bootloaders/eboot/eboot.elf"

shansamp, 2022-11-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```
I am getting the below error sporadically when I try to compile for esp8266.
Exception: Unable to find entry point in file "C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/bootloaders/eboot/eboot.elf"

shansamp, 2022-11-23:
[image]

mcspr, 2022-11-24:
Unable to reproduce. No version of IDE was specified, assuming it is 2.0.2

Quick google of `"Unable to find entry point" esp8266` has some results related to our script, but nothing conclusive.
https://github.com/esp8266/Arduino/issues/7253 *may* be related, but same as above - not something we are able to reproduce (at all; on available hardware, or in CI). Failure in the issue is related to some other .exe, so we are just happen to *sometimes* fail to launch things properly. We are missing `communicate()` call at the very least.
https://docs.python.org/3/library/subprocess.html#subprocess.Popen.communicate
https://github.com/python/cpython/blob/b5b3904f05e77f044f158307bc6bdd2bc1b670a2/Lib/subprocess.py#L506-L573

---

Other than that, make sure that there is `C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed` and you are able to call `C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\bin\xtensa-lx106-elf-readelf.exe` (<kbd>Win</kbd>+<kbd>r</kbd> -> cmd.exe; or, any other terminal app)

You should be able to repeat what elf2bin script tries to do
```
> C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\bin\xtensa-lx106-elf-readelf.exe -h C:\Users\shans\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\bootloaders\eboot\eboot.elf
```
```
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Tensilica Xtensa Processor
  Version:                           0x1
  Entry point address:               0x4010f494
  Start of program headers:          52 (bytes into file)
  Start of section headers:          88296 (bytes into file)
  Flags:                             0x300
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         5
  Size of section headers:           40 (bytes)
  Number of section headers:         17
  Section header string table index: 16
```

panakos, 2022-12-18:
I have the same problem in my desktop. I have tried clean install , deleted all files , rebooted and then installed again. It works with version 2.5.0 but not with any newer including 3.0.2.  It used to work at my desktop untill one day it stopped. Laptop has the same setup but is working fine. No matter what I tried desktop keeps coming back with the same problem. 

Arduino IDE 2.0.3 but same problem had at 2.0.2 and 1.8.19

Dimasikkk1, 2023-01-10:
Same problem. 
`Traceback (most recent call last):
  File "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/elf2bin.py", line 182, in <module>
    sys.exit(main())
  File "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/elf2bin.py", line 166, in main
    to_addr=4096
  File "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/elf2bin.py", line 161, in wrapper
    write_bin(out=out, args=args, **kwargs)
  File "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/elf2bin.py", line 68, in write_bin
    entry = int(get_elf_entry( elf, args.path ))
  File "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/tools/elf2bin.py", line 44, in get_elf_entry
    raise Exception('Unable to find entry point in file "' + elf + '"')
Exception: Unable to find entry point in file "C:\Users\mrdim\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4/bootloaders/eboot/eboot.elf"
exit status 1
`

panakos, 2023-01-10:
If you have Citrix Workspace installed, try updating it or remove it.  That was my problem, I updated it to teh latest version and it fixed it

Dimasikkk1, 2023-01-12:
Updated citrix from 2206 to 2212, didn't help.

panakos, 2023-01-12:
I don't remember which version I have, but I updated it just before Christmas. It's worth trying to uninstall it completely and try