esp8266/Arduino #8731: Should the C++ constructor and destructor tables move into FLASH?

metarutaiga, 2022-11-26:
https://github.com/esp8266/Arduino/blob/fe451a814f0d219c8cc8cbac7735e981f8d46f9c/tools/sdk/ld/eagle.app.v6.common.ld.vtables.h#L15-L26

and

https://github.com/esp8266/Arduino/blob/fe451a814f0d219c8cc8cbac7735e981f8d46f9c/tools/sdk/ld/eagle.app.v6.common.ld.vtables.h#L35-L39

These are all 32bit pointers.
It should move into flash to reduce memory usage.

mcspr, 2022-11-26:
We could just copy-paste those .{c,d}tor entries from the vtables .h .rodata into the .irom0 one? No relation to vtables, just move those to flash at all times

earlephilhower, 2022-11-26:
IIRC they're in RAM now because of IRQ safety.  If any classes are used in IRQ handlers, you can't be sure the flash is accessible so risk crashes.  There was a bug I can't find opened about this after the original memory map shuffling...

mcspr, 2022-11-26:
Arent dtors & ctors (and the whole init array) are supposed to be used for global objects? IRQ code will inject ctor / dtor call when needed, where the actual section can be changed with the IRAM attr

earlephilhower, 2022-11-26:
Good catch, that sounds right.

The `__init_array_start/end` are only used in the CRT initialization (now handled by us in `main.cpp`) https://github.com/esp8266/Arduino/blob/fe451a814f0d219c8cc8cbac7735e981f8d46f9c/cores/esp8266/core_esp8266_main.cpp#L271-L278

BSS start/end, I *think* is used by the ROM code to do the `bzero` but I can't find where we stuff it for the ROM to find right this morning.

I'd say move it at this point.  We're not saving all that much space at this point, but every byte counts...

closed 2023-01-05 by mcspr