esp8266/Arduino #8741: [HTTPS] GET... failed, error: connection failed

bkrajendra, 2022-12-04:
I am using the following code to send data to the server periodically every 1 minute.

```
void sendData(char *h, char *t) {
  const char api_key[] = "myapiKey";
String serve_url = "http://myserver/input/post?";

  char spath[250];
  sprintf(spath, "node=%s&fulljson={\"humidity\":%s,\"temp\":%s}&apikey=%s", clientId.c_str(), h, t, api_key);
  String p = serve_url + spath;
  
  Serial.println("Connecting to: ");
  Serial.println(p);
  HTTPClient https;

  Serial.print("[HTTPS] begin...\n");
  https.begin(client, p);

  Serial.print("[HTTPS] GET...\n");

  int httpCode = https.GET();
  if (httpCode > 0) {

    Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

    if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
      String payload = https.getString();
      Serial.println(payload);
    }
  } else {
    Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
  }

  https.end();
}
```
This works once or twice in a loop, but after that, it fails continuously.
with error:
```
[HTTPS] GET... failed, error: connection failed
```
The same code runs when I use the same URL printed in the console directly in the same code.

What could be wrong here?

Environment:

ESP8266
toll chain SDK "version": "5.100300.220621"
framework: "version": "3.1.0-dev"

Bighoneypot, 2022-12-04:
Dear, you use SSL or not SSL?
Add and configure http.header() in your sketch to connection.

bkrajendra, 2022-12-05:
Ibhaving error for both SSL and non ssl.

What should be http.header()?

Bighoneypot, 2022-12-05:
> I am having error for both SSL and non ssl implementation.
> 
> What should be http.header()?

Dear, I need to read entire sketch.
Another question, in whicn server you need to send your data?

bkrajendra, 2022-12-06:
The issue got resolved with the following approach.
```c++
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include <SPI.h>
#include "Adafruit_Si7021.h"

// #define _TASK_SLEEP_ON_IDLE_RUN
#include <TaskScheduler.h>
bool enableHeater = false;
Adafruit_Si7021 sensor = Adafruit_Si7021();
ESP8266WiFiMulti WiFiMulti;

Scheduler runner;
void dataCollection();
Task cloud_update(5000, TASK_FOREVER, &dataCollection, &runner, true);  // adding task to the chain on creation

WiFiClientSecure client;
// WiFiClient client;
String path = "";
void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("myfi", "mypass");

  if (!sensor.begin()) {
    Serial.println("Did not find Si7021 sensor!");
    while (true)
      ;
  }

  client.setInsecure();

  cloud_update.setInterval(6000);
  runner.startNow();  // set point-in-time for scheduling start
}

void loop() {
  runner.execute();

}

void sendData(String path) {
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, path)) {  // HTTP

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = http.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
    } else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }
}

void formatPath(float t, float h, String& p)
{
  p = "https://mycloudserver/input/post?node=";
  p += "node12";
  p += "&data={\"humidity\":";
  p += String(h);
  p += ",\"temp\":";
  p += String(t);
  p += "}&apikey=hiddenapikey";
}

void dataCollection() {
  float t = sensor.readTemperature();
  delay(200);
  float h = sensor.readHumidity();

  formatPath(t, h, path);
  Serial.println(path);
  sendData(path);

  Serial.print("Humidity:    ");
  Serial.print(h);
  Serial.print("\tTemperature: ");
  Serial.println(t);
}

```