esp8266/Arduino #8768: String variable value is lost when get out from if branch.

Larzu, 2022-12-19:
The following code is losing the string value when it's  used outside of if-statement. Outside of if-statement the "original" variable **payload** is getting back the initial value **x**

esp8266/Arduino  version 3.0.2
Board : LOLIN  D1 mimi v.4.0.0

See the code and the output

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncTCP.h>
#include "LittleFS.h"
#include "time.h"

#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>
---clip---

String httpsGETRequest(const char* serverName) {

    String payload = "x"; 
    String payload1 = "y"; 

     std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    
    client->setFingerprint(fingerprint);
    // Or, if you happy to ignore the SSL certificate, then use the following line instead:
    client->setInsecure();   
    HTTPClient https;
    
    Serial.print("[HTTPS] begin...\n");
    if (https.begin(*client, serverName)) {  // HTTPS

      Serial.print("[HTTPS] GET...\n");
      // start connection and send HTTP header
      int httpCode = https.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          Serial.println("getting payload");
          String payload = https.getString();
          payload1=payload;
          Serial.print ("PRINT payload (inside-if):  ");
          Serial.println(payload);
          payload1=payload;
          Serial.print ("PRINT payload1 (inside-if):  ");
          Serial.println(payload1);**

        }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }
      Serial.print ("PRINT payload (outside-if):  ");
      Serial.println(payload);
      Serial.print ("PRINT payload1 (outside-if):  ");
      Serial.print("print payload1");
      Serial.println(payload1);
      https.end();

      return payload;
    } else {
      Serial.printf("[HTTPS] Unable to connect\n");
    }
}

[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... code: 200
getting payload
**PRINT payload (inside-if):**  {"0":{"time":"2022-12-19 13:00:00","value":"27.36"},"1"-----clip-----
**PRINT payload1 (inside-if)**:  {"0":{"time":"2022-12-19 13:00:00","value":"27.36"},"1"-----clip-----

**PRINT payload (outside-if)**:  x
**PRINT payload1 (outside-if):**  print payload1{"0":{"time":"2022-12-19 13:00:00","value":"27.36"},"1"{-----clip---
</div>

mcspr, 2022-12-19:
Not specific to esp8266, check out what variable scopes are
https://en.cppreference.com/w/cpp/language/scope

`payload` outside of `if` and inside of it are not the same

closed 2022-12-19 by mcspr

Pablo2048, 2022-12-20:
The bug is exactly here:
```cpp
    if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
      Serial.println("getting payload");
     >>> String payload = https.getString(); <<<
```
`String payload =...` create local variable with same name...