esp8266/Arduino #8776: Panic core_esp8266_main.cpp:133 __yield

garudaonekh, 2022-12-23:
Hi,
My code publish data to MQTT Server thru TinyGSM.
`mqttcl->publish(buf,(const uint8_t *)data->cypherbuf_buf,pl,false)`

However, it throw this error:**Panic core_esp8266_main.cpp:133 __yield**

I guess it's because there's a yield which is somehow forbidden. Somehow, I can't figure out where.

Here's my code stacktrace:

`Decoding stack results
0x4020e797: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at /Users/chamroeun/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/functional line 1954
0x401007d6: millis() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x402110d9: __yield() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 135
0x4020fc0f: Stream::timedRead() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Stream.cpp line 39
0x4020ef3c: std::_Function_handler ::_M_invoke(const std::_Any_data &, unsigned int &&) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 423
0x4020fe09: Stream::readBytesUntil(char, char*, unsigned int) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Stream.cpp line 233
0x401007d6: millis() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40202247: TinyGsmModem ::streamGetIntBefore(char) at /Users/chamroeun/Documents/Arduino/libraries/TinyGSM/src/TinyGsmModem.tpp line 307
0x40202da3: TinyGsmSim800::modemGetAvailable(unsigned char) at /Users/chamroeun/Documents/Arduino/libraries/TinyGSM/src/TinyGsmClientSIM800.h line 609
0x4020f324: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x4020f330: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 165
0x402064a1: TinyGsmTCP ::maintainImpl() at /Users/chamroeun/Documents/Arduino/libraries/TinyGSM/src/TinyGsmTCP.tpp line 330
0x401007d6: millis() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40206549: TinyGsmTCP ::GsmClient::available() at /Users/chamroeun/Documents/Arduino/libraries/TinyGSM/src/TinyGsmTCP.tpp line 151
0x40212d5e: TinyGsmTCP ::GsmClient::connected() at /Users/chamroeun/Documents/Arduino/libraries/TinyGSM/src/TinyGsmTCP.tpp line 253
0x40212c38: PubSubClient::connected() at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/PubSubClient.cpp line 694
0x402015d0: PubSubClient::publish(char const*, unsigned char const*, unsigned int, bool) at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/PubSubClient.cpp line 454
0x4021335c: Print::write(char) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Print.h line 75
0x40239741: mqtt_if_output(netif*, pbuf*, ip_addr_t const*) at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/mqttif.cpp line 208
0x4010113b: free(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 398
0x40100acc: vPortFree(void*, char const*, int) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/heap.cpp line 308
0x40226b00: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x401009c1: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/interrupts.h line 29
0x40226b48: ip4_output_if_opt at core/ipv4/ip4.c line 1577
0x40226b6e: ip4_output_if at core/ipv4/ip4.c line 1550
0x40227703: ip_chksum_pseudo at core/inet_chksum.c line 395
0x40222b3a: tcp_output at core/tcp_out.c line 1361
0x40221e25: tcp_create_segment at core/tcp_out.c line 190
0x4022051e: tcp_input at core/tcp_in.c line 771
0x401007d6: millis() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x4022689e: ip4_input at core/ipv4/ip4.c line 1467
0x402397be: mqtt_if_Task(ETSEvent*) at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/mqttif.cpp line 92
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x401009e0: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp line 172
0x401009c1: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/interrupts.h line 29
0x401009c1: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/interrupts.h line 29
0x40100904: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp line 135
0x4020d860: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1096
0x4020d88e: esp8266::MDNSImplementation::MDNSResponder::_udpAppend8(unsigned char) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1112
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x4020d9d1: esp8266::MDNSImplementation::MDNSResponder::_write8(unsigned char, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1254
0x4020dc0d: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSHostDomain(char const*, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1382
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x401004aa: SoftwareSerial::rxBitISR(SoftwareSerial*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 509
0x40100734: ets_post(uint8, ETSSignal, ETSParam) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x401009e0: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp line 172
0x401009c1: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/interrupts.h line 29
0x401009e0: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp line 172
0x40100904: interrupt_handler(void*) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp line 135
0x40101173: malloc(size_t) at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
[... cut ...]
0x40206ddd: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h line 311
0x402071ab: mqtt_loop(mqtt_if_data*) at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/mqttif.cpp line 337
0x40206fd8: loop() at /Users/chamroeun/Documents/Arduino/mqtt_vpn_webserver_tinygsm/mqtt_vpn_webserver_tinygsm.ino line 280
0x402111ac: loop_wrapper() at /Users/chamroeun/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197`

TD-er, 2022-12-23:
My first impression is that it is a large stack.
Are you sure this isn't a stack overflow?

garudaonekh, 2022-12-23:
> My first impression is that it is a large stack. Are you sure this isn't a stack overflow?

Is there any way to increase stack? I will give a try.

 From my search of similar issues, I think it is more likely due to delay or millis().

TD-er, 2022-12-23:
In the stack tracer there are some functions marked with _ISR, which is often a strong indication it might indeed be something like that.
However, you also have a number of interrupt_handler calls in your stack followed by some code which I would not expect to be called from the interrupt code. Interrupt calls should be very short and judging the call stack here, it doesn't seem to be.
Therefore I am not sure about how valid this decoded stack is.
If it is valid, then this code for sure needs some changes.
But then we might need a bit more code to look at than simply a stack dump ;)

bertmelis, 2022-12-23:
You cannot yield in ISRs. If an underlying library uses interrupts, the callbacks of these library obviously also cannot use yield or delay.

garudaonekh, 2022-12-23:
> In the stack tracer there are some functions marked with _ISR, which is often a strong indication it might indeed be something like that. However, you also have a number of interrupt_handler calls [...]

It's a combination of: https://github.com/martin-ger/MQTT_VPN/tree/master/mqtt_vpn_arduino, https://github.com/vshymanskyy/TinyGSM and https://github.com/knolleary/pubsubclient

I have to recode MQTT_VPN to use PubSubClient in order for it to work with TinyGSM. The whole code is too big to share here. The complication here is the tcp/ip protocol stack which I don't quite understand.

garudaonekh, 2022-12-23:
Ok,
I found the place where the error occurred(there's a yield function call of Stream.cpp). But I have no idea how to fix this.

**In  Stream::timedRead() at /Users/xxxx/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Stream.cpp line 39**

```
int Stream::timedRead() {
    int c;
    _startMillis = millis();
    do {
        c = read();
        if(c >= 0)
            return c;
        if(_timeout == 0)
            return -1;
        yield();
    } while(millis() - _startMillis < _timeout);
    return -1;     // -1 indicates timeout
}

```
[image]

mhightower83, 2022-12-24:
This is a pseudo example of how I handle using Stream when yield is not allowed.

```cpp
char buffer[32];
Stream msg;
msg.setTimeout(0);
size_t count = msg.readBytesUntil('\n', buffer, 10);

if (0 == count) {
  // no data
} else if (10 > count) {
  // handle partial data
} else {
  // handle complete message
}
```

garudaonekh, 2022-12-25:
inline int16_t streamGetIntBefore(char lastChar) {
    char   buf[7];
    size_t bytesRead = thisModem().stream.readBytesUntil(
        lastChar, buf, static_cast<size_t>(7));
    // if we read 7 or more bytes, it's an overflow
    if (bytesRead && bytesRead < 7) {
      buf[bytesRead] = '\0';
      int16_t res    = atoi(buf);
      return res;
    }

    return -9999;
  }

> This is a pseudo example of how I handle using Stream when yield is not allowed.
> 
> ```c++
> char buffer[32];
> Stream msg;
> msg.setTimeout(0);
> size_t count = msg.readBytesUntil('\n', buffer, 10);
> 
> if (0 == count) {
>   // no data
> } else if (10 > count) {
>   // handle partial data
> } else {
>   // handle complete message
> }
> ```

Hi, I don't understand your code. The problem is at readBytesUntil itself.

Here's the TinyGSM part that may cause the yield exception:

 ```
 inline int16_t streamGetIntBefore(char lastChar) {
    char   buf[7];
    size_t bytesRead = thisModem().stream.readBytesUntil(
        lastChar, buf, static_cast<size_t>(7));
    // if we read 7 or more bytes, it's an overflow
    if (bytesRead && bytesRead < 7) {
      buf[bytesRead] = '\0';
      int16_t res    = atoi(buf);
      return res;
    }

    return -9999;
  }
```

garudaonekh, 2022-12-26:
I switch to **system_os_task** but still same error. It looks like system_os_task also can't handle yield/delay

mcspr, 2022-12-26:
> I switch to **system_os_task** but still same error. It looks like system_os_task also can't handle yield/delay

In simple terms, we have two tasks: CONT and SYS. CONT is setup() and loop(), SYS is everything else like networking, WiFi driver stuff, timer function execution, etc. In our Core, yield() is a **one-way** switch from CONT task to SYS, hence the panic.

Arduino API is mostly expected to be called from setup() or loop(), so you'd have to check where you happen to be at that point in code. At a glance, I think https://github.com/martin-ger/MQTT_VPN callback happens to be somewhere in SYS, so we can't yield()

garudaonekh, 2022-12-26:
Thanks all, After I queue all the data to publish to MQTT Server and let the loop() to handle it, problem solved.

closed 2022-12-26 by garudaonekh