esp8266/Arduino #8779: String: add debugging message in case of allocation failure

d-a-v, 2022-12-31:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

MacDada, 2022-12-31:
Wow, I did not expect there would be a response o.O Thanks and good luck! <3

MacDada, 2022-12-31:
> Change would be there

Out of curiosity: what kind of change do you have in mind?

1. From my perspective, the best option would be an exception.
* `+` Simple to detect problem, easy to handle "globally".
* `-` But there can not be any exceptions on Arduino :/
2. Second option is to make `replace` return `bool` to signal if it was successful.
* `+` Clear interface
* `-` Could be a little bit backwards incompatible (interface change), but since it returns `void` now, probably nobody really "checks" what it returns RN, so for the new code anybody can just start checking.
*  `-` Check and and handle each and every time it is used might be cumbersome
3. Another way is to add a third parameter to `replace`: a reference bool to check if succeeded
* `+` more BC than above
* `-` even more cumbersome than above (having to create a var in advance on usage)
4. Log info to serial
* `-` You have to be lucky to observe serial monitor and notice
* `+` But actually that's the first thing I do when there is a problem: log to serial everything I suspect might be connected to the issue -> so it would have helped me to spot the problem much faster
5. Any other ideas?

d-a-v, 2023-01-07:
Yes, I think a message in debugging mode is the best we can do when exceptions are not enabled.

PR #8821 (2023-01-26): String: informative messages on issues
- Always enable messages when serial debug port is enabled
- Allocation issue was not reported

fixes #8779

Messages are like:
```
12:14:33.749616204 [String] 'ab1ab2ab3a ... 08ab8309ab': Reallocating large String(48749 -> 48753 bytes)
12:14:33.758096639 [String] 'ab1ab2ab3a ... 11ab8312ab': Reallocating large String(48767 -> 48771 bytes)
12:14:33.766779262 [String] 'ab1ab2ab3a ... 8314ab8315': Reallocating large String(48783 -> 48785 bytes)
12:14:33.774862098 [String] 'ab1ab2ab3a ... 16ab8317ab': Reallocating large String(48797 -> 48801 bytes)
12:14:33. [... cut]
closed 2023-01-26 by mcspr