esp8266/Arduino #8780: DNS lookup problem

brunoudi, 2022-12-31:
### Basic Infos

- [ x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x ] I have searched the issue tracker for a similar issue.
- [ x ] If there is a stack dump, I have decoded it.
- [ x ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[3.0.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I have a separate router and modem. 
After temporary data loss (modem disconnected, internet reconnection, etc), without WiFi connection loss some ESP DNS problem is no longer able to resolve IP addresses.

I have enabled the DEBUG level to WIFI.

### Sketch

```cpp

#include <ESP8266WiFi.h>

const char* ssid     = "my-ssid";
const char* password = "my-password";

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  IPAddress ipResult;
  int result = WiFi.hostByName("www.google.com", ipResult);
  Serial.print("Result: "); Serial.println(result);
  Serial.print("Ip: "); Serial.println(ipResult);
  Serial.println();
  delay(5000);
}

```
Turn on the router WITHOUT DATA (modem connection), and start the ESP.

The output, as expected, is this:

```
[hostByName] request IP for: www.google.com
[hostByName] Host: www.google.com lookup error: -5!
Result: 0
Ip: (IP unset)
```
But after data reestablishing (connecting the router to the modem), ESP still cannot resolve the IP address.

I know there is a DNS cache (I don't know exactly how long it lasts), but I left it running for 24 hours and the output remained the same.

I tried some actions, like reconnecting to WiFi, but is only solves if I restart the device.

mcspr, 2022-12-31:
Why do you think of cache first? You sure networking works at all?

DNS servers are printable
```cpp
Serial.println(WiFi.dnsIP(0));
Serial.println(WiFi.dnsIP(1));
Serial.println(WiFi.dnsIP(2));
```

Same for our own IP, netmask and gateway
```
Serial.println(WiFi.localIP());
Serial.println(WiFi.subnetMask());
Serial.println(WiFi.gatewayIP());
```

I'd expect to see something different when it works and when it doesn't

brunoudi, 2022-12-31:
I'm pretty sure the network starts working fine after I connect the router to modem.

I actually did not thing at DNS cache at first, and I also printed the DNS IP as suggested, but when using dhcp the dns usually is the gateway address. So, the IP didn't change when have or not data connection.

If I turn on the ESP with everything Ok (wifi + data), it solves the IP and print. 

After removing the conection from router to modem, it continuous printing the IP for a while because of DNS cache (not exactly same time every test), and only after some minutes it begins to show error.

It looks like that when it tries to solve the IP and is not successfull, the device never again tries to actually solve the IP and only returns the error from some kind of cache.

mcspr, 2022-12-31:
True, DNS client stores the record based on TTL. My question was - is device actually connected, does anything *but* DNS work?

Error code -5 means IN\_PROGRESS aka it just times out the request
https://github.com/esp8266/Arduino/search?q=ERR_INPROGRESS&type=code
So at least the request is attempted

d-a-v, 2023-01-01:
Network protocol activity can be traced or debugged with [included netdump facility](https://github.com/esp8266/Arduino/blob/3f1d608a19c336f7db0b18ec9faef8e8f38de668/libraries/Netdump/examples/Netdump/Netdump.ino#L108)

d-a-v, 2023-01-05:
@brunoudi a DNS fix was merged.
Can you retry using current master of this core ?

nilo85, 2024-02-25:
I seem to have the same issue.

My device gets ip 192.168.68.XX
GW, DNS etc set to 192.168.68.1
My router is a google nest wifi if that makes a difference.. 

Frustrating =) I tried the both git master and latest stable release with same issue

d-a-v, 2024-02-26:
Can we close this issue ?

nilo85, 2024-02-27:
I’m not the original reporting user and don’t know if you have any rules when to close etc, as I found a valid cause to my issue and original reporting user has not replied within a year, I would feel encouraged to close this one :)