esp8266/Arduino #8800: Multiple Instances of WiFiClientSecure is not working.

mayankbytebeam, 2023-01-12:
### Basic Infos

We are working on something that requires multiple instances of WiFiClientSecure, more specifically we want to connect to a secure mqtt broker and a https server.

#### Platform

- Hardware: ESP-12E
- Core Version: 3.1.0
- Development Env: Arduino IDE 
- Operating System: Windows

### Settings in IDE

- Module: Nodemcu 1.0 (ESP-12E Module)
- Flash Size: 4MB
- lwip Variant: v2 Lower Memory
- CPU Frequency:  80MHz
- Upload Using: SERIAL
- Upload Speed: 115200

### Problem Description

We tried the following code but couldn't get it to work. First client is connected and working perfectly but other is not.

```
#include <Arduino.h>
#include <PubSubClient.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiClientSecureBearSSL.h>
#include <ESP8266httpUpdate.h>

ESP8266WiFiMulti WiFiMulti;

const char* ca_cert = "Root CA Certificate Here";
const char* client_cert = "Client Certificate Here":
const char* client_key = "Client Private Key Here";

PubSubClient PubSub_1;
PubSubClient PubSub_2;
BearSSL::WiFiClientSecure wifi_client_1;
BearSSL::WiFiClientSecure wifi_client_2;

BearSSL::X509List* rootCA_1 = NULL;
BearSSL::X509List* clientCert_1 = NULL;
BearSSL::PrivateKey* clientKey_1 = NULL;

BearSSL::X509List* rootCA_2 = NULL;
BearSSL::X509List* clientCert_2 = NULL;
BearSSL::PrivateKey* clientKey_2 = NULL;

const char* topic = "/tenants/project1/devices/1/actions";

void mqttCallback(char* topic, byte* message, unsigned int length) {
  Serial.println("I am mqttCallback()");

  Serial.print("{topic : ");
  Serial.print(topic);
  Serial.print(", message : ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
  }
  Serial.println("}");
}

void setClock() {
   // Set time via NTP, as required for x.509 validation
  configTime(19800, 0, "pool.ntp.org"); 
 
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {
    delay(500);
    Serial.print(".");
    now = time(nullptr);
  }
  
  struct tm * timeinfo;
  time(&now);
  // Serial.println(now);
  timeinfo = localtime(&now);

  Serial.print("Current time: ");
  Serial.print(asctime(timeinfo));
  Serial.println();
}

void doSomething() {  

    rootCA_1 = new BearSSL::X509List(ca_cert);
    clientCert_1 = new BearSSL::X509List(client_cert);
    clientKey_1 = new BearSSL::PrivateKey(client_key);

    rootCA_2 = new BearSSL::X509List(ca_cert);
    clientCert_2 = new BearSSL::X509List(client_cert);
    clientKey_2 = new BearSSL::PrivateKey(client_key);

    wifi_client_1.setTrustAnchors(rootCA_1);
    wifi_client_1.setClientRSACert(clientCert_1, clientKey_1);

    wifi_client_2.setTrustAnchors(rootCA_2);
    wifi_client_2.setClientRSACert(clientCert_2, clientKey_2);

    PubSub_1.setClient(wifi_client_1);
    PubSub_1.setCallback(mqttCallback);
    PubSub_1.setServer("cloud.bytebeam.io", 8883);

    if(!PubSub_1.connect("Client_1")) {
      Serial.println("error while connecting PubSub_1");
      Serial.printf("PubSub_1 State is %d\n", PubSub_1.state());
    } else {
      Serial.println("PubSub_1 Connected");
    }

    if(!PubSub_1.subscribe(topic)) {
      Serial.println("PubSub_1 subscribe failed");
    } else {
      Serial.println("PubSub_1 subscribed");
    }
    
    PubSub_2.setClient(wifi_client_2);
    PubSub_2.setCallback(mqttCallback);
    PubSub_2.setServer("cloud.bytebeam.io", 8883);

    if(!PubSub_2.connect("Client_2")) {
      Serial.println("error while connecting PubSub_2");
      Serial.printf("PubSub_2 State is %d\n", PubSub_2.state());
    } else {
      Serial.println("PubSub_2 Connected");
    }

    if(!PubSub_2.subscribe(topic)) {
      Serial.println("PubSub_2 subscribe failed");
    } else {
      Serial.println("PubSub_2 subscribed");
    }

    Serial.println();

}

void setup() {

  Serial.begin(9600);
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("Ok, wifi is connected");
  }

  setClock();
  doSomething();
}

void loop() {
  
  if(!PubSub_1.loop()) {
    Serial.println("Pubsub_1 loop failed");
  } else {
    Serial.println("Pubsub_1 looped");
  }

  if(!PubSub_2.loop()) {
    Serial.println("Pubsub_2 loop failed");
  } else {
     Serial.println("Pubsub_2 looped");
  }

  delay(2000);
}
```

### Serial Logs

Ok, wifi is connected
.Current time: Thu Jan 12 22:11:09 2023

PubSub_1 Connected
PubSub_1 subscribed
error while connecting PubSub_2
PubSub_2 State is -2
PubSub_2 subscribe failed

Pubsub_1 looped
Pubsub_2 loop failed
Pubsub_1 looped
Pubsub_2 loop failed
Pubsub_1 looped
Pubsub_2 loop failed
Pubsub_1 looped
Pubsub_2 loop failed

Any help would be greatly appreciated!

Mayank

earlephilhower, 2023-01-12:
You're out of memory.

Each SSL connection requires a contiguous buffer of 16KB to hold the encrypted data before decryption.  There is also a ~7KB additional stack required to run the SSL code that comes out of heap as well.  Fitting 2 connection states in RAM isn't possible by default.

Look at the MFLN example for a way of shrinking the buffers required, but note that it requires support on the remote server end as well.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength/BearSSL_MaxFragmentLength.ino

closed 2023-01-12 by earlephilhower

earlephilhower, 2023-01-12:
All ESP8266 have the same amount of on-chip RAM, so without doing extraordinary things (i.e. adding an external (P)SRAM chip and using the virtual memory mode) it's not really possible.

Note that MFLN will "work," even if unsupported at the server end, as long as any message/packet sent across fits within the allocated buffer.  I..e if you can guarantee never sending or receiving packets > 7KB, then you can set an 8KB MFLN size and it will run fine.  But, as soon as a message grows beyond the MFLN buffer the connection will fail.

mayankbytebeam, 2023-01-13:
Thank you so much @earlephilhower for the clear insights :)

mayankbytebeam, 2023-01-13:
### Problem Description

Hi @earlephilhower, as per your suggestion by changing the buffers size we are able to setup 2 clients, but I observed when the second client is getting connected the first client loses its connectivity. can you please help here?   

#include <Arduino.h>
#include <time.h>
#include <ESP8266WiFiMulti.h>
#include <PubSubClient.h>
#include <WiFiClientSecureBearSSL.h>
#include <ESP8266httpUpdate.h>

ESP8266WiFiMulti WiFiMulti;

char ca_cert[] = "Root CA Cert";
char client_cert[] = "Client Cert";
char client_key[] = "Client Key";

PubSubClient PubSub_1;
BearSSL::WiFiClientSecure wifi_client_1;
BearSSL::WiFiClientSecure wifi_client_2;

BearSSL::X509List* rootCA_1 = NULL;
BearSSL::X509List* clientCert_1 = NULL;
BearSSL::PrivateKey* clientKey_1 = NULL;

const char* topic = "/tenants/project1/devices/1/actions";

bool updateFlag = false;

void mqttCallback(char* topic, byte* message, unsigned int length) {
  Serial.println("I am mqttCallback()");

  Serial.print("{topic : ");
  Serial.print(topic);
  Serial.print(", message : ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
  }
  Serial.println("}");

  updateFlag = true;
}

// function to sync time from ntp server with predefined credentials
void setClock() {
   // Set time via NTP, as required for x.509 validation
  configTime(19800, 0, "pool.ntp.org"); 
 
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {
    delay(500);
    Serial.print(".");
    now = time(nullptr);
  }
  
  struct tm * timeinfo;
  time(&now);
  // Serial.println(now);
  timeinfo = localtime(&now);

  Serial.print("Current time: ");
  Serial.print(asctime(timeinfo));
}

void doSomething() {  
    Serial.printf("Free Heap : %d\n", ESP.getFreeHeap());    

    rootCA_1 = new BearSSL::X509List(ca_cert);
    clientCert_1 = new BearSSL::X509List(client_cert);
    clientKey_1 = new BearSSL::PrivateKey(client_key);
[... 145 lines cut ...]
Pubsub_1 looped
CALLBACK:  HTTP update process started
CALLBACK:  HTTP update process at 0 of 449280 bytes...
publish failed
CALLBACK:  HTTP update process at 0 of 449280 bytes...
CALLBACK:  HTTP update process at 4096 of 449280 bytes...
CALLBACK:  HTTP update process at 8192 of 449280 bytes...
CALLBACK:  HTTP update process at 12288 of 449280 bytes...
CALLBACK:  HTTP update process at 16384 of 449280 bytes...
CALLBACK:  HTTP update process at 20480 of 449280 bytes...
CALLBACK:  HTTP update process at 24576 of 449280 bytes...
CALLBACK:  HTTP update process at 28672 of 449280 bytes...
CALLBACK:  HTTP update process at 32768 of 449280 bytes...
CALLBACK:  HTTP update process at 36864 of 449280 bytes...
CALLBACK:  HTTP update process at 40960 of 449280 bytes...
CALLBACK:  HTTP update process at 45056 of 449280 bytes...
publish failed
CALLBACK:  HTTP update process at 49152 of 449280 bytes...
CALLBACK:  HTTP update process at 53248 of 449280 bytes...
CALLBACK:  HTTP update process at 57344 of 449280 bytes...
```

Always Thankful for any help!

Mayank

earlephilhower, 2023-01-13:
I'd enable full debugging and rerun and see what spits out.

You've got a lot going on w/both the sketch and the 3rd party library.  If the issue is in the core then you're probably hitting an OOM error condition (i.e. shrink those buffers even more if you can, or look at the add'l IRAM heap configs) or a timeout on one or the other because SSL is very slow on the 8266 (make sure you're at 160MHz!).

mayankbytebeam, 2023-01-13:
I really appreciate you help.

We tried with 2K and 4K buffer sizes with 80 MHZ CPU. We will update the CPU to 160MHz.

earlephilhower, 2023-01-13:
>  I tried with 2k buffer size with 160MHz CPU but couldn't make it work.

My guess is that the buffer shrink is causing the failure.  AS I said, the receiver needs to keep a copy of the full encrypted message in RAM and can't decode it unless it's all there.  So, if you got a response from the other end that was larger than your buffer the TLS connection would have to fail.  (With MFLN, the server promises never to send messages larger than whatever was specified so this isn't a problem there...the server will break things up properly to make it happen.)

I'd go up to 8K and try again at 160MHz.  There might be enough space for 2 8K buffers + other stuff in the 8266 RAM.  You could also look at the add'l heap options to eek out a few more (slow) KB.

mayankbytebeam, 2023-01-14:
I really disagree on the point that buffer shrink is causing the issue because I tested both the clients separately with 2k buffer size and 160MHZ CPU frequency. Therefore, it's hard to believe that server response is larger than the buffer size. can you check for the possibility of error in the core with 2k and 4k buffer sizes because we are building for the community, and we really want to optimize the things here as much as possible.

Also, I really don't know much about add heap options, I will be happy if you can share some relevant reference.

d-a-v, 2023-01-14:
> Also, I really don't know much about add heap options, I will be happy if you can share some relevant reference.

https://arduino-esp8266.readthedocs.io/en/latest/mmu.html

mayankbytebeam, 2023-01-14:
> > Also, I really don't know much about add heap options, I will be happy if you can share some relevant reference.
> 
> https://arduino-esp8266.readthedocs.io/en/latest/mmu.html

Thanks @d-a-v