esp8266/Arduino #8801: OTA upload fails with SDK 3.0.5

hmueller01, 2023-01-13:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266 device]
- Core Version:			[latest git (3.0.5(b29dcd3), framework-arduinoespressif8266 @ 3.2.0-dev+sha.840ef78)]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[other]
- Flash Mode:			[qio]
- Flash Size: 		[2MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]

### Problem Description

Upload starts and gets slower after some percent. Finally it stops and breaks with an error.

```
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = espota
Uploading .pio/build/upstream_develop_upload/firmware.bin
18:26:45 [DEBUG]: Options: {'esp_ip': '192.168.x.x', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 58294, 'auth': '', 'image': '.pio/build/upstream_develop_upload/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
18:26:45 [INFO]: Starting on 0.0.0.0:58294
18:26:45 [INFO]: Upload size: 545696
18:26:45 [INFO]: Sending invitation to: 192.168.x.x
18:26:45 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [=                                                           ] 1% 
...
Uploading: [===========================================                 ] 71% 
Uploading: [===========================================                 ] 72% 
Uploading: [============================================                ] 72% 
Uploading: [============================================                ] 72% 
Uploading: [============================================                ] 73% 
18:27:08 [ERROR]: Error Uploading
*** [upload] Error 1
```
On Serial Monitor I see
```
Start updating sketch
Progress: 75%
Error[4]: End Failed
```

Doing the same thing with SDK 2.2.2-dev(38a443e) works reliable.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
 ...
  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else {  // U_FS
      type = "filesystem";
    }

    // NOTE: if updating FS this would be the place to unmount FS using FS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("\nError[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
}

void loop() {
    ...
    ArduinoOTA.handle();
}

```

earlephilhower, 2023-01-13:
Can you please run this with full debugging enabled and capture the `Serial` output?  That should provide a more complete picture of what's happening.

mcspr, 2023-01-13:
> Hardware: [ESP8266 device]
> Flash Mode: [qio]
> Flash Size: [2MB]

Also, what device is it?
`printf("%08X\n", ESP.getFlashChipId());`?

At a minimum, we need `-DDEBUG_ESP_UPDATER` and `-DDEBUG_ESP_PORT=Serial`
https://github.com/esp8266/Arduino/blob/840ef782371d2c0310aa9cbde8a311fd8b80bf28/cores/esp8266/Updater.h#L29-L33
(https://docs.platformio.org/en/stable/platforms/espressif8266.html#debug-level)

mhightower83, 2023-01-13:
Not to be forgotten when changing SDKs, you should erase the WiFi Settings. Commonly covered by selecting Arduino IDE Tools->Erase Flash: "Sketch + WiFi Settings" once with a change of SDK version.

hmueller01, 2023-01-13:
Ok, it's a Shelly1 what I am using.
`FlashChipId  : 0015605E`

Board config file `shelly1.json` (created by me in my boards folder):
```json
{
  "build": {
    "arduino": {
      "ldscript": "eagle.flash.2m.ld"
    },
    "core": "esp8266",
    "extra_flags": "-DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_SHELLY1",
    "f_cpu": "80000000L",
    "f_flash": "40000000L",
    "flash_mode": "qio",
    "mcu": "esp8266",
    "variant": "nodemcu"
  },
  "connectivity": [
    "wifi"
  ],
  "frameworks": [
    "arduino",
    "esp8266-rtos-sdk",
    "esp8266-nonos-sdk"
  ],
  "name": "Shelly1",
  "upload": {
    "maximum_ram_size": 81920,
    "maximum_size": 2097152,
    "require_upload_port": true,
    "resetmethod": "nodemcu",
    "speed": 115200
  },
  "url": "http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family",
  "vendor": "Espressif"
}
```
I will provide more details later.

hmueller01, 2023-01-13:
> Not to be forgotten when changing SDKs, you should erase the WiFi Settings. Commonly covered by selecting Arduino IDE Tools->Erase Flash: "Sketch + WiFi Settings" once with a change of SDK version.

Is `esptool.py erase_flash` sufficient too?

mhightower83, 2023-01-13:
> Is esptool.py erase_flash sufficient too?

Yes, that should work. It is the same as running Arduino IDE Tools->Erase Flash: "All Flash Contents".

hmueller01, 2023-01-13:
Ok, this time I see an Exception:

```
sleep disable
[begin] roundedSize:       0x00086000 (548864)
[begin] updateEndAddress:  0x001F0000 (2031616)
[begin] currentSketchSize: 0x00086000 (548864)
[begin] _startAddress:     0x0016A000 (1482752)
[begin] _currentAddress:   0x0016A000 (1482752)
[begin] _size:             0x00085F30 (548656)
Start updating sketch
Header: 0xE9 2 0 30
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec20 end: 3fffffb0 offset: 0190
3fffedb0:  000000d4 3ffef3a8 000000d4 401013bc  
3fffedc0:  000000e0 40105b87 3ffee308 ffffffff  
3fffedd0:  40256d8b 3ffef3a8 00000002 40256d20  
3fffede0:  00000002 40256cd7 00000002 00000001  
3fffedf0:  40255e2d 3fffee90 3ffef254 3ffeebc4  
3fffee00:  4025385f 3fffee90 3ffef254 3ffeebc4  
3fffee10:  75652040 00006d68 7fffffff 00000001  
3fffee20:  00000001 00000080 3fffb060 40100dfa  
3fffee30:  4024fa39 3ffeb150 05b10b26 3ffeb120  
3fffee40:  3ffeca8c 3ffeed4c ffffffce 3ffeca9c  
3fffee50:  3ffee308 0000002e 00000000 4025451f  
3fffee60:  00000000 3fff37ec ffffffce 00000000  
3fffee70:  00000000 00000030 c224c8c5 ffffffff  
3fffee80:  40000f58 00000000 0000c8c5 00000000  
3fffee90:  00000000 14310101 00640100 00000045  
3fffeea0:  3ffecab0 00000169 3ffecac5 3ffecaa4  
3fffeeb0:  00000000 3ffecab0 3ffecab8 3ffecb0d  
3fffeec0:  00000000 00000000 3ffecad9 3ffecaf3  
3fffeed0:  3ffecb1a 3ffecb36 00000000 3ffecbdd  
3fffeee0:  00000000 00000000 0000002e 00000000  
3fffeef0:  3fff3be4 402540e0 3ffee308 3fff37ec  
3fffef00:  00000000 3ffeca8c 3ffee308 3ffeca8c  
3fffef10:  00000191 3ffef3a8 00000000 0000002e  
3fffef20:  00000191 3ffeca96 00000000 40269333  
3fffef30:  3ffee308 3ffeca80 3fffdcc0 3ffea978  
3fffef40:  00000080 00000080 00000000 3fffdab0  
3fffef50:  40268bf7 3fffdab0 00000000 402195ea  
3fffef60:  3ffea978 40000f49 ffff0081 40000f49  
3fffef70:  40000e19 00085f2c bff00000 0000bfff  
3fffef80:  00000005 40106af1 00085f2c bff00000  
3fffef90:  40106af7 fbadf9ce 198e343a 50406ff1  
3fffefa0:  4010000d bff00000 00085f2c 401000ab  
3fffefb0:  4025d580 3fffef4c 4025d531 3ffffea8  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
[... 231 lines cut ...]
3fffff00:  3fff37ec 4020d840 40223930 1702a8c0  
3fffff10:  40223930 1702a8c0 3fff13cc 4020be97  
3fffff20:  3fff052c 00000218 3fff15a4 4020fa03  
3fffff30:  00000000 3fff0050 00000000 3fff0814  
3fffff40:  3fffdad0 00000000 3fff04a8 40211b88  
3fffff50:  3fffdad0 00000000 3fff00cc 4020c051  
3fffff60:  3fffdad0 402232c4 3fff00d0 00000001  
3fffff70:  3fff0698 3fff0694 00000001 40216bac  
3fffff80:  00000000 3fff984c 00000020 3fff0814  
3fffff90:  00000005 3fff972c 3fff00e4 3fff0814  
3fffffa0:  3fffdad0 00000000 3fff07e8 3fff0814  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
�@*r��P���K�t�dr�,@��.�Ѫ�Ko8a ��*�T������P�-���*����nzAl )�m~�P'!�(P�+t׬X@j|����*���r�(�8x�vA,PZ�r�(�+N�P���l /Ѣ�boot not set
ota1 not set
ota2 not set
V2
Mo
,�rf cal sector: 508
freq trace enable 0
rf[112] 
SDK:3.0.5(b29dcd3)/Core:3.1.0-8-g840ef78=30100008/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
```
[dump-20230113-decoded.txt](https://github.com/esp8266/Arduino/files/10414787/dump-20230113-decoded.txt)

hmueller01, 2023-01-20:
This is really strange.
I **never** had problems with ESPhttpUpdate. Only ArduinoOTA was causing it.
In the meantime I updated to platformio/espressif8266 @ 4.1.0 and latest framework-arduinoespressif8266 @ 3.2.0-dev+sha.bed2fa3 and all of a sudden I can't reproduce the problem. Closing this and hoping it won't come again. ;-)

closed 2023-01-20 by hmueller01