esp8266/Arduino #8803: Compile failed using board manager 3.1.0 of ESP8266

kagouraki, 2023-01-14:
Hello,

- i installed the sloeber IDE bundle fresh on linux.
- i added the link `https://arduino.esp8266.com/stable/package_esp8266com_index.json` to the third party tab
- i installed through the sloeber preferences the esp8266 3.1.0 
- when i try to compile the blink example i get this error

```
18:46:06 **** Incremental Build of configuration Release for project testesp8266 ****
make all 
Building file: ../sloeber.ino.cpp
Starting C++ compile
"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 "@/home/labcnc/eclipse-workspaceNew/testesp8266/Release/core/build.opt" "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/tools/sdk/include" "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/tools/sdk/lwip2/include" "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/tools/sdk/libc/xtensa-lx106-elf/include" "-I/home/labcnc/eclipse-workspaceNew/testesp8266/Release/core" -c "@/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/tools/warnings/extra-g++"  -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DARDUINO_BOARD_ID="" -DLED_BUILTIN=2 -DFLASHMODE_DOUT     -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/cores/esp8266" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.0/variants/generic" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
xtensa-lx106-elf-g++: error: /home/labcnc/eclipse-workspaceNew/testesp8266/Release/core/build.opt: No such file or directory
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

18:46:06 Build Failed. 2 errors, 0 warnings. (took 609ms)
```
on the arduino IDE i make the exact same steps for installing the 3.1.0 board manager and i tried again to compile the blink sketch. The results are these
```
/home/labcnc/Programs/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/labcnc/Programs/arduino-1.8.13/hardware -hardware /home/labcnc/.arduino15/packages -tools /home/labcnc/Programs/arduino-1.8.13/tools-builder -tools /home/labcnc/Programs/arduino-1.8.13/hardware/tools/avr -tools /home/labcnc/.arduino15/packages -built-in-libraries /home/labcnc/Programs/arduino-1.8.13/libraries -libraries /home/labcnc/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10813 -build-path /tmp/arduino_build_276859 -warnings=default -build-cache /tmp/arduino_cache_475570 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/labcnc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=/home/labcnc/.arduino15/packages/esp8266/tools/mklittlefs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/mklittlefs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.python3.path=/home/labcnc/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/labcnc/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=/home/labcnc/.arduino15/packages/esp8266/tools/mkspiffs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/mkspiffs/3.1.0-gcc10.3-e5f9fec -verbose /home/labcnc/Programs/arduino-1.8.13/examples/01.Basics/Blink/Blink.ino
/home/labcnc/Programs/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/labcnc/Programs/arduino-1.8.13/hardware -hardware /home/labcnc/.arduino15/packages -tools /home/labcnc/Programs/arduino-1.8.13/tools-builder -tools /home/labcnc/Programs/arduino-1.8.13/hardware/tools/avr -tools /home/labcnc/.arduino15/packages -built-in-libraries /home/labcnc/Programs/arduino-1.8.13/libraries -libraries /home/labcnc/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10813 -build-path /tmp/arduino_build_276859 -warnings=default -build-cache /tmp/arduino_cache_475570 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/labcnc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs.path=/home/labcnc/.arduino15/packages/esp8266/tools/mklittlefs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mklittlefs-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/mklittlefs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.python3.path=/home/labcnc/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/labcnc/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=/home/labcnc/.arduino15/packages/esp8266/tools/mkspiffs/3.1.0-gcc10.3-e5f9fec -prefs=runtime.tools.mkspiffs-3.1.0-gcc10.3-e5f9fec.path=/home/labcnc/.arduino15/packages/esp8266/tools/mkspiffs/3.1.0-gcc10.3-e5f9fec -verbose /home/labcnc/Programs/arduino-1.8.13/examples/01.Basics/Blink/Blink.ino
Using board 'generic' from platform in folder: /home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0
Using core 'esp8266' from platform in folder: /home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0
/home/labcnc/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 /home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0/tools/mkbuildoptglobals.py /home/labcnc/Programs/arduino-1.8.13 10813 /tmp/arduino_build_276859 /tmp/arduino_build_276859/core/build.opt /home/labcnc/Programs/arduino-1.8.13/examples/01.Basics/Blink/Blink.ino.globals.h /home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0/cores/esp8266/CommonHFile.h
Traceback (most recent call last):
  File "/home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0/tools/mkbuildoptglobals.py", line 846, in <module>
    sys.exit(main())
  File "/home/labcnc/.arduino15/packages/esp8266/hardware/esp8266/3.1.0/tools/mkbuildoptglobals.py", line 759, in main
    if time.time_ns() < os.stat(commonhfile_fqfn).st_mtime_ns:
AttributeError: module 'time' has no attribute 'time_ns'
exit status 1
Error compiling for board Generic ESP8266 Module.
```

the 3.0.2 version of the boards manager works perfectly on both sloeber and arduino IDE

mcspr, 2023-01-15:
> on the arduino IDE i make the exact same steps for installing the 3.1.0 board manager and i tried again to compile the blink sketch. The results are these

https://github.com/esp8266/Arduino/pull/8804 should fix that, assuming the issue is with locally-installed python <3.7
(which it looks like based on the error message)

What happens with Sloeber is whole other thing. Searching in the issue tracker there, there is a mention of stm32duino (which is similar to the implementation of build options here)
* https://github.com/Sloeber/arduino-eclipse-plugin/issues/1143 

But, exception for stm32duino in tests is removed in the current version... so, it works? I wonder what our differences, then.

JAndrassy, 2023-01-15:
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1546

JAndrassy, 2023-01-15:
workaround. in platform.sloeber.txt change `recipe.hooks.prebuild.2.pattern` to `recipe.hooks.prebuild.1.pattern`

jantje, 2023-01-15:
Note that if you do the workaround as described above you also need to do the following steps to clear the caching
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533

jantje, 2023-01-16:
After more investigation in this issue my proposed workaround is to comment out the following line in the platform.txt and execute the cache clearing steps https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533
`build.opt.flags="@{build.opt.fqfn}"`

This is a better solution because Sloeber uses the build command for discovery purposes. As the discovery is triggered early on it is very unlikely the build has run before the discovery runs. Therefore the opt file is missing and the discovery fails.
If one of your project uses build options you can add the "@{by build opt file}" to the build options in the project properties->sloeber pages.
If you point to your real file there is no need for a prebuild copying action. Still both the discovery and build will work and take it into account.
Note you will have to do a manual clean if you change the opt file.

closed 2023-01-22 by mcspr

jantje, 2023-01-22:
I'm glad to see this got fixed