esp8266/Arduino #8809: VS Code reports language server error related to generated compilerPath in c_cpp_properties.json by Arduino extension

milanbx, 2023-01-15:
## Problem
After upgrading **ESP8266/Arduino core** to **v3.1.1**, the extension regenerated th `c_cpp_properties.json` file with a `compilerPath` attribute having value causing **C/C++ configuration error**:
`Unable to resolve configuration with compilerPath: "@D:\Portable\Arduino\portable\packages\esp8266\hardware\esp8266\3.1.1\tools\warnings\default-g++"`
On top of this line from `C/C++ Configuration Warning` output, the Code UI shows 5 instances of `cpptools` output, all having same content (see also snapshot below):
`[Error - 18:20:27] Connection to server got closed. Server will not be restarted.`
[image]
Then, the intellisense stops warking at all, co cross reference can be followed (Ctrl+left mouse click) to see details of function or variable declaration etc.

## Environment
### Code
Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T16:59:02.252Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No

### Extensions
Arduino: 0.4.12
C/C++ Extension Pack: 1.3.0

## Steps to reproduce
Just run VS Code, open any sketch with Arduino code (*.ino), even empty one. Popup error start to come up in the lower right corner - see snapshot:
[image]

## Code snippet
Every, does not matter, even this simple blinking LED example:
``` cpp
byte go = LOW;
void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, go);
}
void loop() {
    delay(500);
    go = 1 - go;
    digitalWrite(LED_BUILTIN, go);
}
```

## Dev Tools Console Log
see attached file please
[vscode-app-1673804966418.log](https://github.com/microsoft/vscode-arduino/files/10420833/vscode-app-1673804966418.log)

Bighoneypot, 2023-01-15:
Use direct LOW and not byte go= LOW;

mcspr, 2023-01-17:
Also reported at https://github.com/microsoft/vscode-arduino/issues/1567

I wonder if changing `"@{compiler.warning_flags}"` to `@"{compiler_warning.flags}"` in compiler properties would help Intellisense out, unless it really does not understand `@file-path` syntax.

i.e.
```diff
diff --git a/platform.txt b/platform.txt
index 120cfb00..9ca484fb 100644
--- a/platform.txt
+++ b/platform.txt
@@ -22,11 +22,11 @@ runtime.tools.mkdir={runtime.platform.path}/tools/mkdir.py
 runtime.tools.cp={runtime.platform.path}/tools/cp.py
 runtime.tools.eboot={runtime.platform.path}/bootloaders/eboot/eboot.elf

-compiler.warning_flags=@{runtime.platform.path}/tools/warnings/none
-compiler.warning_flags.none=@{runtime.platform.path}/tools/warnings/none
-compiler.warning_flags.default=@{runtime.platform.path}/tools/warnings/default
-compiler.warning_flags.more=@{runtime.platform.path}/tools/warnings/more
-compiler.warning_flags.all=@{runtime.platform.path}/tools/warnings/extra
+compiler.warning_flags={runtime.platform.path}/tools/warnings/none
+compiler.warning_flags.none={runtime.platform.path}/tools/warnings/none
+compiler.warning_flags.default={runtime.platform.path}/tools/warnings/default
+compiler.warning_flags.more={runtime.platform.path}/tools/warnings/more
+compiler.warning_flags.all={runtime.platform.path}/tools/warnings/extra

 build.lwip_lib=-llwip_gcc
 build.lwip_include=lwip/include
@@ -76,18 +76,18 @@ compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_S
 compiler.libraries.ldflags=

 compiler.c.cmd=xtensa-lx106-elf-gcc
-compiler.c.flags=-c "{compiler.warning_flags}-gcc" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.c.flags=-c @"{compiler.warning_flags}-gcc" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.S.cmd=xtensa-lx106-elf-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls "-I{runtime.tools.xtensa-lx106-elf-gcc.path}/include/"

-compiler.c.elf.flags=-g "{compiler.warning_flags}-gcc" -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read
+compiler.c.elf.flags=-g @"{compiler.warning_flags}-gcc" -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read

 compiler.c.elf.cmd=xtensa-lx106-elf-gcc
 compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 {build.stdcpp_lib} -lm -lc -lgcc

 compiler.cpp.cmd=xtensa-lx106-elf-g++
-compiler.cpp.flags=-c "{compiler.warning_flags}-g++" {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.cpp.flags=-c @"{compiler.warning_flags}-g++" {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.as.cmd=xtensa-lx106-elf-as
```

Unable to reproduce, still. VSCode, Arduino extension and C++ pack does not want to enable Arduino language mode. Maybe something missing... are there any extra settings you have applied? In VSCode .json, or the sketch one.
```json
{
    "arduino.useArduinoCli": true,
    "arduino.path": "C:\\Program Files\\Arduino CLI",
    "C_Cpp.loggingLevel": "Debug",
    "arduino.commandPath": "arduino-cli.exe",
    "arduino.logLevel": "verbose"
}
```

milanbx, 2023-01-18:
Hi  @mscpr and thanks for your answer. Regarding the "@" prefix in `cimpilerPath`, here is some more info: https://github.com/microsoft/vscode-cpptools/issues/10385#issuecomment-1386343166
And here is the content of my `arduino.json`, which is almost the same for month now and used in other my projects too, so thus may not be the cause:
``` json
    "port": "COM3",
    "configuration": "xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600",
    "board": "esp8266:esp8266:generic",
    "sketch": "zaluzieRF.ino",
    "output": "d:\\VSBuildTemp"
```

mcspr, 2023-01-18:
Thanks for the link! We indeed have a valid syntax here, I was just wondering about how should we place `@` - just before `"` or keep it within quotes.

Intellisense generation you are seeing somehow misses the fact that actual compiler path to g++ is not our warning file, but the path just before.
e.g. for me it is at `"C:\Users\maxim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\bin\xtensa-lx106-elf-g++.exe`

I was hoping to test some things with Intellisense myself, but consistently get a very nondescript message. As it seems to just stay in Win32 (cl.exe) mode and failing, Arduino extension never even tries to create `c_cpp_properties.json`
> [Warning] Failed to generate IntelliSense configuration.
[Done] Analyzing sketch 'arduino8809.ino'

milanbx, 2023-01-19:
Are you saying, that the `c_cpp_properties.json` is populated by C/C++ extension?
I am asking particulary because the fact, that when I change the path to compiler manualy to `d:\Portable\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\bin\xtensa-lx106-elf-g++.exe `, next run of rebuilding intellisense info chagnes the path back to the one that causes the error.
So, question is, who rewrites it?

milanbx, 2023-01-19:
I have used vanilla Code portable installation v1.74.3 with only Arduino extension, which requires C/C++ extension v1.13.9, without ESP8266 or any other aditional board, and empty sketch configured for Arduino Uno (on Windows 11), and the bug did not appear any more. It meand, the bug has to do something with ESP8266 Arduino core.

mcspr, 2023-01-20:
> 

I would assume so - command palette has Arduino: action for Intellisense. Same thing mentioned in settings.

Have you tried the patch above? `"@{compiler blah}"` -> `@"{compiler blah}"`, to see whether its the quotes that are a problem.

milanbx, 2023-01-20:
well I did try it but every run of a compiler rewrites the line again

mcspr, 2023-01-20:
> well I did try it but every run of a compiler rewrites the line again

`D:\Portable\Arduino\portable\packages\esp8266\hardware\esp8266\3.1.1\platform.txt` should not be modified by extension, it is our file

milanbx, 2023-01-20:
I have applied all changes as you have described in the patch 2 days ago, and it didnot helped. See console output:
```
[Starting] Analyzing sketch 'xxx.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Error while detecting libraries included by C:\Users\milan\AppData\Local\Temp\arduino_build_145046\sketch\xxx.ino.cpp
xtensa-lx106-elf-g++: error: "D:\Portable\Arduino\portable\packages\esp8266\hardware\esp8266\3.1.1/tools/warnings/none-g++": Invalid argument
exit status 1
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'xxx.ino': Exit with code=1
```

milanbx, 2023-01-20:
I have downgraded ro 3.0.2, where all works fine and made a diff:
``` diff
--- platform302.txt	2021-07-27 00:35:15.000000000 +0200
+++ platform311.txt	2023-01-20 13:22:55.335208900 +0100
@@ -5,8 +5,8 @@
 # For more info:
 # https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification
 
-name=ESP8266 Boards (3.0.2)
-version=3.0.2
+name=ESP8266 Boards (3.1.1)
+version=3.1.1
 
 # These will be removed by the packager script when doing a JSON release
 
@@ -17,15 +17,16 @@
 runtime.tools.elf2bin={runtime.platform.path}/tools/elf2bin.py
 runtime.tools.sizes={runtime.platform.path}/tools/sizes.py
 runtime.tools.makecorever={runtime.platform.path}/tools/makecorever.py
+runtime.tools.mkbuildoptglobals={runtime.platform.path}/tools/mkbuildoptglobals.py
 runtime.tools.mkdir={runtime.platform.path}/tools/mkdir.py
 runtime.tools.cp={runtime.platform.path}/tools/cp.py
 runtime.tools.eboot={runtime.platform.path}/bootloaders/eboot/eboot.elf
 
-compiler.warning_flags=-w -Werror=return-type
-compiler.warning_flags.none=-w -Werror=return-type
-compiler.warning_flags.default=-Werror=return-type
-compiler.warning_flags.more=-Wall -Werror=return-type
-compiler.warning_flags.all=-Wall -Wextra -Werror=return-type
+compiler.warning_flags=@{runtime.platform.path}/tools/warnings/none
+compiler.warning_flags.none=@{runtime.platform.path}/tools/warnings/none
+compiler.warning_flags.default=@{runtime.platform.path}/tools/warnings/default
+compiler.warning_flags.more=@{runtime.platform.path}/tools/warnings/more
+compiler.warning_flags.all=@{runtime.platform.path}/tools/warnings/extra
 
 build.lwip_lib=-llwip_gcc
 build.lwip_include=lwip/include
@@ -52,25 +53,41 @@
 #build.sdk=NONOSDK22x_191024
 #build.sdk=NONOSDK22x_191105
 
+# These are not overriden when FS is not configured
+build.eeprom_start=
+build.spiffs_start=
+build.spiffs_end=
+build.spiffs_blocksize=
+
+# Fully qualified file names for processing sketch global options
+globals.h.source.fqfn={build.source.path}/{build.project_name}.globals.h
+commonhfile.fqfn={build.core.path}/CommonHFile.h
+build.opt.fqfn={build.path}/core/build.opt
+build.opt.flags="@{build.opt.fqfn}"
+mkbuildoptglobals.extra_flags=
+
 compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/
 compiler.sdk.path={runtime.platform.path}/tools/sdk
 
 compiler.libc.path={runtime.platform.path}/tools/sdk/libc/xtensa-lx106-elf
-compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-I{compiler.sdk.path}/include" "-I{compiler.sdk.path}/{build.lwip_include}" "-I{compiler.libc.path}/include" "-I{build.path}/core"
+compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 {build.opt.flags} "-I{compiler.sdk.path}/include" "-I{compiler.sdk.path}/{build.lwip_include}" "-I{compiler.libc.path}/include" "-I{build.path}/core"
+
+# support precompiled libraries in IDE v1.8.6+
+compiler.libraries.ldflags=
 
 compiler.c.cmd=xtensa-lx106-elf-gcc
-compiler.c.flags=-c {compiler.warning_flags} -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.c.flags=-c "{compiler.warning_flags}-gcc" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
 
 compiler.S.cmd=xtensa-lx106-elf-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls "-I{runtime.tools.xtensa-lx106-elf-gcc.path}/include/"
[... 44 lines cut ...]
 ## Compile S files
-recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -D{build.sdk}=1 -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" {build.led} {build.flash_flags} {compiler.S.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
+recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -D{build.sdk}=1 -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" -DARDUINO_BOARD_ID="{_id}" {build.led} {build.flash_flags} {compiler.S.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Create archives
 recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
 
 ## Combine gc-sections, archives, and objects
-recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {build.exception_flags} -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{archive_file_path}" {compiler.c.elf.libs} -Wl,--end-group "-L{build.path}"
[... cut ...]
-recipe.objcopy.hex.3.pattern="{runtime.tools.python3.path}/python3" -I "{runtime.tools.sizes}" --elf "{build.path}/{build.project_name}.elf" --path "{runtime.tools.xtensa-lx106-elf-gcc.path}/bin" --mmu "{build.mmuflags}"
+recipe.objcopy.hex.3.pattern="{runtime.tools.python3.path}/python3" -X utf8 -I "{runtime.tools.sizes}" --elf "{build.path}/{build.project_name}.elf" --path "{runtime.tools.xtensa-lx106-elf-gcc.path}/bin" --mmu "{build.mmuflags}"
 
 ## Save hex
 recipe.output.tmp_file={build.project_name}.bin

```
Some of the changed lines above has to cause this problem. But I cannot find out, which ones...

mcspr, 2023-01-20:
Well, if `@` is not a problem...

Another guess, after reading extension dependencies and seeing this loop
https://github.com/elektronikworkshop/cocopa/blob/ce125785574735b777e2a40bad0e85dd41580692/src/ParserGcc.ts#L64

What about removing `-g++` and `-gcc` suffixes from warnings?
```diff
diff --git a/platform.txt b/platform.txt
index 120cfb00..04e00110 100644
--- a/platform.txt
+++ b/platform.txt
@@ -76,18 +76,18 @@ compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_S
 compiler.libraries.ldflags=

 compiler.c.cmd=xtensa-lx106-elf-gcc
-compiler.c.flags=-c "{compiler.warning_flags}-gcc" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.c.flags=-c "{compiler.warning_flags}" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.S.cmd=xtensa-lx106-elf-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls "-I{runtime.tools.xtensa-lx106-elf-gcc.path}/include/"

-compiler.c.elf.flags=-g "{compiler.warning_flags}-gcc" -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read
+compiler.c.elf.flags=-g "{compiler.warning_flags}" -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read

 compiler.c.elf.cmd=xtensa-lx106-elf-gcc
 compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 {build.stdcpp_lib} -lm -lc -lgcc

 compiler.cpp.cmd=xtensa-lx106-elf-g++
-compiler.cpp.flags=-c "{compiler.warning_flags}-g++" {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.cpp.flags=-c "{compiler.warning_flags}" {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.as.cmd=xtensa-lx106-elf-as
```

milanbx, 2023-01-20:
no, no, error again:
`xtensa-lx106-elf-g++: error: D:\Portable\Arduino\portable\packages\esp8266\hardware\esp8266\3.1.1/tools/warnings/none: No such file or directory`

mcspr, 2023-01-20:
Sorry, I thought that would've been a no-op for gcc. But .json is generated correctly?

Just removing this warning arg should work then.
```diff
diff --git a/platform.txt b/platform.txt
index 120cfb00..79b26fb2 100644
--- a/platform.txt
+++ b/platform.txt
@@ -76,18 +76,18 @@ compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_S
 compiler.libraries.ldflags=

 compiler.c.cmd=xtensa-lx106-elf-gcc
-compiler.c.flags=-c "{compiler.warning_flags}-gcc" -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.c.flags=-c -std=gnu17 {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.S.cmd=xtensa-lx106-elf-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls "-I{runtime.tools.xtensa-lx106-elf-gcc.path}/include/"

-compiler.c.elf.flags=-g "{compiler.warning_flags}-gcc" -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read
+compiler.c.elf.flags=-g -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/lib/{build.sdk}" "-L{build.path}" "-L{compiler.libc.path}/lib" "-Tlocal.eagle.flash.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read

 compiler.c.elf.cmd=xtensa-lx106-elf-gcc
 compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 {build.stdcpp_lib} -lm -lc -lgcc

 compiler.cpp.cmd=xtensa-lx106-elf-g++
-compiler.cpp.flags=-c "{compiler.warning_flags}-g++" {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}
+compiler.cpp.flags=-c {build.stacksmash_flags} -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 {build.stdcpp_level} -MMD -ffunction-sections -fdata-sections {build.exception_flags} {build.sslflags} {build.mmuflags} {build.non32xferflags}

 compiler.as.cmd=xtensa-lx106-elf-as

```

We'll need figure out a better name for warning files then.

milanbx, 2023-01-21:
YES! This helped! All working fine, back on track, thanks a lot!
Although I don't understand properly, what caused the issue and how you've fixed it, I'm glad it works.

PR #8829 (2023-01-26): Don't use -g++ or -gcc suffixes in command line arguments
It may confuse some greedy parsers matching compiler either with a glob or by iterating each argument and not stopping on the first one

ref. this library used by VSCode Arduino extension
https://github.com/elektronikworkshop/cocopa/blob/ce125785574735b777e2a40bad0e85dd41580692/src/ParserGcc.ts#L63-L68

fix #8809
commit 79dde219 (2023-01-26): Don't use -g++ or -gcc suffixes in command line arguments (#8829)

It may confuse some greedy parsers matching compiler either with a glob
or by iterating each argument and not stopping on the first one

ref. this library used by VSCode Arduino extension
https://github.com/elektronikworkshop/cocopa/blob/ce125785574735b777e2a40bad0e85dd41580692/src/ParserGcc.ts#L63-L68

fix #8809
closed 2023-01-26 by mcspr